betaflight/betaflight #9032: OSD main battery voltage accuracy when vbat < 10V

Dope-johnny, 2019-10-13:
**Describe the bug**
The main bat voltage OSD element shows only 0.1V accuracy although it has 3 siginficant digits when a 2S lipo is plugged in. Average cell voltage, betaflight configurator main bat voltage and telemetry data show 0.01V accuracy.
e.g: OSD: 8.20V, actual voltage in configurator: 8.16V

**To Reproduce**
Plug in 2S battery with BF4.1.0-RC6

**Expected behavior**
0.01V accuracy when Vbat <10V

**Flight controller configuration**
diff:
```# diff

# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct  6 2019 / 06:47:27 (c6452a55c) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF411   custom defaults: YES

# start the command batch
batch start

board_name MATEKF411
manufacturer_id MTKS

# serial
serial 1 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# master
set bat_capacity = 450
set force_battery_cell_count = 2
set ibata_scale = 1283
set osd_vbat_pos = 2346
set osd_avg_cell_voltage_pos = 2314
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 1
set vtx_freq = 5658
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

resource:
```resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

**Setup / Versions**
FC: NamelessRC AIO412
Bat: 2S 450mAh

mikeller, 2019-10-15:
@Dope-johnny: This is an intentional change, done in #7952. (Note that the '3 significant digits' in the pull request title is misleading, it simply rounds to tenths of volts.

Dope-johnny, 2019-10-15:
Alrite. I was indeed peeking at those previous changes and misunderstood it. Although I liked .01 accuracy for 1S/2S quads. What [AndersHoglund](https://github.com/betaflight/betaflight/issues/7868#issuecomment-476218045) said seems like the best approach to me. That extra 0 in the OSD now doesn't have any benefit.

So I think this can be closed now.

closed 2019-10-15 by Dope-johnny