betaflight/betaflight #9033: Kakute F7 - bi-directional DSHOT no longer working

stylesuxx, 2019-10-13:
**Describe the bug**
On 4.1.0 RC6 Bi-Directional DSHOT is no longer working. I have been using it on 4.0.6 without any issues. Now when I enable it, the motors sometimes start twitching randomly, without the copter being armed. In the motors tab 100% Error is shown and I can not spin up the motors. Motors will also not spin when arming the copter.

I tried both, unified and legacy targets.

Trying to find a working version I tried 4.1.0 RC 1 (unified target) after flashing that my FC does not react at all anymore. It poweres up when attached via USB, but that is it - it is no longer recognized as a USB device.

`dshot_telemetry_info` output:
```
# dshot_telemetry_info
Dshot reads: 0
Dshot invalid pkts: 0
Dshot directionChange cycles: 0, micros: 0

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0   NO DATA
    1         0        0       0   NO DATA
    2         0        0       0   NO DATA
    3         0        0       0   NO DATA

0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

**To Reproduce**
Enable bi-directional DSHOT on a Kakute F7 board. Using BLHELI_32 ESC's with firmware 6.4.

**Expected behavior**
I am expecting Bi-Directional DShot to work in 4.1.0.

**Flight controller configuration**
```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.1.0 Oct  6 2019 / 06:29:55 (c6452a55c) MSP API: 1.42

# start the command batch

batch start

board_name KAKUTEF7

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set pid_process_denom = 1
set debug_mode = GYRO_SCALED

profile 0

rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
Kakute F7 flight-controller, Tekko32 4in1 ESC.

**Additional context**
Here is a diff of my working 4.0.6 setup:
```
# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.5 Aug  1 2019 / 12:01:12 (839661b20) MSP API: 1.41

# diff

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.5 Aug  1 2019 / 12:01:12 (839661b20) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name
name stylesuxx

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 16384 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 2 900 1300 0 0
aux 1 13 1 1300 2100 0 0
aux 2 35 0 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 200
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 80
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 128
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set small_angle = 180
set pid_process_denom = 1
set osd_rssi_pos = 2105
set osd_tim_2_pos = 2519
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2507
set osd_warnings_pos = 14601
set osd_avg_cell_voltage_pos = 2487
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 2
set vtx_freq = 5917
set vcd_video_system = PAL

# profile
profile 1

set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set p_pitch = 60
set i_pitch = 45
set d_pitch = 60
set f_pitch = 350
set p_roll = 65
set i_roll = 55
set d_roll = 65
set f_roll = 325
set p_yaw = 100
set f_yaw = 125
set d_min_roll = 45
set d_min_pitch = 45
set d_min_boost_gain = 30
set d_min_advance = 0

# rateprofile
rateprofile 1

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80
set tpa_rate = 80
set tpa_breakpoint = 1750

# end the command batch
batch end
```

IvoFPV, 2019-10-13:
You must update your ESC firmware. BF4.1 uses a different signal to communicate with the ESC which doesn't work on older blheli32 versions.
>bidirectional Dshot, which is the basis for RPM based filtering, has been improved and is now available with BLHeli_32 (from version 32.7 on) and on BLHeli_S hardware (by using the JESC firmware). Use these instructions to get it set up:
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

stylesuxx, 2019-10-13:
OK, thank you - now I feel stupid, I checked in BLHELI_32 suite and the latest version that is offered to me is  32.64. And the newest version of BLHELI_32 suite seems to have a dependency problem with my libcurl - I guess that one would offer version 32.7. I do not see that version for download anywhere...

I guess my issue is solved once I can aquire version 32.7 until then I will stick with 4.0.6. Thanks for your time.

closed 2019-10-13 by stylesuxx