betaflight/betaflight #9034: OSD causing cycle time to spike to over 1000 on iFlight SucceX Mini F4 V2.1 Flight Control

ipocalypse, 2019-10-13:
**Describe the bug**
When trying to run @ 4k/4k or 8k/8k the cycle time will have constant spikes to over 1000 when the OSD is enabled. This causes twitching while flying. I had Iflight send me another board. This one does the exact same thing. CPU never goes over 5%. 

**To Reproduce**
Steps to reproduce the behavior:
Very easy to reproduce. Just set to 4k/4k with DSHOT300 and OSD enabled. Reboot and watch the cycle times go from 250 to over 1000 and back to 250. Turn off OSD and reboot and issue will go away. I have tried this with other F4 boards and have not seen this issue.
**Expected behavior**
I would expect the OSD to not cause a cycle time issue. 

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
'''# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct  6 2019 / 06:47:09 (c6452a55c) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id AIRB

# feature
feature -OSD

# serial
serial 2 8192 115200 57600 0 115200

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 25 200 500 800
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set serialrx_provider = SBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_arming_disable = OFF
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_batt_critical = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_crash_flip = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_tim1 = 2562
set osd_tim2 = 2562
set osd_warnings_pos = 329
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 4
set vtx_freq = 5695
set vcd_video_system = NTSC
set gyro_1_align_yaw = 2700

profile 0

rateprofile 0

# end the command batch
batch end

# '''

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
'''# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: PREINIT 5
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
iFlight SucceX Mini F4 V2.1 Flight Control (Came in the BNF MegaBee V2)
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
SUCCEX MINI F4 V2 FLIGHT TOWER SYSTEM 2-6S WITH 500MW VTX
Tried firmware all the way back to 3.4.1 and issue was there with all of them.
 - how are the different components wired up.
plug and play
**Additional context**
Add any other context about the problem here.

etracer65, 2019-10-13:
> I have tried this with other F4 boards and have not seen this issue.

Well as you already noted, the behavior is the same on firmware versions all the way back to 3.4.1 so that would rule out something firmware specific. Perhaps you should take this up with the board manufacturer.

> I would expect the OSD to not cause a cycle time issue.

Why would you expect that? Flight controllers only have a single microcontroller. There is no multi-threading. Anything that uses processing time affects other parts that need processing time. The OSD is generally CPU intensive - particularly if you add lots of elements.

Also, select the appropriate video type - NTSC or PAL instead of AUTO. The auto detection can cause a big delay as the mode changes or other video sync issues are detected.

etracer65, 2019-10-13:
It's also a possibility that your camera or the cloned max7456 OSD overlay chip used on that flight controller may be out of spec and losing video sync. Every time that happens there is a delay while the OSD chip resynchronizes. You could try capturing a log with `debug_mode = MAX7456_SIGNAL`. Then debug(2) will show a count of the number of re-sync's and the timing (value increases for each re-sync).

ipocalypse, 2019-10-13:
I did all my tests with just the FC and nothing else connected. I can send a video of what I see as well if you would like to see it.  Sent from Mail for Windows 10 From: Bruce LuckcuckSent: Sunday, October 13, 2019 2:47 PMTo: betaflight/betaflightCc: ipocalypse; AuthorSubject: Re: [betaflight/betaflight] OSD causing cycle time to spike to over 1000 on iFlight SucceX Mini F4 V2.1 Flight Control (#9034) It's also a possibility that your camera or the cloned max7456 OSD overlay chip used on that flight controller may be out of spec and losing video sync. Every time that happens there is a delay while the OSD chip resynchronizes. You could try capturing a log with debug_mode = MAX7456_SIGNAL. Then debug(2) will show a count of the number of re-sync's and the timing (value increases for each re-sync).

ipocalypse, 2019-10-13:
I tried many OSD options including turning everything off but still having it enabled. Same issue. Once it is enabled it has the issue. I am sure it is an issue with the manufacture and I suspect it is a CLONE. But I wanted to let people know because I had SOOOO many issues trying to get this thing to fly correctly and disabling the OSD was all it took to fix it.  Sent from Mail for Windows 10 From: Bruce LuckcuckSent: Sunday, October 13, 2019 11:57 AMTo: betaflight/betaflightCc: ipocalypse; AuthorSubject: Re: [betaflight/betaflight] OSD causing cycle time to spike to over 1000 on iFlight SucceX Mini F4 V2.1 Flight Control (#9034) I have tried this with other F4 boards and have not seen this issue.Well as you already noted, the behavior is the same on firmware versions all the way back to 3.4.1 so that would rule out something firmware specific. Perhaps you should take this up with the board manufacturer.I would expect the OSD to not cause a cycle time issue.Why would you expect that? Flight controllers only have a single microcontroller. There is no multi-threading. Anything that uses processing time affects other parts that need processing time. The OSD is generally CPU intensive - particularly if you add lots of elements.Also, select the appropriate video type - NTSC or PAL instead of AUTO. The auto detection can cause a big delay as the mode changes or other video sync issues are detected.

etracer65, 2019-10-13:
> I did all my tests with just the FC and nothing else connected.

Well then that's not really a valid test. Without a camera video signal the OSD overlay chip is going to continuously reinitialize trying to synchronize to the (nonexistent) video input.

Also, not every flight controller powers the overlay chip completely when only on USB. So if that was the case then there would definitely be periodic delays.

ipocalypse, 2019-10-13:
Well it’s the same in the drone as out as far as this issue goes. I just know for 100% that with OSD on the drone twitches horribly and with it off there is no twitch at all. I even put the OSD on a switch and can make it start twitching just by turning it on.  Sent from Mail for Windows 10 From: Bruce LuckcuckSent: Sunday, October 13, 2019 3:39 PMTo: betaflight/betaflightCc: ipocalypse; AuthorSubject: Re: [betaflight/betaflight] OSD causing cycle time to spike to over 1000 on iFlight SucceX Mini F4 V2.1 Flight Control (#9034) I did all my tests with just the FC and nothing else connected.Well then that's not really a valid test. Without a camera video signal the OSD overlay chip is going to continuously reinitialize trying to synchronize to the (nonexistent) video input.Also, not every flight controller powers the overlay chip completely when only on USB. So if that was the case then there would definitely be periodic delays.

etracer65, 2019-10-13:
Then I would suspect that the CPU is having communications problems with the OSD overlay chip. Possibly bad circuit design or bad/failing OSD chip.

ipocalypse, 2019-10-13:
I sent IFLIGHT all this info as well. I just wanted to report it for the community in case someone else has this same board and keeps trying to fight the twitching that issues caused. Sent from Mail for Windows 10 From: Bruce LuckcuckSent: Sunday, October 13, 2019 3:55 PMTo: betaflight/betaflightCc: ipocalypse; AuthorSubject: Re: [betaflight/betaflight] OSD causing cycle time to spike to over 1000 on iFlight SucceX Mini F4 V2.1 Flight Control (#9034) Then I would suspect that the CPU is having communications problems with the OSD overlay chip. Possibly bad circuit design or bad/failing OSD chip.

closed 2019-11-20 by stale[bot]