betaflight/betaflight #9035: SmartAudio softserial does not work on Crazybee F4 pro v1.0 no rx version

barneyflorrick, 2019-10-13:
VTX: FXT fx868t
FC: crazybee f4 pro v1.0 no rx version

SmartAudio wire is soldering on led strip pin.

SmartAudio softserial doesn't work.

Flight controller configuration:
```
# diff

# version
# Betaflight / CRAZYBEEF4DX (C4DX) 4.1.0 Oct  6 2019 / 06:19:50 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

board_name CRAZYBEEF4DX

# resources
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A00

# feature
feature SOFTSERIAL

# serial
serial 30 2048 115200 57600 0 115200

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 200

# master
set vtx_band = 4
set vtx_channel = 2
set vtx_power = 1
set vtx_freq = 5760

profile 0

rateprofile 0

# end the command batch
batch end
```

mikeller, 2019-10-13:
@barneyflorrick: Can we please get the output of `resource show all` as well?

ChenRiTianS, 2019-10-14:
DALRC F405      RUSH Tank
Pitmode is always on, ground station and osd modulation cannot exit,
You can only exit pitmode through the button on the image

etracer65, 2019-10-14:
A couple of things:

1. You've configured the `vtxtable` power section as if this VTX supports SA 2.0. Do you know this to be the case? Has the manufacturer published documentation that says it is? You should look at https://github.com/betaflight/betaflight/wiki/VTX-tables which includes details on how to determine.

2. You have `vtx_low_power_disarm = ON` so any power level changes you make in the OSD menu won't appear to work as it will always stay at the lowest level when disarmed. Try turning this option off and test.

barneyflorrick, 2019-10-14:
> A couple of things:
>
> 1. You've configured the `vtxtable` power section as if this VTX supports SA 2.0. Do you know this to be the case? Has the manufacturer published documentation that says it is? [...]

1. The vtx support SA 2.0, i have tested it on uart2, it works.
2. I have set vtx_low_power_disarm=off , smartaudio softserial does not work either.

RipperDrone, 2019-10-14:
Haven't there been reports about some boards where SA over Softserial didn't work? I remember I changed SA to hard UART in 2 cases in the past due to Softserial not working, presumably it was on SucceX F4 Mini and on Magnum F4 v2 Mini boards. Trying to dig it up...

RipperDrone, 2019-10-14:
Here are some references: https://github.com/betaflight/betaflight/issues/6400, https://github.com/betaflight/betaflight/pull/5076,

closed 2019-11-21 by stale[bot]