betaflight/betaflight #9037: Problem with GPS in SPRACINGF3

firewales, 2019-10-14:
Hello everybody,
I have a problem to configure the GPS in my SPRACINGF3. I know that the GPS in this model are disactivate , and i try to build another version of the betaflight_4.0.6 enabling this features and removing others, following these manuals:

Building in Windows
https://github.com/betaflight/betaflight/blob/master/docs/development/Building%20in%20Windows.md

Customized version
https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

The features i like to change in target.h: 

Eneable:
#define GPS
#define GPS_PROTO_UBLOX

Disable:
#undef USE_DSHOT     // Because i don use this
//#undef TELEMETRY      Is comment because i dont know if it affects other features 

The firmware works good, but i can't eneable the GPS. I don't know if i'm doing something wrong  or something else needs to be change.

[betaflight_4.0.6_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/3724192/betaflight_4.0.6_SPRACINGF3.zip)

There are another version of Betaflight that I can see to make my version?

Thank you so much for everything.

etracer65, 2019-10-14:
You didn't correctly enable the GPS code. This is wrong:
```
Eneable:
#define GPS
#define GPS_PROTO_UBLOX
```
Try:
```
#define USE_GPS
#define USE_GPS_NMEA
#define USE_GPS_UBLOX
```

firewales, 2019-10-14:
In other thread it's commented that #define USE_GPS is deprecated
https://github.com/betaflight/betaflight/issues/4415

>  You need to change #define USE_GPS to #define GPS in this case - a lot of the defines got 

> renamed between 3.2 and 3.3.

And in the manual: https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md

#define GPS

For this reason I put it like this, the manual is wrong or obsolete?

etracer65, 2019-10-14:
The information you're referring to is out of date.

firewales, 2019-10-14:
I try it this afternoon and I close the thread if it works.

Thank you!!!!

firewales, 2019-10-15:
It´s works!!!

Upload the firmware in case someone needs it

The change from 4.0.6 version:
#define USE_PWM

#define USE_GPS
#define USE_GPS_NMEA
#define USE_GPS_UBLOX

#undef USE_DSHOT

[betaflight_4.0.6_SPRACINGF3.zip](https://github.com/betaflight/betaflight/files/3727852/betaflight_4.0.6_SPRACINGF3.zip)

closed 2019-10-15 by firewales