betaflight/betaflight #9039: Motors rotate more and more with Props

nickgrossm, 2019-10-14:
Hello I'm Nick,

i have a problem with mit JB4 with the 4 in1 ESCs. I did calibrate the ESCs and thought I check every settings double, but the Motors are begin to rotate more and more after i push my throttle stick minimal.

When the props are on the quad, and the motors are rotate more and more, the quad is doing a front flip.

My Quad is a Reverb with JB4 and 4 in1 ESC + Xing Motors 2207 2450kV +DJI FPV 

I show you my settings:

# name
name Reverb

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource INVERTER 1 C00
resource INVERTER 2 NONE
resource INVERTER 3 NONE
resource INVERTER 4 NONE
resource INVERTER 5 NONE
resource INVERTER 6 NONE
resource INVERTER 7 NONE
[... 1132 lines cut ...]
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# restore original rateprofile selection
rateprofile 2

# save configuration
save

It will be glad if you have any idea where my problem is.

Thank you 

Nick

IvoFPV, 2019-10-14:
Please provide a `diff` and not a `dump`. As always check if your props are on correctly, not upside down, motor direction and motor order are correct, board orientation.

etracer65, 2019-10-14:
Issues are for reporting problems or bugs with the firmware. This is not the place for help with your configuration or hardware. Try:

https://www.rcgroups.com/forums/showthread.php?2464844-Betaflight-Flight-Controller-Firmware-Discussion-Thread/

https://www.facebook.com/groups/betaflightgroup

https://slack.betaflight.com

closed 2019-10-14 by etracer65

nickgrossm, 2019-10-14:
> 
> 
> Please provide a `diff` and not a `dump`. As always check if your props are on correctly, not upside down, motor direction and motor order are correct, board orientation.

I checked everything double motor direction board orientation.

nickgrossm, 2019-10-14:
where do i check the motor order?

jflyper, 2019-10-14:
Spin motors one by one using the motor tab individual slider.

In addition to motor oder, pay attention if direction of rotation matches with the graphics on the page, too.

I will lock this, as this is clearly a support request. Further assistance can be obtained at places mentioned above.