betaflight/betaflight #9043: Openlog will not record flight data after upgrading to BF 4.06

ResinSmoker, 2019-10-16:
**Describe the bug**
Openlog serial data recorder was working when using BF 3.5.3 with BetaFPV 2-4S F4 (F411) Flight controller. After flashing BF 4.0.6 all thats available is the header information on the SD card. After the flight I can see Copter tune settings but no flight data etc.

Note: Blue and Green LED's can bee seen working on the SD card. Thus indicating that it's attempting to record. ***Copter flies fine, just needs tuning hence the need for Openlog***

**To Reproduce**
Connect the Openlog card to the Flight Controller pad T1 for RX, then 5V and Ground pads respectively.
 
In BF ports tab set up UART 1 for Peripherials "Blackbox" & 250000 baud. 
Please note that "Blackbox"  option is not shown under the "Configuration" tab. From what I understand if this normal. Instead go to the CLI tab and ensure the following is set:

blackbox_p_ratio = 32  (Far as i know of this is the setting for the 1k logging rate)
Allowed range: 0 - 32767

blackbox_device = SERIAL
Allowed values: NONE, SPIFLASH, SDCARD, SERIAL

blackbox_record_acc = ON
Allowed values: OFF, ON

blackbox_mode = ALWAYS
Allowed values: NORMAL, MOTOR_TEST, ALWAYS
Default value: NORMAL

Next save then reset the FC. Check the Blackbox tab to ensure that the Serial option is selected for 1k.

Insert the newly formatted SD card and then power up the copter.

On the SD card after the first power-up, the following should be shown in "config.txt".
250000,26,3,0,1,1,0
baud,escape,esc#,mode,verb,echo,ignoreRX

Do a short flight, maybe 20 seconds. Pull the battery and then the SD card. A newly created log file should be shown but saved in txt format. Rename the .TXT to .BFL and export to BB Explorer.

**Expected behavior**
Some of the flight data logged so the step response can later be plotted.

**Flight controller configuration**
```# 

# diff

# version
# Betaflight / MATEKF411 (MK41) 4.0.6 Sep  1 2019 / 00:50:01 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF411
manufacturer_id 

# name
name Invert WhrlyGig

# resources
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B05
resource MOTOR 4 B04
resource SERVO 1 A08
resource LED_STRIP 1 NONE
resource LED 1 NONE

# timer

# dma

# mixer
mixer TRI

# servo
servo 5 1329 1929 1629 100 -1

# servo mix

# feature
feature -OSD

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 128 115200 57600 0 250000
serial 1 64 115200 57600 0 115200

# led
led 0 0,0::A:0
led 1 1,0::A:0
led 2 2,0::L:0
led 3 3,0::L:0

# color

# mode_color
mode_color 5 0 9
mode_color 5 1 11
mode_color 5 2 2
mode_color 5 3 13
mode_color 5 4 10
mode_color 5 5 3
mode_color 6 0 2

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 2 2 900 1300 0 0
aux 3 28 2 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 100
set gyro_lowpass2_type = BIQUAD
set gyro_lowpass2_hz = 300
set acc_calibration = -155,7,505
set baro_hardware = NONE
set rc_smoothing_type = INTERPOLATION
set serialrx_provider = FPORT
set airmode_start_throttle_percent = 32
set serialrx_halfduplex = ON
set blackbox_mode = ALWAYS
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT300
set align_board_pitch = 180
set align_board_yaw = 180
set bat_capacity = 450
set vbat_detect_cell_voltage = 30
set ibata_scale = 580
set servo_center_pulse = 1520
set servo_pwm_rate = 330
set small_angle = 30
set deadband = 10
set yaw_deadband = 10
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set runaway_takeoff_deactivate_throttle_percent = 25
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_vbat_pos = 2433
set osd_rssi_pos = 2049
set osd_tim_1_pos = 2104
set osd_tim_2_pos = 407
set osd_flymode_pos = 2073
set osd_throttle_pos = 2265
set osd_vtx_channel_pos = 2081
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_current_pos = 374
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 2439
set osd_warnings_pos = 2377
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set debug_mode = GYRO_FILTERED
set vcd_video_system = NTSC

# profile
profile 0

set vbat_pid_gain = ON
set iterm_relax = OFF
set p_pitch = 20
set i_pitch = 10
set d_pitch = 10
set p_roll = 20
set i_roll = 10
set d_roll = 10
set p_yaw = 16
set i_yaw = 10
set d_yaw = 16
set d_min_roll = 0
set d_min_pitch = 0

# rateprofile
rateprofile 0

set roll_expo = 50
set pitch_expo = 50
set yaw_srate = 30
set tpa_rate = 10
set tpa_breakpoint = 1650
set throttle_limit_type = SCALE
set throttle_limit_percent = 50

# end the command batch
batch end

# 
```
Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
``` 
# resource
resource BEEPER 1 B02
resource MOTOR 1 B06
resource MOTOR 2 B07
resource MOTOR 3 B05
resource MOTOR 4 B04
resource MOTOR 5 B03
resource MOTOR 6 B10
resource SERVO 1 A08
resource PPM 1 A03
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 2 C14
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
resource USB_DETECT 1 C15
``` 

**Setup / Versions**
- BetaFPV F4 2-4S https://betafpv.com/collections/brushless-flight-controller/products/f4-2-4s-aio-brushless-flight-controller-12a-blheli_s
- FrSky R-XSR connected via F-Port on TX2 pad (Telemetry enabled)
- Openlog serial data logger (TX1): https://www.sparkfun.com/products/13712

 - how are the different components wired up.
[image]

[image]

**Additional context**
Thank you in advance!

mikeller, 2019-10-16:
@ResinSmoker: Can you please post an example log file?

etracer65, 2019-10-17:
Couple of things:
```
set blackbox_mode = ALWAYS
```
This is not the normal way to perform logging and it's very possible that this may not work well with serial logging. Try setting this back to default with the value `NORMAL`. You'll have to set up a blackbox mode associated with an aux channel to activate/deactivate logging.

```
set debug_mode = GYRO_FILTERED
```
This just adds substantial data to the log for no extra value. The output is the same as the normal gyro traces. Set to `NONE` to reduce the data volume logged.

In reality the OpenLog serial logger is really no longer viable as it can't keep up with the data volume. We recommend the more modern OpenLager.

jflyper, 2019-10-18:
Btw, how large is a log record now?

jflyper, 2019-10-18:
@ResinSmoker Ah, sorry, my question was for devs, and it was "a size per record". I'm suspecting the cause of this issue as {record size} x {record / sec} exceeding output bandwidth or output buffer size not enough for batch output behavior of the black box.

jflyper, 2019-10-19:
@ResinSmoker

Can you do another test with lower logging rate at 500Hz? 

You don't actually have to fly; just arm for few seconds and disarm.

jflyper, 2019-10-19:
I did some follow up test.

Test equipments
Matek F411 + 4.1.0 (Legacy)
OpenLager + 8GB microSDHC, speed class 4 (el cheapo $3 card I could find at drug store across the street)

In summary, I would say 250Kbps (in conjunction with internal UART transmit buffer size) is not a enough bandwidth for 1KHz logging rate with ACC under 4.1.0 firmware. If I excluded the ACC logging, this logging rate can be handled. But I suspect it is at a marginal region.

I also did the faster logging rates (all with ACC records)

| Logging        | Minimum |
| ------------- |:-------------:| 
| 8KHz      |  2Mbps |
| 4KHz      | 1.5Mbps  | 
| 2KHz      | 1.5Mbps |

Too bad OpenLager does not support more speeds between 250K and 1.5M that I couldn't conduct more detailed test.

etracer65, 2019-10-19:
> In summary, I would say 250Kbps (in conjunction with internal UART transmit buffer size) is not a enough bandwidth for 1KHz logging rate with ACC under 4.1.0 firmware. If I excluded the ACC logging, this logging rate can be handled. But I suspect it is at a marginal region.

So in the context of this issue, this just reinforces the expectation that 1KHz logging with a legacy OpenLog on 4.1 is not practical. Possibly might be barely functional if the ACC is disabled and definitely disable the `debug_mode = GYRO_FILTERED` (which adds a lot to the log). The point is that 4.1 adds additional data to the logging over previous versions. So with the limited bandwidth of the legacy OpenLog it becomes unworkable.

jflyper, 2019-10-19:
Yet another follow up.

I inserted a code in `blackbox_io.c` so that average output bit rate in [Kbps] every 100 [ms].
One byte output is counted as 10-bits on wire to account for start and stop bits.

For 1KHz logging rate, the output bit rate stays over 250Kbps most of the time (peak at 270Kbps).
[image]

If ACC was not recorded, it consistently stayed below 250Kbps. (Peak was at 232Kbps, just barely more than 230,400bps can handle).
[image]

So, to conclude, blackbox logging rate (by volume) increased in 4.1.0 (and may be 4.0.x) as a result of  users demanding more and more items to be logged for better and more diverse flight behavior analysis, that 1KHz logging rate can not be handled within 250Kbps bandwidth that legacy OpenLog can handle. If ACC is not logged, then it could still be handled within the 250Kbps bandwidth, but I believe it is a matter of time before this method breaks also.

ResinSmoker, 2019-10-19:
> @ResinSmoker
> 
> I did some test with Matek F411 flashed with 4.1.0 and openlogger, and found records dropping at logging rate of 1KHz. Lowering the rate to 500Hz showed some improvement.
> 
> Can you do another test with lower logging rate at 500Hz?
> 
> You don't actually have to fly; just arm for few seconds and disarm.

Well i currently cant do that with the copter in question and would need to install the OpenLog to another unit. Beyond that I’ve experienced log gaps even at 500Hz but never have seen the log totally drop out after creating the header.

jflyper, 2019-10-19:
Regarding the how log files are broken, I was never able to reproduce the "header only" result. All I could produce was a header followed by butchered logs. And as the logging module in 4.1.0 has been proven to produce perfect log stream if enough bandwidth is given, your solution is one of the followings.

1. Stick with 3.5.x.

2. Build a custom firmware with decreased log items.

3. Migrate to OpenLager.

4. Migrate to an FC with onboard flash or uSD card.

mikeller, 2019-10-19:
@jflyper: Did you try with `blackbox_mode = ALWAYS`? This is a somewhat hackish mode that is intended for use when testing things like gyro behaviour with no RC input set up, but it has many shortcomings, so I am not surprised if the format gets mangled up when logging over serial.

jflyper, 2019-10-20:
I used `ALWAYS` for measurement using #9060.

jflyper, 2019-10-21:
Here is an updated table of varying logging rates and output bandwidths.

Measured under following conditions:
- 8K gyro sampling, 8K PID loop
- Blackbox recording mode normal (`blackbox_mode = NORMAL`)
- ACC recording on (`blackbox_record_acc = ON`)

| Logging rate (KHz) | Max rate (Kbps) | Peak (Kbps) | OpenLager bps setting for no drop recording |
|--------------------|-----------------|-------------|----------------------------------------------|
| 0.5                            | 281                      | 162               |  250K                                                                     |
| 1                                | 395                     | 302               |  1.5M                                                                     |
| 2                               | 622                      | 590               |  1.5M                                                                     |
| 4                               | 1122                     | 1187              |  2.0M                                                                     |
| 0.5                            | 2308                   | 2308              |  2.47M                                                                     |

haslinghuis, 2019-10-21:
@jflyper Request: Please Change text `Openlog` to `Openlager` BF Configurator / Blackbox tab: 
```
Outboard serial logging device`
`You can log to an external logging device (such as an **OpenLog** or compatible clone) by using a serial port. Configure the port on the Ports tab.
```

ResinSmoker, 2019-10-22:
> @ResinSmoker
>
> I did some test with Matek F411 flashed with 4.1.0 and openlogger, and found records dropping at logging rate of 1KHz. Lowering the rate to 500Hz showed some improvement.
>
> Can you do [...]

[3 inch Whirlygig using BF 4-1 and RPM filters.zip](https://github.com/betaflight/betaflight/files/3754683/3.inch.Whirlygig.using.BF.4-1.and.RPM.filters.zip)

With the kids feeling better and in school, i was able to make another log. This time, using BF 4.1, RPM filtering and a Serial rate of 500Hz. Please keep in mind that the little Tricopter isn't too happy with the BF 4.1 default PID's (hot motors) and as such will have the values dropped considerably before the next flight.

ResinSmoker, 2019-10-22:
Here is another flight with reduced PID;s and logging at 1000 Hz.
[2nd Flight with reduced PIDs and serial rate at 1Khz.zip](https://github.com/betaflight/betaflight/files/3754782/2nd.Flight.with.reduced.PIDs.and.serial.rate.at.1Khz.zip)

The BB Explorer plot does have gaps this time but was able to record enough to create a Step-Response plot.

ResinSmoker, 2019-10-22:
Ok, got one that failed to log but still shows the header information. I've left it in TXT format so you can easily open it with just Notepad.
[LOG00003.TXT](https://github.com/betaflight/betaflight/files/3754874/LOG00003.TXT)

jflyper, 2019-10-22:
@ResinSmoker 
> Ok, got one that failed to log (3rd short flight) but still shows the header information. Not sure why this works on some flights and not others. I've left it in TXT format so you can easily open it with just Notepad.

Anything can happen when buffer is overflowing. Again, we are not doing any bandaid work on this matter.

ResinSmoker, 2019-10-23:
@jflyper
If you read further up, you originally requested this information.

Cheers

closed 2019-11-25 by ResinSmoker

ilya-epifanov, 2021-02-02:
Theoretically, >600kbps UART will be enough.

The openlog blackbox firmware supports up to 1Mbps: https://github.com/cleanflight/blackbox-firmware/blob/master/OpenLog_v3_Blackbox/OpenLog_v3_Blackbox.ino#L87

[image]

192.3µs / 499µs * 1500000bps
578056.1122244489bps

haslinghuis, 2021-02-04:
@ilya-epifanov better use openlager