betaflight/betaflight #9044: Burn-down for the creation of Unified Target configurations

mikeller, 2019-10-16:
The following targets still need to have a Unified Target configuration created:

- [ ]  AIRBOTF4
- [ ]  AIRBOTF4SD
- [ ]  ALIENWHOOPF4
- [ ]  BETAFLIGHTF4
- [ ]  BLUEJAYF4
- [ ]  COLIBRI
- [ ]  COLIBRI_OPBL
- [ ]  CRAZYBEEF4DX
- [ ]  CRAZYBEEF4FS
- [ ]  CRAZYFLIE2
- [ ]  CRAZYFLIE2BQ
- [ ]  DALRCF405
- [ ]  ELLE0
- [ ]  EXUAVF4PRO
- [ ]  F4BY
- [ ]  FF_FORTINIF4
- [ ] FF_FORTINIF4_REV03
- [ ] FF_PIKOF4
- [ ] FF_PIKOF4OSD
- [ ] FISHDRONEF4
- [ ] FOXEERF405
- [ ] FRSKYF4
- [ ] FURYF4
- [ ] HAKRCF405
- [ ] KAKUTEF4
- [ ] KIWIF4
- [ ] KIWIF4V2
- [ ] KROOZX
- [ ] LUXF4OSD
- [ ] MATEKF405
- [ ] NOX
- [ ] NUCLEOF446RE
- [ ] OMNINXT4
- [ ] PIRXF4
- [ ] PLUMF4
- [ ] PODIUMF4
- [ ] PYRODRONEF4
- [ ] REVO
- [ ] REVOLT
- [ ] REVOLTOSD
- [ ] REVONANO
- [ ] REVO_OPBL
- [ ] SKYZONEF405
- [ ] SOULF4
- [ ] SPARKY2
- [ ] SPEKTRUMF400
- [ ] SPRACINGF4EVO
- [ ] SPRACINGF4NEO
- [ ] STACKX
- [ ] STM32F405
- [ ] STM32F411
- [ ] STM32F411DISCOVERY
- [ ] STM32F4DISCOVERY
- [ ] UAVPNG030MINI
- [ ] VRRACE
- [ ] WORMFC
- [ ] XRACERF4
- [ ] YUPIF4
- [ ] AG3XF7
- [ ] ALIENWHOOPF7
- [ ] ANYFCF7
- [ ] ANYFCM7
- [ ] DALRCF722DUAL
- [ ] HAKRCF722
- [ ] KAKUTEF7V2
- [ ] KISSFCV2F7
- [ ] NUCLEOF7
- [ ] NUCLEOF722
- [ ] OMNIBUSF7
- [ ] OMNINXT7
- [ ] STM32F745
- [ ] STM32F7X2
- [ ] YUPIF7

Docteh, 2019-10-17:
> COLIBRI_OPBL
> [these are not sequential in the list]
> REVO_OPBL

My understanding is that OPBL means Open Pilot Boot Loader, and might require a different unified target.

>- [ ]  STM32F405
>- [ ] STM32F411
>- [ ] STM32F745
>- [ ] STM32F7X2

Aren't these the unified targets?

mikeller, 2019-10-17:
Good points, @Docteh.

Re _OPBL, they are, and they do / would require a different Unified Target. Also, the hardware can be used (in an easier way) in Betaflight by wiping the OPBL and installing the respective non-OPBL targets, so the only reason for the _OPBL version (to my knowledge) is so that users can trial Betaflight without having to re-flash OPBL if they want to switch back - not sure if this is still a relevant use case, or if we should stop supplying these targets. I'll get some analytics data.

> STM32F405

Oops, the list was generated by a script, don't script at 2 am kids. :sleeping:

IvoFPV, 2019-10-17:
>MATEKF405

This one can be removed from the list. That one is separated into MATEKF405CTR, MATEKF405STD, MATEKF405MINI configs which have different gyro alignments.

mikeller, 2019-10-17:
@IvoFPV: Thanks, updated.

@Docteh: I had a look at the analytics data, and uptake for the OPBL targets is very small:

Out of 1,109,478 targets that were flashed in the last month, only 586 were OPBL targets (0.05%). So I think creating a Unified Target for these is probably not worth the effort, we can just keep releasing the non-unified targets for these.

Docteh, 2019-10-17:
@mikeller that is 586 more than I expected. 😮 Is anyone connecting to the OPBL targets after flashing?

mikeller, 2019-10-17:
@Docteh: That's still 10000 less than users flashing a NAZE target in the same timeframe, and we have not released those in years. :wink: 

Recognising OPBL targets when connecting to them isn't really possible, as they identify / behave the same as the respective non-OPBL targets, which is kind of the idea (same target, only different memory layout).

closed 2026-09-26 by blckmn