betaflight/betaflight #9045: F3 board (colibri_race) - compile working GPS support later 3.2.5

vandusinae, 2019-10-16:
I wanted to add GPS support to my colbri_race and started with 3.2.x branch and it works. Then I compiled 4.0, 3.3, 3.4, 3.5, which won't work. In the Betaflight Configurator I can enable GPS on UART 3 (ports tab) and can enable GPS (UBLOX) in the configurator tab. Then I get the GPS tab, but it won't find any satellites and the GPS module status led flashes already green. If I flash the fc back to 3.2.5 GPS works fine. 

I used #define USE_GPS to add GPS support (which is disabled since 3.2.x afaik).  

And to clear up some space I removed obsolete functions

#undef USE_SERIALRX_SUMD
#undef USE_SERIALRX_SUMH
#undef USE_SERIALRX_XBUS
#undef USE_SERIALRX_IBUS
#undef USE_SERIALRX_SPEKTRUM
#undef USE_SPEKTRUM_BIND
#undef USE_SPEKTRUM_BIND_PLUG
#undef USE_SPEKTRUM_REAL_RSSI
#undef USE_SPEKTRUM_FAKE_RSSI
#undef USE_SPEKTRUM_RSSI_PERCENT_CONVERSION
#undef USE_SPEKTRUM_VTX_CONTROL

I believe something has changed since 3.3.x. Is there something I'am missing?

etracer65, 2019-10-16:
Make sure you define:
```
#define USE_GPS
#define USE_GPS_NMEA
#define USE_GPS_UBLOX
```

vandusinae, 2019-10-16:
#define USE_GPS_UBLOX does the trick.

Tested 3.5.x and 4.0.x

Thank you very much!

closed 2019-10-16 by vandusinae