betaflight/betaflight #9050: Separate pilot name from craft name for OSD

smitty97, 2019-10-17:
I'm requesting that a separate "Pilot Name" field be added to Config...Personalization so that it can be displayed on the OSD.  Optionally, this could be part of the OSD tab so it could have a different value for each of the OSD profiles.

The craft name is used when saving diff files, so it should contain something that identifies the build.  I don't need this displayed on the OSD however.  Since cameras get smashed a lot, I'd prefer not to fiddle with its separate OSD, and use the craft name to show my pilot name.  To keep diff files sane, my pilot name is appended with abbreviated frame/FC/motor names, which looks terrible on screen but necessary to keep diffs organized.

If there is a single pilot name in the personalization section, it could also be a prefix on the cli diff filename. (BTFL_PILOT_CRAFT_date.txt)

McGiverGim, 2019-10-17:
There are two OSD elements: craft/pilot name (used to save files, etc.), and display name (that can be shown only in the OSD). But I don't know if this is an OSD profile property or is global.

etracer65, 2019-10-17:
There are already two distinct elements.

`set name = `
- Optionally displayed in the OSD as the "Craft Name" element.
- Also used in blackbox file naming.

`set display_name = `
- Optionally displayed in the OSD as the "Display Name" element.

Sounds like you should put the pilot name in the `display_name` and put the quad's name in the `name` field.

smitty97, 2019-10-17:
Oh!  That's wonderful.  I guess at this point it's more of an issue for bfconfigurator then, unless I am blind and didn't see it there.

Docteh, 2019-10-18:
@smitty97 I think the "Display Name" OSD element was added for 4.1.0, so changing the actual text isn't available in the CLI yet. New features tend to be CLI only when first added

[image]

Also of interest are OSD, PID and rate profile names
[image]

If you want to share a quad between some pilots...

smitty97, 2019-10-18:
Actually, that is what i'm heading towards.  My girlfriend and I share almost identical builds- if i set an aux channel high on my tx, and low on hers and have it tied to name/rateprofile, we can just rebind if we need to switch quads at a race in a hurry and itll be all set up.

closed 2019-10-18 by smitty97