betaflight/betaflight #9052: Gyro drifting in flight

Azuxul, 2019-10-17:
**Describe the bug**
In angle mode, after few seconds the quadcopter get a bad default target angle.
Gyro was correctly calibrated
At the end i need to push my sticks at max to stabilize.

https://youtu.be/6fQ3QYtDiq0

**To Reproduce**
On tinywoop with racestar BR0804 motors

**Expected behavior**
No gyro or low gyro drift over time

**Flight controller configuration**
```
# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.1.0 Oct 16 2019 / 11:48:35 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name OMNIBUSF4SD

# name: azuxul

# resources
resource MOTOR 1 A02
resource MOTOR 2 A03
resource MOTOR 3 B01
resource MOTOR 4 B00
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B06

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature SOFTSERIAL

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 0 900 1700 0 0
aux 2 13 3 1700 2100 0 0
aux 3 28 0 1700 2100 0 0
aux 4 36 4 1700 2100 0 0

# rxrange
rxrange 1 2000 1000

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 2
vtxtable powervalues 0 1
vtxtable powerlabels 25 200

# master
set acc_calibration = 35,111,-339
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1075
set max_check = 1930
set rssi_channel = 14
set serialrx_provider = SBUS
set motor_pwm_protocol = DSHOT600
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 8
set yaw_deadband = 8
set pid_process_denom = 2
set osd_vbat_pos = 2423
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2401
set osd_craft_name_pos = 2412
set osd_disarmed_pos = 2282
set debug_mode = SMARTAUDIO
set vtx_band = 1
set vtx_channel = 8
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5725
set vcd_video_system = NTSC
set name = azuxul

profile 0

rateprofile 0

# end the command batch
batch end

# 
```
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 1
A03: MOTOR 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A15: OSD_CS
B00: MOTOR 4
B01: MOTOR 3
B03: PREINIT 2
B04: BEEPER
B05: LED 1
B06: SERIAL_TX 11
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM3:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM4:
    CH1: SERIAL_TX 11
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Betaflight / OMNIBUSF4SD (OBSD) 4.1.0 Oct 16 2019 / 11:48:35 (c37a7c91a) MSP API: 1.42
 - Flight controller : Tenny F4 Pro (F4 1'')

**Additional context**
Same issue with Betaflight 4.x

Thanks for your support

etracer65, 2019-10-17:
This has nothing to do with the gyro. It is caused by either an improperly calibrated accelerometer, or by having too much vibration from the motors/props affecting the accelerometer (vibrations are acceleration). Start by calibrating the accelerometer by placing the quad on a known level surface and calibrate in the Setup tab. If you still have problems you're going to need to work on reducing vibration - better props, replace motors, soft-mount the flight controller, etc.

closed 2019-10-17 by etracer65

jesenator, 2020-05-14:
I have a similar problem. I was having no problems before flashing the newest firmware on the omnibus f4 flight controller, but then after, the gyro was misaligned by 90 degrees for some reason, and there is excessive drifting like described. You can even see the heading value slowly count up when the quad is sitting on the table, so that means it couldn't be caused by vibrations Also, in the sensors tab all of the x, y, and z values are offset by a couple of deg/s. This means that the flight controller thinks the quad is continuously rotating. This is also with the quad sitting stationary and unpowered on the table.
[image]

Robokishan, 2020-10-05:
Hey i am also having same problem gyro starts drifting no accelerometer perticular only gyro values start drifting

haslinghuis, 2020-10-05:
No drifting with BFC 10.8 on MATEKF411 4.3. Recalibrate as @etracer65 suggests otherwise it's probably faulty hardware.

Robokishan, 2020-10-06:
After  throttling down it goes to normal again. And then again start drifting so it's I don't think it's an hardware issue

haslinghuis, 2020-10-06:
Sound like electrical noise to me. Try to add a low esr capacitor.

Robokishan, 2020-10-06:
So after doing pid to 8khz everything works fine
 Mine was first at 1khz. I was doing some experiments with 1khz but 8khz everything is smooth