betaflight/betaflight #9054: Blackbox log does not record the unified target name

Cleric-K, 2019-10-18:
Blackbox logs the `Firmware revision` header line which is visible in the bb explorer as:
[image]

Since unified targets have the common target (e.g. `STM32F405`), the actual target name is not recorded. It can be taken from the `board_name` setting.

We have two basic approaches:
* Add the `board_name` to the `Firmware revision` string. This will be quick and easy solution.
* Add separate header for the `board_name`. This will require also modification of the explorer to display it.

McGiverGim, 2019-10-18:
The `version` CLI command shows this and other information. So I think is a good idea to use the same format:

https://github.com/betaflight/betaflight/blob/1c8d41b952c18d6c760b0a9ab52adb24234b784c/src/main/cli/cli.c#L4764-L4773

Cleric-K, 2019-10-18:
The `version` command seems to return two line output:
```
# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4   custom defaults: YES
```
I'm not sure if is possible to have two-line header in the blackbox log.

mikeller, 2019-10-19:
As a heads up, I'll probably look into changing the second line in CLI to return the changeset id for a Unified Target configuration's last modification after `custom defaults:` in CLI - this will allow us to pinpoint the exact contents of the configuration.

chrissng, 2020-01-22:
Addressed in https://github.com/betaflight/betaflight/pull/9372 and https://github.com/betaflight/blackbox-log-viewer/pull/428

McGiverGim, 2020-01-22:
True, I close this, thanks!

closed 2020-01-22 by McGiverGim