betaflight/betaflight #9057: Issues with SmartAudio 2.1 in 4.1 Rush Tank

rickgomez2003, 2019-10-18:
The issue i found is with SmartAudio 2.1 and the Rush Tank. after upgrading from RC6 to 4.1 I can no longer take the VTX out of pit mode nor can I change the settings IE (power, Band, Channel). I have tested on four different rigs #1 Iflight tiwng F7, #2 Matek F722SE, #3 SP Racing H7 Extreme, #4 Iflight F411 Pro.   All caused the same issue.

I am looking for a solution to this issue to allow me to adjust the setting as before either from the OSD or theough the LUA script.

**Describe alternatives you've considered**
I have not tested any other way other that button presses.

**Additional context**
This appeared to be working in RC6 but now that I have upgraded it does not work and even after the downgrade the issue persist. 

Please let me know if I can provide any further information. Thank you for all that you do.

IvoFPV, 2019-10-18:
Rush Tank should be SmartAudio 2.0. Have you checked your version by using the debug mode as explained on the wiki or you got that it is 2.1 from the vtx specifications?

rickgomez2003, 2019-10-19:
> Rush Tank should be SmartAudio 2.0. Have you checked your version by using the debug mode as explained on the wiki or you got that it is 2.1 from the vtx specifications?

Thank you so much for the reply.  I have checked the version via the debug and get an initial of 216 that 222. This just started changing after I upgraded to 4.1.  Please let me know what I can provide.

rickgomez2003, 2019-10-19:
[image]

[image]

rickgomez2003, 2019-10-19:
@IvoFPV Here are screen shots of Betaflight.

IvoFPV, 2019-10-19:
That would be smartaudio 2.0 so get the proper vtxtable from the wiki and it should work.

rickgomez2003, 2019-10-19:
> That would be smartaudio 2.0 so get the proper vtxtable from the wiki and it should work.

I will try it right away thank you. What would constitute 2.1 out of curiosity?

IvoFPV, 2019-10-19:
You have that information on the wiki. But SA 2.1 would be 300 or 316.

rickgomez2003, 2019-10-19:
> You have that information on the wiki. But SA 2.1 would be 300 or 316.

Thank you I am testing now. I will update shortly.

rickgomez2003, 2019-10-19:
No change even with new settings. I have tried on all four rigs.  What else can I try? 

I am using this config:

SmartAudio 2.0 device
# This example enables a lot of power levels and channels.
# Almost nobody will be able to legally use this without modification.
# Check your local laws and regulations before use!
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

This matches the Rush Tank that I am using.

IvoFPV, 2019-10-19:
Please provide a `diff`.

rickgomez2003, 2019-10-19:
# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF7_TWIN_G
manufacturer_id IFRC
mcu_id 0049002e3237510231313935
signature 

# name: Real-1-F7

# feature
feature -SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 4 1024 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1400 1600 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# rxrange
rxrange 0 1000 2001
rxrange 1 1000 2001
rxrange 2 1000 2001
rxrange 3 1000 2001

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set bat_capacity = 1250
set ibata_offset = 20
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_link_quality = ON
set osd_rssi_alarm = 80
set osd_cap_alarm = 1250
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_link_quality_pos = 14380
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14826
set osd_warnings_pos = 14345
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_2_align_yaw = 900
set name = Real-1-F7

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

IvoFPV, 2019-10-19:
You have low power disarm enabled. This means when you change the power setting while testing right now they wont be applied. Only when you arm vtx will go to the set power.

rickgomez2003, 2019-10-19:
Set it to:

set vtx_low_power_disarm = OFF

rebooted and still no change. When I try to make changes via the OSD it just goes back.

rickgomez2003, 2019-10-19:
Here is the DIff from one of the others doing the same thing:

# 

# Diff all

# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Oct 16 2019 / 11:56:09 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SPRACINGH7EXTREME
mcu_id 003100343437511931353431
signature 

# name: GEP-LX5-H7

# feature
feature -LED_STRIP
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1625 2075 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# rxrange
rxrange 0 1000 2001
rxrange 1 1000 2001
rxrange 2 1000 2001
rxrange 3 1000 2001

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1250
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_link_quality_pos = 46
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14826
set osd_warnings_pos = 14346
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set vcd_video_system = PAL
set name = GEP-LX5-H7

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-19:
Here is another. 

# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct  6 2019 / 06:48:40 (c6452a55c) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id MTKS
mcu_id 002200383237510c31313935
signature 

# name: SpaceOne-F722

# feature
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1400 1600 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0
aux 7 40 2 1400 2100 0 0
aux 8 41 2 1800 2100 0 0

# rxrange
rxrange 0 1000 2001
rxrange 1 1000 2001
rxrange 2 1000 2001
rxrange 3 1000 2001

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 425
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 340
set dyn_lpf_gyro_max_hz = 850
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1550
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_link_quality_pos = 45
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14825
set osd_warnings_pos = 14345
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = SpaceOne-F722

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 119
set dyn_lpf_dterm_max_hz = 289
set dterm_lowpass2_hz = 255

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-19:
The one rig i have that runs the Tramp protocol function perfectly.   Here is the Diff for this quad:

# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct  6 2019 / 06:47:27 (c6452a55c) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF411_PRO   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF411_PRO
manufacturer_id IFRC
mcu_id 004f002a3437510c32363736
signature 

# name: TurboBee136R

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1300 1700 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# rxrange
rxrange 0 1000 2001
rxrange 1 1000 2001
rxrange 2 1000 2001
rxrange 3 1000 2001

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOW      L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 25 100 200
vtxtable powerlabels 25 100 200

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 350
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 550
set osd_alt_alarm = 300
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_link_quality_pos = 45
set osd_tim_1_pos = 14808
set osd_tim_2_pos = 14776
set osd_flymode_pos = 14368
set osd_vtx_channel_pos = 14400
set osd_craft_name_pos = 14826
set osd_warnings_pos = 14345
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14842
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5806
set vcd_video_system = PAL
set name = TurboBee136R

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set p_pitch = 55
set i_pitch = 108
set d_pitch = 46
set f_pitch = 114
set p_roll = 50
set i_roll = 102
set d_roll = 42
set f_roll = 108
set p_yaw = 36
set i_yaw = 108
set f_yaw = 108
set d_min_roll = 24
set d_min_pitch = 26

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-20:
I have not found this issue on on any of my quads with Tramp VTXs.

rickgomez2003, 2019-10-20:
Have the same issue with 2.1 using the TBS Unify-Pro32 Nano.

rickgomez2003, 2019-10-20:
[image]

rickgomez2003, 2019-10-20:
# 

# diff all

# version
# Betaflight / MATEKF405 (MKF4) 4.1.0 Oct 16 2019 / 11:42:57 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
mcu_id 003a00583436510339373635
signature 

# name: Matek-F405

# feature
feature -SOFTSERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 2048 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1400 1600 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1550
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14825
set osd_warnings_pos = 14346
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = SMARTAUDIO
set vtx_band = 3
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5925
set vcd_video_system = PAL
set name = Matek-F405

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-20:
Sorry Uploaded the wrong diff.

# 

# diff all

# version
# Betaflight / MATEKF405 (MKF4) 4.1.0 Oct 16 2019 / 11:42:57 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
mcu_id 003a00583436510339373635
signature 

# name: Matek-F405

# feature
feature -SOFTSERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 2048 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1400 1600 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOW      L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 3
vtxtable powervalues 14 20 26
vtxtable powerlabels 25 100 400

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1550
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14825
set osd_warnings_pos = 14346
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = GYRO_SCALED
set vtx_band = 3
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5925
set vcd_video_system = PAL
set name = Matek-F405

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

IvoFPV, 2019-10-20:
>I am looking for a solution to this issue to allow me to adjust the setting, as before either from the OSD or theough the LUA script.

If you select the channel,  band, power in the configurator and then hit save, does it get changed? When changing a setting in the OSD you must SET the values. Just going back and hitting save will not apply the settings. Same thing for the LUA script, after changing, you must save the settings by long pressing menu and selecting 'Save page'

rickgomez2003, 2019-10-20:
Thank you for all the replies I really appropriate it .

I have tried in the OSD, through the LUA script on my X9D, and through the configurator.  The issue still persist I have tried on all my set ups just as you have described.  

Please let me know what else to try.   

Thank you again!!!!!!

IvoFPV, 2019-10-20:
You do have the VTX PIT MODE on a switch. Can you confirm you have the switch/mode off when testing this?

IvoFPV, 2019-10-20:
Also, you have failsafe channel fallback set on AUX 3 where VTX PIT MODE is set to value of 2000, meaning it is enabling the mode in that case.

rickgomez2003, 2019-10-20:
I have tested with and with out the failsafe set up. I have also tried with the switch configured and not configured. 

I will test all again just to be sure. I will update shortly after I test all the set ups.

rickgomez2003, 2019-10-20:
Just tested and issue still persist in all my set ups with 4.1. 

Just as a test I rolled back my Matek F722SE to 4.06 and tested and it works fine in that release, I am able to use the LUA script and the OSD to make changes with out issue.  I then re-flashed with full erase and the set all the setting by hand as not to introduce any prior miss configurations and the problem came back.

Please let me know your thoughts thank you for your time.

IvoFPV, 2019-10-20:
Could you provide that 'diff' that you configured now from scratch?

rickgomez2003, 2019-10-20:
Here is the diff from the BF4.06 config.

# 

# diff all

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.6 Sep  1 2019 / 00:51:20 (2a64051a2) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id 
mcu_id 002200383237510c31313935
signature 

# name
name SpaceOne-F722

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 26 0 1900 2100 0 0
aux 2 30 1 1400 1600 0 0
aux 3 35 1 1900 2100 0 0
aux 4 36 3 1900 2100 0 0
aux 5 39 2 1400 2100 0 0
aux 6 40 2 1900 2100 0 0
aux 7 41 2 1900 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set min_throttle = 1070
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_vtx_channel_pos = 14570
set osd_warnings_pos = 14346
set vtx_band = 1
set vtx_freq = 5865

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-20:
Here is the 4.1 by hand.

# 

# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id MTKS
mcu_id 002200383237510c31313935
signature 

# name: SpaceOne-F722

# feature
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 26 0 1900 2100 0 0
aux 2 35 1 1900 2100 0 0
aux 3 36 3 1900 2100 0 0

# rxrange
rxrange 0 1000 2001
rxrange 1 1000 2001
rxrange 2 1000 2001
rxrange 3 1000 2001

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 400
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 320
set dyn_lpf_gyro_max_hz = 800
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1550
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14826
set osd_warnings_pos = 14346
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
set vtx_freq = 5806
set vcd_video_system = PAL
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = SpaceOne-F722

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 112
set dyn_lpf_dterm_max_hz = 272
set dterm_lowpass2_hz = 240

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-20:
In the OSD when I try to change things it switches back right away, same with the LUA script. Just an FYI.

This does not happen in 4.06

rickgomez2003, 2019-10-20:
If someone could point me to the spot were this is configured in the source I can try and do a stare and compare between 4.06 and 4.1.

mikeller, 2019-10-21:
@rickgomez2003: Is this now solved with your statement in https://github.com/betaflight/betaflight/issues/9078#issuecomment-544305564?

rickgomez2003, 2019-10-21:
> @rickgomez2003: Is this now solved with your statement in [#9078 (comment)](https://github.com/betaflight/betaflight/issues/9078#issuecomment-544305564)?

It is solved for me, but is this not suppose to work with the custom setting?

mikeller, 2019-10-21:
@rickgomez2003: That depends on the VTX hardware. Essentially, using `CUSTOM` tries to control the VTX by telling it 'use frequency a', whereas `FACTORY` tells it 'use the frequency in band b, slot c in your internal tables'. Some VTX models can be controlled by either of these, others (like yours apparently) work with only the first or second way.

rickgomez2003, 2019-10-21:
Thank you so much for the information. Please mark this as solved.

closed 2019-10-21 by mikeller

luispequeno, 2020-09-11:
So what exactly was the fix?  I have SmartAudio 2.1 and I have tried some of these fixes and I still have no OSD with Rush Tank.  I’m lost and just cant seem to figure it out!

haslinghuis, 2020-09-11:
See this comment: https://github.com/betaflight/betaflight/issues/9078#issuecomment-544305564