betaflight/betaflight #9058: Tmotor F7 with Tmotor F55A reads no current on 4.1

allopez27, 2019-10-19:
Tmotor F7 paired with Tmotor F55A does not read current. In the ports tab UART4 is assigned as ESC sensor input. This is the UART that the esc natively plugs into the FC. In ESC/Motor Features the ESC_SENSOR is turned on. In Power and Battery the Current Meter Source is set to ESC Sensor. 

```# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: TMTR   board_name: TMOTORF7   custom defaults: YES

# start the command batch
batch start

board_name TMOTORF7
manufacturer_id TMTR

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature ESC_SENSOR

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200

# aux
aux 0 0 0 900 1275 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 325
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set current_meter = ESC
set pid_process_denom = 1
set osd_vbat_pos = 2400
set osd_current_pos = 2080
set osd_mah_drawn_pos = 2048
set osd_avg_cell_voltage_pos = 2432
set osd_stat_battery = ON
set vtx_band = 1
set vtx_channel = 1
set vtx_power = 4
set vtx_freq = 5865
set vcd_video_system = NTSC
set gyro_rpm_notch_harmonics = 1

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195

rateprofile 0

# end the command batch
batch end
```

Tmotor F7 and F55A from pyrodrone, Rush Tank mini VTX, spektrum quad race serial, lumenier 2407 1700kV.

IvoFPV, 2019-10-19:
@allopez27 I'm pretty sure that 4in1 doesnt have current over telemetry and instead has a separate current pin so you should configure that as the current meter source. However, this is not a right place to discuss configuration issues.
>Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

etracer65, 2019-10-19:
As mentioned, that ESC doesn't provide current sensor data over ESC telemetry.

closed 2019-10-19 by etracer65