betaflight/betaflight #9063: Flyaway w/FURYF4OSD (FY4O) 4.0.5 on Mamba F405 mini MK2 Diatone 249+

SaltyCracka, 2019-10-19:
**Describe the bug**
I upgraded my firmware from 3.5.1 to 4.0.5 to try out some of the new features. It's a 2.5" so I applied the UAV tech PIDs and took it out for a test flight. Worked fine until a crash and failed motor. Replace motor and decided to reverse prop direction. Changed the config in Betaflight and BL_Heli & props successfully. Went to do a hover test and it went to full throttle with the slightest touch. Checked everything, same result. Reversed everything back, same result.

At this point I realized my buzzer was no longer working either. I re-flashed the hardware with full chip erase with 4.0.5, performed the basic configuration changes, same result. Upgraded to 4.1...same issues, same results. Flyaway & buzzer non-functional. Thought I might have a failed piece of hardware so I bought a new stack and dumped it's configuration. It's a 3.5.1 so I re-flashed my existing stack to 3.5.1 and applied the diff from the new stack and my buzzer works again and it doesn't flyaway anymore.

**To Reproduce**
Install 4.0.5 OR 4.1 on a Mamba F405 mini MK2 stack with the FURYF4OSD image...

**Expected behavior**
I expected my drone to NOT go into a full throttle flyaway state and my buzzer to continue to work.

**Flight controller configuration**
'''# diff all

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.5 Aug  1 2019 / 11:57:21 (839661b20) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id 
mcu_id 002c00363436510237363537
signature 

# name
name diatone_249+

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature DISPLAY

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST

# map

# serial
serial 2 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 13 2 1800 2100 0 0
aux 3 35 1 1800 2100 0 0

# adjrange

# rxrange
rxrange 0 1000 1986
rxrange 1 988 2010
rxrange 2 990 2008
rxrange 3 988 2010

# vtx

# rxfail
rxfail 3 h

# display_name

# master
set dyn_lpf_gyro_max_hz = 700
set acc_calibration = -128,64,-332
set min_check = 1010
set max_check = 1990
set rssi_channel = 16
set fpv_mix_degrees = 20
set blackbox_p_ratio = 256
set motor_pwm_protocol = DSHOT600
set failsafe_throttle_low_delay = 400
###ERROR: CORRUPTED CONFIG: failsafe_throttle_low_delay = 400###
set bat_capacity = 550
set current_meter = ADC
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_crash_flip = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_tim2 = 1
set osd_vbat_pos = 14785
set osd_tim_2_pos = 14806
set osd_crosshairs_pos = 366
set osd_ah_pos = 175
set osd_motor_diag_pos = 472
set osd_craft_name_pos = 14794
set osd_warnings_pos = 14378
set osd_avg_cell_voltage_pos = 14753
set osd_flip_arrow_pos = 14768
set osd_stick_overlay_left_pos = 14660
set osd_stick_overlay_right_pos = 14675
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_power = 3

# profile
profile 0

set dterm_lowpass2_hz = 150
set vbat_pid_gain = ON
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10
set pidsum_limit = 1000
set p_pitch = 33
set i_pitch = 85
set d_pitch = 35
set p_roll = 28
set i_roll = 78
set d_roll = 32
set d_min_roll = 16
set d_min_pitch = 18
set d_min_boost_gain = 30
set d_min_advance = 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150
set roll_srate = 60
set pitch_srate = 60
set yaw_srate = 60
set tpa_rate = 80
set tpa_breakpoint = 1800

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# '''

I am currently at 3.5.1 and do not have a resource dump available from the 4.0.5 or 4.1 flashes/configurations. It should be the defaults that are available via the FURYF4OSD.

**Setup / Versions**
 - FC: MAMBA F405 mini MK2
 - 4in1 ESC: MAMBA F25 MK2
 - VTX: RUNCAM TX200U
 - RX: AC900 X-Boss Reciever
bought from DIATONE as a GT 249+ via Grayson Hobby;

**Additional context**
Add any other context about the problem here.

IvoFPV, 2019-10-19:
> Worked fine until a crash and failed motor. Replace motor and decided to reverse prop direction. Changed the config in Betaflight and BL_Heli & props successfully. 

That seems self explanatory, if I understand correctly that you flashed 4.0.5 and it worked before you changed prop direction and the motor.

>bought a new stack and **dumped it's configuration**. 
set failsafe_throttle_low_delay = 400
###ERROR: CORRUPTED CONFIG: failsafe_throttle_low_delay = 400###

You should not be dumping the config from older versions i.e 3.5.1->4.0.5 or 4.1.

I would recommend you start from scratch, flash the latest 4.1 with full chip erase enabled, and see if you still have issues. Please confirm your motor order and motor direction, as well as board/gyro orientation.

SaltyCracka, 2019-10-21:
Every time I flash firmware I do a full chip erase. I only selectively apply saved information from a diff all.
I do believe it worked before I reversed props...but I think I did have a runaway as part of the crash/motor failure. I had already done a full erase re-flash of 4.0.5 and 4.1 and still had the buzzer not working. Full erase flash of 3.5.1 restored buzzer and stopped flyaway (flew it yesterday until my antenna broke).
I could save my current config and return to normal prop rotation. Then begin a version by version upgrade checking my buzzer and flyaway status to determine where and when it starts happening.

jflyper, 2019-10-21:
@SaltyCracka Are you saying that stock, freshly full-chip erased board doesn't have a working buzzer? No diff or your own custom config loading, just reflash FURYF4OSD legacy target or FURYF4OSD unified target then testing the buzzer. Can you actually execute the procedure and confirm non-working buzzer for us please? No flyaway test necessary at this point.

SaltyCracka, 2019-10-24:
I just finished a baseline test sequence for the flyaway situation until it occurred. I started with a 3.5.1 w/full chip erase and applied a baseline changelog which will be found below. No problems. I went straight to 4.0.0 but found an issue regarding flyaways  (#8043) in the release notes and went straight to 4.0.1. This tested fine and went to 4.0.2. Buzzer is still working (I will continue with buzzer testing to current to verify I am not hallucinating as requested) however it appears the flyaway situation is occurring.

Testing Notes.
Diatone GT 249+ baseline 3.5.1 w/reversed props works fine.
Standard Changelog:
UART3 to IRC Tramp
UART6 to Camera Control
Mixer Motor Direction is Reversed
ESC/Motor Features DSHOT600
Arming 180 degrees
Modes Arm Aux1 1800-2100
Modes Angle Aux2 900-1200
Modes BEEPER Aux3 1800-2100
Modes Flip After Crash Aux2 1800-2100

Upgraded to 4.0.0
Found a note that may apply to my problem regarding flyaways:
Target: FURYF4OSD 
Version: 4.0.0 
Binary: betaflight_4.0.0_FURYF4OSD.hex 
Date: 11-04-2019 02:58 
Release notes:
Important
A number of users reported issues with their craft taking off with throttle at idle when running on Betaflight 4.0.0's default settings. It was identified that the default settings did not have enough filtering for setups with a lot of vibrations. More filtering was added to the defaults in 4.0.1. It is recommended that you upgrade directly to 4.0.1, to eliminate the risk of unintended flyaways introduced by this problem (#8043).

Upgraded to 4.0.1
Applied Standard Changelog
Desktop Beeper test successful
Desktop Throttle test w/o props successful

Upgraded to 4.0.2
Applied Standard Changelog
Desktop Beeper test successful
Desktop Throttle test w/o props FAILURE
Slight throttle increase after arm and return to min motors respond.
High throttle increase w/high speed after arm and return to min MOTORS HESITATE THEN ACCELERATE W/O RESPONSE. DISARM required to stop motors.

SaltyCracka, 2019-10-25:
> @SaltyCracka Are you saying that stock, freshly full-chip erased board doesn't have a working buzzer? No diff or your own custom config loading, just reflash FURYF4OSD legacy target or FURYF4OSD [...]

In case you haven't seen the detail in the previous comment, I did update to 4.1 & 4.0.5 w/full erase and tested the beeper w/switch. It works...don't know why I was having problems before but was probably user error. Still having the flyaway problem.

jflyper, 2019-10-25:
Okay, then you have proven yourself that something you believe is not always true.

It suggest that you would start from scratch configuration after fresh flashing, adding one category of configuration at a time, to find out what is causing the flyaway issue. It wouldn't be so hard as you are having this issue at launch.

kllpw, 2019-10-26:
I too am having runaway with a FURYF4OSD with freshly flashed ESCs (JESC) and flashed flight controller, i have had no crashes this thing was flying very well on blheli_s.  I notice i am getting errors in the Motors tab when i am testing motor direction.

[betaflight config.txt](https://github.com/betaflight/betaflight/files/3775067/betaflight.config.txt)
[Jesc log.txt](https://github.com/betaflight/betaflight/files/3775068/Jesc.log.txt)

mikeller, 2019-10-27:
@SaltyCracka:

> ###ERROR: CORRUPTED CONFIG: failsafe_throttle_low_delay = 400###

This clearly shows that you tried to use a config from an older version of Betaflight, and **not** follow the instructions for configuring 4.1:

> Please **do not paste tuning configurations from previous versions of the firmware**. Some defaults have been changed, and some parameters are used in different ways, so previous tuning settings will not work well with Betaflight 4.1.

Please try again, and follow the instructions this time, and then report back if the problem still occurs.

illmillrig, 2019-11-10:
I think your `gyro_1_align_yaw` should be `1800` if it's not.  which would be CW180.  It doesn't show up in your diff, which makes me think it could be wrong.

closed 2019-12-18 by stale[bot]

pattam2010, 2020-05-20:
Hi, 
I know this is ticket has been closed. I just built a f405mini with the F25 20A  ESC. I had the exact same issue. The quad shoot to max throttle even it's tightly move the throttle.  That is the issue with the betaflight 4.x firmware. It wouldn't happen stay in the 3.5.x fimrware. Can someone tell me what is the solution for it.

illmillrig, 2020-05-22:
> Hi,
> I know this is ticket has been closed. I just built a f405mini with the F25 20A ESC. I had the exact same issue. The quad shoot to max throttle even it's tightly move the throttle. That is the [...]

post the dump

f0x52, 2020-06-04:
Im having the same issue, F405 mini mk2 and after update to latest betaflight I have runaway throttle

pattam2010, 2020-06-08:
okay, my solution is update the latest 4.x betaflight firmware and make a dump from the 3.x setting in CLI., I believe that is an issue with the PID, even I did manually change the PID as setting in 3.x. I still had the issue with max throttle out. I need to use the 4.x firmware because the of the UART ports available. So again, my solution is :  1. upgrade to the last 4.x firmware, and then dump the setting I have in 3.x in CLI.

pattam2010, 2020-06-08:
> > Hi,
> > I know this is ticket has been closed. I just built a f405mini with the F25 20A ESC. I had the exact same issue. The quad shoot to max throttle even it's tightly move the throttle. That is the issue with the betaflight 4.x firmware. It wouldn't happen stay in the 3.5.x fimrware. Can someone tell me what is the solution for it.
> 
> post the dump. I fly this mamba FC every day on 4.1

I will do it later, thank you for your reply.

Dukky805, 2020-07-17:
Hey Guys, i know this is dead but i am having a similar flyaway issues and would really appreciate any help. I have a new build with a MAMBA F405 MK2 F40 stack. On first test flight it just shot to the moon after i raised the throttle slightly. 

Reset to defaults. No change

I've tried BF 3.5.7 - 4.2 all on defaults and they all do it. 

Tried using Diatone's config dump for the board. No change. https://www.diatone.us/apps/help-center#!hc-config-file

Looked at a black box log and i Term seems to be really high when it does a full throttle flyaway (no clue if that has anything to do with it.)

Tried running both reversed and regular motor directions

Tried a few different BLHeli versions. no change

My Current Diff all (after adding diatone's config):

# version
# Betaflight / FURYF4OSD (FY4O) 4.1.0 Oct 16 2019 / 11:37:34 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
mcu_id 004c004d3252500320383856
signature 

# map
map TAER1234

# serial
serial 2 8192 115200 57600 0 115200

# aux
aux 0 0 2 1700 2100 0 0
aux 1 1 1 1300 1700 0 0

# master
set gyro_sync_denom = 2
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_src_frame_errors = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set current_meter = ADC
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set gyro_1_sensor_align = DEFAULT
set gyro_2_spibus = 1

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

Dukky805, 2020-07-17:
Also, the flyaway only happens if i go above about 35% throttle. It is a 5inch build.

illmillrig, 2020-07-17:
why `set align_board_yaw = 180`?

illmillrig, 2020-07-17:
> 
> 
> Also, the flyaway only happens if i go above about 35% throttle. It is a 5inch build.

why `set align_board_yaw = 180`?

Dukky805, 2020-07-17:
The boards default settings have the gyro at CW180. I was just testing if doing it with board alignment instead of gyro made any difference. It made no change.

I did make some progress. I disabled air mode but the motors still spin when armed, but it doesn’t try to fly away now. I’m going to go out and try flying it with air mode off (it feels like it is on still tho, the Pid loop still has authority and it holds its angle when I drop to 0 throttle. 

I also tried turning my i term down to 0 the fly aways where gone. So to me it seems to be an issue with i Term in air mode. Check the BB log and that confirms that the i term is ramping up after going above about half throttle 

Maybe a bug with airmode on this board or is there something I’m missing??

illmillrig, 2020-07-17:
> The boards default settings have the gyro at CW180. I was just testing if doing it with board alignment instead of gyro made any difference. It made no change. I did make some progress. I disabled [...]

I think you may be confusing two different settings.  There is the Gyro (sensor) alignment, but then there is also the board alignment.  Correct, the sensor alignment is CW180 - which shows up as `set gyro_1_align_yaw = 1800` in a diff.  But you are also telling betaflight that the flight controller is actually turned around on the frame `set align board_yaw = 180` - notice the subtle difference .  if your flight controller is not actually turned around on the frame, then you need to set 'Yaw Degrees' back to 0 and leave the `First Gyro` alignment set to CW180

Dukky805, 2020-07-17:
Okay I will double check this when I’m home. The alignment seemed fine tho. Everything matched up when moving the quad around and looking at the BF gyro screen, also I’ve reset back to default since posting that and the issue still occurs. See my notes on airmode in my last reply. I think it has something to do with that. 

Thank you!

illmillrig, 2020-07-17:
> Okay I will double check this when I’m home. The alignment seemed fine tho. Everything matched up when moving the quad around and looking at the BF gyro screen, also I’ve reset back to default since [...]

I would just flash to bf4.2, use the defaults that betaflights asks to apply, and start from there.  only use the diatone config for reference when you need to look at it.  don't paste it in.

Dukky805, 2020-07-17:
That is my current config, 4.2 on defaults. The motors still jump to full throttle after going past half and have to disarm to get it to stop. Turned off air mode and the quad flys fine and is acting like air mode is on. Very confusing.

Stormdad, 2020-07-24:
> That is my current config, 4.2 on defaults. The motors still jump to full throttle after going past half and have to disarm to get it to stop. Turned off air mode and the quad flys fine and is acting like air mode is on. Very confusing.
> […](#)

I am having the same exact issue. I flashed back to the stock 3.5.7 but still runaway throttle after 1/2.

DawnFPV, 2020-11-11:
Hi , im using Mamba F405FC with crossfire Nano RX wired to Uart 6 Rx 6 and TX 6 . no stick inputs . ive tried everything . pls help

jpwhitney, 2022-02-20:
The ramping up of the I term is still an issue with this board on 4.3