betaflight/betaflight #9068: NO Telemetry on Fport with R9mm receiver

enrimoizo, 2019-10-20:
I followed Oscarlang Guide (https://oscarliang.com/setup-r9m-r9-mini-betaflight/) and everything looks good (binding, betaflight sticks \ aux working) but no telemetry in the openTX except for RSSI. I updated everyfirmware to last version.

Firmware:
Taranis: OpenTX 2.3.1
R9m & R9mm: Last firmware Flex / Flex Fport

Trasmitter Wiring: Ground ----> Ground
5v ---> 5v
Inverted S.Port ---> plugged to S-Bus Receiver you can see it in the picture. (UART6 RX)

This is the DUMP from CLI

Thank you very much for your attention.

get serialrx_provider
serialrx_provider = FPORT
Allowed values: SPEK1024, SPEK2048, SBUS, SUMD, SUMH, XB-B, XB-B-RJ01, IBUS, JETIEXBUS, CRSF, SRXL, CUSTOM, FPORT
Default value: SPEK1024

dump
version
Betaflight / OMNIBUSF4SD (OBSD) 4.0.6 Sep 1 2019 / 00:57:23 (2a64051a2) MSP API: 1.41
start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id

name
name Crocodile 7

resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
[... 952 lines cut ...]
set launch_control_gain = 40

rateprofile
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

jflyper, 2019-10-20:
The pad designated as SBUS is receive only. You have to find and use A TX port and set serialrx_halfduplex = on. For details, please visit the places mentioned in the pre formatted text when you opened this issue.

closed 2019-10-20 by jflyper

enrimoizo, 2019-10-20:
Ok thank you very much for your help.