betaflight/betaflight #9069: [BUG] OSD "VTX channel" always showing default power level

Lord-Quake, 2019-10-20:
Installed the latest official Betaflight 4.1 software on my HGLRC XJB-145 using the IRC Tramp protocol. 
I've set up 4 power levels (25,100,200,350) and have 100mv set as default.
"Low Power Disarm" is also activated (ON).
OSD's "VTX channel" is active as well.

**Issue**
OSD is showing 100mv even though the QC is disarmed.

**Expected behavior**
Shouldn't it show the lowest power output level (25mv) until I arm the quad?

IvoFPV, 2019-10-20:
@Lord-Quake If you are using the Tramp protocol you should get the appropriate VTX table. This one seems to have power levels set up for Smartaudio.

Lord-Quake, 2019-10-20:
According to Betaflight Wiki example:

vtx_power = #
Allowed range: 0 - 5
for SmartAudio:  0=25mW, 1=25mW, 2=200mW, 3=500mW, 4=800mW
for TBS Unify Nano:  0=25mW, 1=25mW, 2=50mW
for IRC-Tramp:  0=25mW, 1=25mW, 2=100mW, 3=200mW, 4=400mW, 5=600mW

IvoFPV, 2019-10-20:
IRC Tramp power levels are done like this. Could you try this out?

vtxtable powervalues 25 100 200 350
vtxtable powerlabels 25 100 200 350

https://github.com/betaflight/betaflight/blob/master/docs/VTX.md#power-levels

Lord-Quake, 2019-10-20:
Bingo! OSD is now showing the correct power levels. Thank you! 
Here the link where documentation seems to be incorrect or just outdated.
https://github.com/betaflight/betaflight/wiki/VTX-CLI-Settings

closed 2019-10-20 by Lord-Quake

IvoFPV, 2019-10-20:
@Lord-Quake Great! BTW Added a note to the wiki. Hopefully it is now clear that those are for BF versions prior to 4.1.