betaflight/betaflight #9078: SmartAudio 2.1 issue with 4.1

rickgomez2003, 2019-10-20:
The issue i found is with SmartAudio 2.1 and the TBS Unify-Pro 32 Nano, after upgrading from RC6 to 4.1 I can no longer operate the VTX  IE (power, Band, Channel). I have tested on four different rigs #1 Iflight tiwng F7, #2 Matek F722SE, #3 SP Racing H7 Extreme, #4 Iflight F411 Pro. All caused the same issue.

I am looking for a solution to this issue to allow me to adjust the setting, as before either from the OSD or theough the LUA script.

Describe alternatives you've considered
I have not tested any other way other that button presses.

Additional context
This appeared to be working in RC6 but now that I have upgraded it does not work and even after the downgrade the issue persist.

Please let me know if I can provide any further information. Thank you for all that you do.

rickgomez2003, 2019-10-20:
Here is the Diff:  
# 

# diff all

# version
# Betaflight / MATEKF405 (MKF4) 4.1.0 Oct 16 2019 / 11:42:57 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
mcu_id 003a00583436510339373635
signature 

# name: Matek-F405

# feature
feature -SOFTSERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RTEA1234

# serial
serial 0 2048 115200 57600 0 115200

# aux
aux 0 0 0 1900 2100 0 0
aux 1 13 2 1900 2100 0 0
aux 2 26 0 1900 2100 0 0
aux 3 30 1 1400 1600 0 0
aux 4 35 1 1900 2100 0 0
aux 5 36 3 1900 2100 0 0
aux 6 39 2 1400 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# rxfail
rxfail 4 s 1000
rxfail 6 s 2000
rxfail 7 s 1000
rxfail 11 s 1000

# master
set gyro_sync_denom = 2
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set fpv_mix_degrees = 35
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1550
set osd_alt_alarm = 400
set osd_vbat_pos = 14360
set osd_rssi_pos = 14336
set osd_tim_1_pos = 14776
set osd_tim_2_pos = 14808
set osd_flymode_pos = 14400
set osd_vtx_channel_pos = 14368
set osd_craft_name_pos = 14825
set osd_warnings_pos = 14346
set osd_avg_cell_voltage_pos = 14392
set osd_disarmed_pos = 14816
set osd_log_status_pos = 14840
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_fft = ON
set osd_stat_total_time = ON
set debug_mode = SMARTAUDIO
set vtx_band = 3
set vtx_channel = 7
set vtx_power = 1
set vtx_freq = 5925
set vcd_video_system = PAL
set name = Matek-F405

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_rc_rate = 130
set pitch_rc_rate = 130
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 75
set pitch_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#

rickgomez2003, 2019-10-20:
Currently testing on Matek F405-OSD trying all the boards that I can.

rickgomez2003, 2019-10-20:
[image]

IvoFPV, 2019-10-20:
Duplicate of #9057. No need to open multiple issues. 

>vtxtable powervalues **25 200 400**
vtxtable powerlabels 25 200 400

This is wrong. For smartaudio 2.1 your power values must be dBm representation of power. Example 14 dBm = 25mW

rickgomez2003, 2019-10-20:
Sorry uploaded the wrong diff. Sorry about the post just did not know if this should be separate.

rickgomez2003, 2019-10-20:
I have found the issue. If the bands are set to Custom instead of Factory the issues appears. If I set it to Factory then it works properly. 

Issue caused by this:
# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

Issue fixed by this:
# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK I FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 LOWBAND  L CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

yoodog, 2019-10-22:
I found an other issue. I Save US version of vtx_table_smart_audio_2_1_us.json and it saves as .txt
I than remove and save it as .json. 
Then I go to BF Configurator 10.6, connect BF4.1 F7v2 omnibus. Try to load from local and it quits program on MacOSx 10.15

Oko001, 2019-10-22:
I had the same problem as rickgomez2003. Smartaudio can switch every parameters except Power, Band and Channel. 
Kakute F7mini FC, Eachine TX1200 VTX. New build with RPM filter on, so I wouldn't like to switch back to 4.0.6. I did it for a try, and on 4.0.6. everything is ok.
 Debug is 216.
Some youtuber told that it is Smartaudio 2.1. IT's NOT!
Some youtuber set powerlevels to 1.0; 1.2 above 1000 mW. 
So I set this: 
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 1.0

The reason why not worked the Power, Band and Channel setting is wrong powerlabel!
Without point, doesn't matter that I write 10 or 999, it works!
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 999

I hope that this saves a few hours for others...

mikeller, 2019-10-23:
@yoodog: What browser are you using?

Oko001, 2019-10-23:
Chrome.
An add: my AKK Vtx with Matek F7 has no problem with powerlabels...

closed 2019-11-29 by stale[bot]