betaflight/betaflight #9079: Rateprofile adjustment not working in 4.1

matoy, 2019-10-20:
Hi,

I have a switch on RC to switch between rates profiles 1, 2 and 3.

I have this feature on 4 quads, different FCs, some quads are on SBUS, some other on CRSF. This even works in flight.

In configuration, that's done by;
adjrange 0 0 2 900 1100 12 2 0 0
adjrange 1 0 2 1400 1600 12 2 0 0
adjrange 2 0 2 1900 2100 12 2 0 0
(to AUX3 in my case)

Until BF 4.0.6, it's working on all my quads.

Since BF 4.1 (even RC1), switching beetween rates profiles is not working well anymore. I got the same behavior on all my 4 quads.

At FC boot, i can switch to rate profile 2 and then 3 with the switch but once this is done, i can NOT switch back to rate profile 1 or 2. This was clearly working until 4.0.6.

If anyone could help or investigate, that would be very nice.
I don't think that BF new 4.1 has configuration changes on this matter, if this is a mistake from myself, thank you for your indulgence :)

Regards,

IvoFPV, 2019-10-20:
Check out #8997

matoy, 2019-10-20:
thanks a lot; it's working.

closed 2019-10-20 by matoy

mikeller, 2019-10-21:
Duplicate of #8997.