betaflight/betaflight #9082: rpm filter resonances while ascending

Mode-1, 2019-10-21:
switching on the rpm-filter in 4.1.0 causes resonances (like chirping) when ascending.
Quad swings around 25Hz on different axes.
switching off the rpm-filter causes the same quad to fly smoothly.

this can be reproduced with 3 totally independent quads:
125mm quad with 1104/7500kv motors, Lumenier Micro Lux V2 AIO, Target LUXF4OSD on 3S
250mm quad with 1806/2280kv motors, Lumenier Micro Lux V2 AIO, Target LUXF4OSD on 3S
250mm quad with 2306/2400kv motors, Mamba Mini F7 AIO, Target MAMBAF722 on 4S & 5S
checked cycle times - ok
checked motors - ok
tried DSHOT 150 - 600 - no change in behavior
tried 3S to 4S/5S - no change in behavior
tuned pids - no real change in behavior
All 3 quads esc's are on BLheli32 V:32.7
tried  BLheli32 V:32.6 - no change:

[image]

here is the 125mm quad's diff and blackbox with and without rpm filter:
[diff and blackbox data.zip](https://github.com/betaflight/betaflight/files/3752427/diff.and.blackbox.data.zip)
i'm using the lastest configurator 10.6.0 - should be ok

with firmwares around #17xx i had manually installed rpm-filters on the 250mm quad
with 1806/2280kv's which did not show the resonance, so the frame,motors, esc and fc
could handle rpm filtering with no issue.

tried different debug_modes, but can't get down to the source of the problem.
switching off the rpm-filter is my only cure at the moment

IvoFPV, 2019-10-21:
@Mode-1 Too much filtering. Enabling RPM it adds a significant amount of phase delay on top of pretty conservative default lowpass and dynamic notch settings. That creates a positive feedback loop and you have these sine wave looking pid traces, oscillations you are experiencing.

To fix this, when you enable RPM you **must** make changes to at least dynamic notch filter and gyro lowpass filters. Read the wiki for recommended settings when running RPM filtering.

Mode-1, 2019-10-21:
Thanks IvoFPV,
i took filtering down like you recommended:
[image]
Oscilallations are down now:
[rpm on - filters down.zip](https://github.com/betaflight/betaflight/files/3752761/rpm.on.-.filters.down.zip)
...but at the price of hot motors in only a 30 second run.
The working window of rpm-filtering seems to be quite narrow for my quads,
so i'll wait for progress in BF before i smoke motors unnecessarily in otherwise
working quads.
cheers IvoFPV

closed 2019-10-21 by Mode-1