betaflight/betaflight #9083: Motor noise statistics

haybeseret, 2019-10-21:
Hello guys,

Since we now can utilize rpm filters per motor - we can also collect noise statistics per motor.

In a commercial airliners when a motor is out of balance (i.e generating too much noise / vibrations) there is an alert to the technical crew that "motor x is out of balance"
keeping the same line of thought I think it will be a good feature to add the ability to alert pilot 
on the summary screen the following data:

1.Which motor was the "noisiest"
2.suggest filter level as per the collected noise of the "noisiest" motor

Also maybe it can be possible to add a "test motor" (for noise) screen. 

I can be very good to see which one of your motors is about to die, has a bad bearing, etc..

Thanks for all your hard work, in any case.

Cheers.

etracer65, 2019-10-21:
We don't have independent "vibration" sensors on each motor like a commercial airliner might. RPM filtering just tells us the frequency of each motor, not the amplitude of the vibrations it's causing. Any "vibration" seen by the gyro is a combination of all the motors along with any other sources like frame resonance, etc.

haybeseret, 2019-10-22:
Thanks for the quick reply.
Can we tell how many notch filters are being applied to each motor by the rmp filters and thereby know which motor is causing more noise ?
If you cannot think of any other way we can determine single motor noise level, feel free to close my request

McGiverGim, 2019-10-22:
All the motors the same, by default three notches by motor and by axis. So 9 notches by motor. 36 notches total. If a fixed value.

etracer65, 2019-10-22:
@haybeseret You're confusing cause & effect. We only have the RPM of the motor so we "expect" that there will be vibration at that frequency (and integer harmonics). So we apply notch filters under this expectation that there will be noise at this frequency that needs to be reduced. We don't actually have any measure of the noise to test whether the notch really needs to be applied.

closed 2019-11-28 by stale[bot]