betaflight/betaflight #9084: VTX going to wrong frequencies on 4.1

jalalaba, 2019-10-22:
On 4.1 when i setup the VTX tables and choose a band and channel from the table the vtx goes to a somewhat nearby frequency( i can see glimpses of parts of the OSD in the static). When i enter the frequency manually it still doesn't work. That frequency has to be deleted( just 0 instead of the frequency) from the table to get the VTX there. 

``` 
# 

# diff

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4FW   custom defaults: YES

# start the command batch
batch start

board_name OMNIBUSF4FW
manufacturer_id AIRB

# resources
resource MOTOR 1 A03
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# feature
feature SOFTSERIAL
feature ESC_SENSOR

# beacon
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 0 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 27 3 1700 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 3 1300 1700 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 FATSHARK F FACTORY    0 5760 5780 5800 5820 5840 5860    0
vtxtable band 4 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable band 5 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 1
vtxtable powerlabels 25

# master
set gyro_lowpass2_hz = 325
set gyro_to_use = SECOND
set dyn_notch_range = LOW
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -1,178,252
set mag_hardware = NONE
set serialrx_provider = SBUS
set min_throttle = 1070
set dshot_idle_value = 500
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set motor_poles = 12
set align_board_roll = 180
set align_board_yaw = 270
set current_meter = ESC
set battery_meter = ESC
set yaw_motors_reversed = ON
set deadband = 3
set yaw_deadband = 5
set pid_process_denom = 1
set osd_rssi_alarm = 30
set osd_cap_alarm = 600
set osd_vbat_pos = 6539
set osd_throttle_pos = 4259
set osd_current_pos = 4501
set osd_mah_drawn_pos = 4483
set osd_power_pos = 4278
set vtx_power = 1
set vtx_freq = 5740
set gyro_1_align_yaw = 1800
set gyro_2_sensor_align = CW180FLIP
set gyro_2_align_pitch = 1800
set gyro_2_align_yaw = 1800

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195

rateprofile 0

# rateprofile 0
set thr_mid = 30
set thr_expo = 66
set roll_rc_rate = 101
set pitch_rc_rate = 101
set yaw_rc_rate = 101
set roll_expo = 30
set pitch_expo = 30
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
set throttle_limit_type = SCALE
set throttle_limit_percent = 90

# end the command batch
batch end

# 
``` 

 - Omnibus Nano V6;
 - TBS Unify Pro V2(SmartAudio 2.0);

IvoFPV, 2019-10-22:
@jalalaba I would assume this only happens on the IMD6 band where there are custom frequencies? Maybe the VTX is not capable of transmitting on specific frequencies that are outside of its own build in band/channel table. Most likely it is just jumping to a closest one that's why you get that behavior.

jalalaba, 2019-10-22:
> @jalalaba I would assume this only happens on the IMD6 band where there are custom frequencies? Maybe the VTX is not capable of transmitting on specific frequencies that are outside of its own build [...]

Happens on all channels. That is why Fatshark band 1st channel is 0 because i use that usually.

etracer65, 2019-10-22:
Have you followed the instructions from the manufacture to unlock the vtx so it can access all channels?

https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf

jalalaba, 2019-10-22:
> Have you followed the instructions from the manufacture to unlock the vtx so it can access all channels?
> 
> https://www.team-blacksheep.com/tbs-unify-pro-5g8-manual.pdf

Yes. Yes i have. My Smart Audio Debug[0] key is 216 (SA 2.0 unlocked)

nodueck, 2019-10-27:
I've got same issues:
- Infinite DVR VTX
- HGLRC Titan Plus F460
```
# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4   custom defaults: YES
```

I've set up Band F Channel 3 and receiving a signal von Band R Channel 1 and Band E Channel 3.
Then I did some tests (all on channel 3):
Set up R Band -> Receiving a signal on F Band
Set up F Band -> Receiving a signal on E Band
Set up A Band -> Receiving a signal on A Band

I was just watching into my googles, didn't try to measure any frequency.
```

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 4 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable band 5 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 1K
```

functionpointer, 2019-10-28:
@nodueck take a look at the frequency table of your vtx:
https://www.akktek.com/infinite-dvr-vtx.html
The bands you have set-up as `FACTORY` do not match the built-in frequency table of your vtx.
Change them to match, e.g. like this:

They need to! In `FACTORY` mode, Betaflight tells the vtx to go into, for example, band 2 and channel 3. The vtx then uses its own frequency table to translate that into a frequency.
Using `FACTORY` only makes sense when `vtxtable` is set up to match the vtxes built-in frequency table exactly (except for blocked channels using 0).

Therefore, your `vtxtable` should be set up something like this:

If you don't want to match the built-in frequency table, use `CUSTOM` mode, like you did for the IMD6 band.

functionpointer, 2019-10-28:
@jalalaba Your `vtxtable` has the same problem. Check your frequency table: https://www.team-blacksheep.com/tbs-unify-pro-manual.pdf
You are missing the E band.

mikeller, 2019-10-29:
@functionpointer: Well spotted! This is probably one of the trickier aspects of how the `FACTORY` bands implementation is working.

@jalalaba, @nodueck: To get the frequencies to be available as you have in your original VTX table, you'll have to do something like this:

```
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 UNUSED   U FACTORY    0    0    0    0    0    0    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860    0
vtxtable band 5 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
```

nodueck, 2019-10-29:
Thank you.
I downloaded the european version of the vtx table (Smart Audio 2.0) from [here](https://github.com/betaflight/betaflight/wiki/VTX-tables) and imported them into betaflight. I think Band E didn't show up, because it's not allowed in Europe. However, I added them as shown by @mikeller .

mikeller, 2019-10-30:
@nodueck: Turns out the SmartAudio tables for the EU on the [VTX tables wiki page](https://github.com/betaflight/betaflight/wiki/VTX-Tables) had this bug. I fixed them and re-uploaded them.

closed 2019-12-06 by stale[bot]