betaflight/betaflight #9085: baroproblem, altitude wrong (>10000m) after some times using rescue mode

Ralfde, 2019-10-22:
**Describe the bug**
at testing the GPS rescue mode a few times, suddenly the altitude was shown over 10000m too high. So the motor throttle was reduced and the copter did sink instead of rise.
A few times the rescue mode worked, before the altitude was shown wrong.
It happend about 2min20 to 2min30s, when switching into the rescue mode.

**To Reproduce**
Steps to reproduce the behavior: use the rescue mode some times. See log file.

**Flight controller configuration**
OmnibusF7NXT with dual gyro
logfiles:
[20191022.zip](https://github.com/betaflight/betaflight/files/3757784/20191022.zip)

Please see also the Video: 
https://drive.google.com/open?id=1MCyhAqBDb41M7YNWwzM9XN55YDvjD8-r

diff all:
[BTFL_cli_20191018_143402_Alpha220_BF4_1.zip](https://github.com/betaflight/betaflight/files/3757850/BTFL_cli_20191018_143402_Alpha220_BF4_1.zip)

etracer65, 2019-10-23:
The altitude is derived from the barometer (if available), GPS (if available), or a combination of both. So I would suspect that one of your sensors had a malfunction - most likely the GPS.

You could experiment with selecting only one altitude source to see if you can isolate the problem. `DEFAULT` means use both if available, otherwise use only the one detected. Set to the other values to force using the specific source.
```
# get position_alt_source
position_alt_source = DEFAULT
Allowed values: DEFAULT, BARO_ONLY, GPS_ONLY
```

Ralfde, 2019-10-23:
Thanks, I will try.

etracer65, 2019-10-23:
You could also capture a log with `debug_mode = ALTITUDE`. Then:
* debug(0) = the ratio of GPS "trust" used for altitude
* debug(1) = altitude from baro
* debug(2) = altitude from GPS

Ralfde, 2019-10-23:
Thanks. I will do a log. Hopefully the error occurs again. At the testflight today it did not occur.

Ralfde, 2019-10-26:
With baro only I did not have a problem today. I think GPS and Baro combined is no good combination, when flips and rolls are done, as this can cause GPS glitches. I made a logfile today, but I'm not sure, if it does show the effect. Maybe not. Maybe somebody can check the GPS code, what could happen at a GPS glitch. I also had a problem with GPS only in the whoop with GPS rescue, but I can not make a logfile there. It has no logging function and no baro. The effect at flying looked similar.
[btfl_001.zip](https://github.com/betaflight/betaflight/files/3774816/btfl_001.zip)

etracer65, 2019-10-27:
I don't see any issues in your log. The values from both the barometer and GPS look reasonable.

If you don't want to use GPS for altitude you're free to change the `position_alt_source` setting and limit it to only the barometer.

Ralfde, 2019-10-27:
Yes, I believe this GPS glitches happen only, if the numbers of satellites get too low at acrobatic flying.
Anyway I will use barometer only for this copter. 
Is it possible to simulate a GPS glitch to check what the software would do with this?

I also want to say, that the BF4.1 firmware is very great. Thanks a lot for such a great fc software.

etracer65, 2019-10-27:
Define "glitch". Like if the GPS returns some incorrect value for the altitude? Then that "incorrect" value will be used for the altitude calculation, simple as that. Same as if any sensor returned erroneous data. Since we only have one of each we have to trust the sensor readings.

mikeller, 2019-10-27:
@Ralfde: Are you by any chance disarming during flight? There is a change in 4.1 that when disarmed with a GPS enabled, absolute altitude is reported, and not relative altitude.

Ralfde, 2019-10-27:
A GPS glitch will return very wrong data for the position and also for the altitude.
But why was the wrong altitude shown so long? Normally it should be corrected again as soon as the correct datas are read again or maybe a few seconds later.
Maybe the datas of the GPS should not be used, in case the number of satellites are under the number, which is adjusted. Is it implemented like this? Maybe it will need also a filter to make the effect of such a glitch lower.
@mikeller: no, I never disarm during the flight. At the DVR video I landed and disarmed and after arming again the altitude was correct again.

etracer65, 2019-10-27:
> A GPS glitch will return very wrong data for the position and also for the altitude.

How would you possibly know that? Are you aware of what bugs your GPS module might have or how it behaves for example if it has some kind of hardware failure?

Even so, without logging we can't even say the GPS was responsible for your "glitch". Maybe the barometer malfunctioned. You have the baro sensor properly shielded from airflow and light, right? The baro sensor is very sensitive to both airflow (like from the props) and bight light (like sunlight) and these can radically affect the readings.

Ralfde, 2019-10-28:
I know, it's hard to say anything without having a logfile, only the OSD video.
The really strange thing is, that the altitude was counting upwards exactly from the moment of switching into the rescue mode. At this time, there should be no wrong altitude data from any sensor.
Was there anything changed at switching into the rescue mode from BF4.0 to 4.1? I never had a problem with BF4.0.
The reason why I think it's coming from the GPS or rescue software is, that I had the same problem with the BetaFPV85X with build in GPS. This one does have no Barometer, so the problem can not be caused by the barometer, if there is no barometer available. The bad thing is, this BetaFPV85X does have no logging function build in and I didn't make a DVR of this flight with the problem. I'm afraid, we need to wait until somebody or I can bring a logfile.

closed 2019-12-04 by stale[bot]