betaflight/betaflight #9086: Hexacopter Current Sensor Question

rgluckman, 2019-10-22:
Hello. I’m looking to build a Hexacopter using 32Bit Individual ESC’s. I’m trying to figure out how to get Betaflight to read current accurately so my OSD shows me accurate current mah used. 
Can I run 6 32Bit ESC telemetry wires to a single UART?
Will Betaflight know what to do with that information?

**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

IvoFPV, 2019-10-22:
Set the ESC telemetry in the Ports tab on the correct UART, enable ESC_SENSOR on the Configuration tab, in Power & Battery for current meter source select ESC Sensor.
However, you should not ask for configuration help here.
>Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.
Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

rgluckman, 2019-10-22:
Sorry about that. I’m not on GitHub too often. Where would I ask a config question like that?
So just to confirm. Betaflight will read 6 motor outputs no problem?

IvoFPV, 2019-10-22:
You have a link in my previous post to Betaflight's Slack where you can ask anything and will get an answer much quicker. It should work.

etracer65, 2019-10-23:
ESC_SENSOR telemetry can handle up to 8 motors.

closed 2019-11-29 by stale[bot]