betaflight/betaflight #9091: Ibus telemetry setup not working

lpb06, 2019-10-23:
Ibus telemetry setup not working. During setup if I set serial RX to any uart and try to setup setup Ibus telemetry on the same uart, i hit save reboot, and the ports go back to default, not remembering any of the choices. Even tried to set this from CLI, but does the same

flight controller: diatone mamba f722  , betaflight 4.1

lpb06, 2019-10-23:
**Describe the bug**
Ibus telemetry setup not working.
**To Reproduce**
  During setup if I set serial RX to any uart and try to setup setup Ibus telemetry on the same uart, i hit save reboot, and the ports go back to default, not remembering any of the choices. Even tried to set this from CLI, but does the same:

# serial 1 4160 115200 57600 115200 115200
serial 1 4160 115200 57600 115200 115200

save

# diff

# version
# Betaflight / MAMBAF722 (MBF7) 4.1.0 Oct 16 2019 / 11:42:41 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name MAMBAF722

# serial
serial 3 0 115200 57600 0 115200

profile 0

rateprofile 0

# end the command batch
batch end

**Expected behavior**
saving uart after i hit save, detailed:
https://github.com/betaflight/betaflight/wiki/Single-wire-FlySky-(IBus)-telemetry

**Flight controller configuration**
'''
# diff

# version
# Betaflight / MAMBAF722 (MBF7) 4.1.0 Oct 16 2019 / 11:42:41 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name MAMBAF722

# serial
serial 3 0 115200 57600 0 115200

profile 0

rateprofile 0

# end the command batch
batch end
'''

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

'''
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: FREE
C13: FREE
C14: LED 2
C15: LED 1
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
'''
**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
Diatone mamba f722 from banggood
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
nothing you cannot setup the flight controller on the bench 

- how are the different components wired up.
nothing
**Additional context**
Add any other context about the problem here.

lpb06, 2019-10-24:
just tested this, it has worked with 3.5 but 4.0 has the same problem, both on legacy and normal targets.

jflyper, 2019-10-25:
@lpb06 sharing IBUS RX and telemetry must be accompanied by RX set to IBUS which allow sharing of RX and telemetry ports. Otherwise, the sharing will be reset.

lpb06, 2019-10-25:
THX, its working now, I'm an idiot:)

closed 2019-10-25 by lpb06