betaflight/betaflight #9095: VTX power disabled after last update

enrimoizo, 2019-10-24:
As per object, i had to downgrade to previous Betaflight firmware becouse with the last update i 'm not able anymore to set VTX power using  Irc Tramp or Betaflight GUI.

robherb, 2019-10-24:
I too am having this issue. Unfortunately I am not equipped to do any hardware diagnostics.

I noticed immediately that once I moved from BF 4.0.6 to 4.1.0, my vtx started acting up and dropping out. It appears that it is stuck on a high output power level and is overheating and then reseting itself midflight. The status leds for the vtx are junk and I have been unable to use the button to change the power level manually. Also, low power disarm does not appear to work anymore (no change in signal when disarmed). I'll try and find the time to go back to BF 4.0.6 and test if the issue goes away.

Beta75 Pro 2 with an Eachine Nano VTX using IRC Tramp

robherb, 2019-10-25:
I'll do some more in depth testing this weekend, but this evening I flashed 4.0.6 and can clearly see the video signal change in response to arming and disarming with low power disarm enabled.

enrimoizo, 2019-10-25:
rotfl i provvided them

enrimoizo, 2019-10-25:
Can you please explain what's the missing information you need ? Becouse  the problem is very clear to me and i know at least 3 people who had to downgrade. :)

IvoFPV, 2019-10-25:
>rotfl i provvided them

@enrimoizo How can you expect help when you are not willing to read the instructions when opening an issue? Read them for real this time, provide the information and someone might help you out, unless you are just having a configuration issue (most likely) which has instructions as well 🤔