betaflight/betaflight #9097: Smartport via SS was working in 4, broken in 4.1. CLRacing F4 Mini

SlipstreamFPV, 2019-10-24:
1. I was running BF4.0.0 with smartport on softseria. I updated to 4.1.0 and applied the same configuration steps and it no longer works.

2. ```
feature TELEMETRY

set tlm_halfduplex = on
set tlm_inverted = off

resource SERIAL_TX 6 NONE
resource SERIAL_RX 11 NONE
resource SERIAL_TX 11 C07

serial 30 32 115200 57600 0 115200

save
```
3. Using a CLRacing F4 Mini. Frsky R9 Mini smartport telemetrey wire is connected to "SAT" pin on CLRacing F4 Mini (C07).
4. Betaflight 4.1.0, tried both new and legacy firmware versions of 4.1
5. ```# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.1.0 Oct 16 2019 / 11:28:41 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# name: +61422411774

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C07
resource SERIAL_RX 4 NONE
resource SERIAL_RX 6 NONE

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beacon
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 35 1 1400 1600 0 0

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 350
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set acc_hardware = NONE
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set bat_capacity = 520
set battery_meter = ESC
set ibata_scale = 170
set deadband = 5
set yaw_deadband = 5
set frsky_unit = METRIC
set vcd_video_system = PAL

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210

rateprofile 0

# rateprofile 0
set roll_srate = 79
set pitch_srate = 79

# end the command batch
batch end```

functionpointer, 2019-10-24:
What is your vtxtable set up? Post the output of the cli command `vtxtable`.

jflyper, 2019-10-24:
I've commented in the slack about this.

> I was running BF4.0.0 with smartport on softserial.

CLRACINGF4 never had a valid timer channel configured for PC7 as default, so a softserial using this pin could not have worked.

If it actually was working, there should have been an additional timer channel configuration for the pin.

If you are not certain about what I'm talking about, I think you have to go back to 4.0.x, make sure the software serial over PC7 works, take “diff all” and paste the result here as an additional information.

SlipstreamFPV, 2019-10-25:
Okay, I reflashed 4.0.0 with my old settings and telmetery definitely works. Here is the diff:

`# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr 11 2019 / 07:01:44 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# resources
resource SERIAL_TX 1 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C07
resource SERIAL_RX 4 NONE
resource SERIAL_RX 6 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 35 1 1400 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set dyn_lpf_gyro_max_hz = 840
set acc_hardware = NONE
set acc_calibration = 26,26,-91
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT600
set bat_capacity = 520
set battery_meter = ESC
set ibata_scale = 170
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set frsky_unit = METRIC
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 35
set osd_cap_alarm = 420
set osd_tim1 = 769
set osd_vbat_pos = 2497
set osd_rssi_pos = 2510
set osd_tim_1_pos = 2519
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 200
set osd_ah_sbar_pos = 200
set osd_ah_pos = 200
set osd_current_pos = 2503
set osd_mah_drawn_pos = 2514
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 441
set osd_gps_lon_pos = 33
set osd_gps_lat_pos = 49
set osd_gps_sats_pos = 417
set osd_home_dir_pos = 65
set osd_home_dist_pos = 66
set osd_compass_bar_pos = 266
set osd_altitude_pos = 86
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 33
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 391
set osd_disarmed_pos = 2154
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_max_fft = ON
set debug_mode = D_MIN
set vtx_band = 3
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5645
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_min_hz = 60
set dyn_lpf_dterm_max_hz = 175
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set p_pitch = 32
set i_pitch = 60
set d_pitch = 24
set f_pitch = 0
set p_roll = 28
set i_roll = 50
set d_roll = 20
set f_roll = 0
set d_min_roll = 0
set d_min_pitch = 0

# rateprofile
rateprofile 0

set roll_srate = 79
set pitch_srate = 79
set tpa_rate = 75
set tpa_breakpoint = 1600

# end the command batch
batch end`

So maybe there's accidently a timer assigned to pin C07 in the BF4.0.0 config? I emember I had to try a few pins before one would work. Is there a way to find out what timer is working on C07 in BF4.0.0 and then apply that in 4.1?

SlipstreamFPV, 2019-10-25:
@jflyper I forgot we've been through this before. Apologies. It was a bug that made it work. The bug has been fixed so it doesn't work anymore.

I have done the following:

```
# timer
timer B09 AF3
# pin B09: TIM11 CH1 (AF3)
timer B00 AF1
# pin B00: TIM1 CH2N (AF1)
timer B01 AF3
# pin B01: TIM8 CH3N (AF3)
timer A03 AF1
# pin A03: TIM2 CH4 (AF1)
timer A02 AF1
# pin A02: TIM2 CH3 (AF1)
timer B04 AF2
# pin B04: TIM3 CH1 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer C07 AF2
# pin C07: TIM3 CH2 (AF2)
```

Notice the last two lines. It works in 4.1 now. Is that a valid method for making it work? Is there a reason more timer channels aren't assigned to pins? Does it use more CPU or something?

jflyper, 2019-10-26:
@SlipstreamFPV 
>Notice the last two lines. It works in 4.1 now. Is that a valid method for making it work?

Yes, the `timer C07 AF2` was indeed the line I was looking for ;)

>Is there a reason more timer channels aren't assigned to pins? Does it use more CPU or something? In this case I think TIM3 is only assigned to the "beeper" on B04 which is unused.

There are many reasons that pins are not assigned timer channels. In your case, I imagine that the designer/maintainer of the board did not want timer collisions (we call a case multiple different functions using a same timer this way; it breaks both functions in variety of ways) when a user decided to use passive beepers. Other reasons include lack of imagination, of course 😂 

If you are satisfied, you can close this issue.

SlipstreamFPV, 2019-10-26:
Satisfied that it's working. I can't find documentation on the timer labeling which would be nice to know, any chance you can fill in the blanks?

TIM# = timer number
CH# = ? 
CH#N = ?
(AF#) = ?

jflyper, 2019-10-26:
@SlipstreamFPV Those are details hidden deep inside STMicro's forest of STM MCU architecture.

TIM# = Names of timer units/functions. There are multiple timers in the MCU.

SlipstreamFPV, 2019-10-26:
@jflyper You’re awesome. Thanks a lot for that! Keep up the great work.

closed 2019-10-26 by SlipstreamFPV