betaflight/betaflight #9098: Role of death with Bidirectional DSHOT on H7EXTREME

maGfpv, 2019-10-24:
**Describe the bug**
Role of death with activated Bidirectional DSHOT and RPM Filter on SPRACINGH7EXTREME

**To Reproduce**
Bidirectional DSHOT was fixed with this issue betaflight/betaflight#8836. 
I updated to Build betaflight/betaflight#1900 (28.09.2019 22:50:24) and activated rpm filter. Everything else is set to default. Sometimes it happens that the quad performs a role of death. (Log is attached [LOG00008.zip](https://github.com/betaflight/betaflight/files/3726415/LOG00008.zip))

Without activated bidir and rpm filter it works great with no issues.

**Flight controller configuration**
```# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 28 2019 / 23:25:54 (fcaa16cd7) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# name: Marmotte

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A00

# feature
feature -TRANSPONDER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 11,11::C:2
led 1 12,12::C:2
led 2 13,13::C:2
led 3 14,14::C:2
led 4 15,15::C:2
led 5 5,15::C:2
led 6 6,15::C:2
led 7 7,15::C:2
led 8 8,15::C:2
led 9 9,15::C:2
led 10 4,11::C:2
led 11 3,12::C:2
led 12 2,13::C:2
led 13 1,14::C:2
led 14 0,15::C:2

# aux
aux 0 0 0 900 1300 0 0
aux 1 13 2 1700 2100 0 0
aux 2 15 6 900 1075 0 0
aux 3 35 4 900 1300 0 0

# adjrange
adjrange 1 0 6 1100 2100 30 6 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 990 2011
rxrange 2 987 2011
rxrange 3 990 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 13 20 26
vtxtable powerlabels 25 100 400

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set ledstrip_race_color = GREEN
set osd_vbat_pos = 384
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2423
set osd_current_pos = 406
set osd_mah_drawn_pos = 2456
set osd_altitude_pos = 2401
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2347
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC
set name = Marmotte

profile 0

# profile 0
set vbat_pid_gain = ON
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10

rateprofile 0

# rateprofile 0
set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 105
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

# end the command batch
batch end
```

```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: MOTOR 2
A03: MOTOR 1
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: SERIAL_TX 5
B14: FREE
B15: SERIAL_RX 1
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: ADC_RSSI
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: SERIAL_TX 2
D06: SPI_MOSI 3
D07: BEEPER
D08: FREE
D09: FREE
D10: SDCARD_DETECT
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: OSD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 4
    CH2: MOTOR 3
    CH3: MOTOR 2
    CH4: MOTOR 1
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: MOTOR 1
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

 
```
# dshot_telemetry_info
Dshot reads: 268399
Dshot invalid pkts: 466
Dshot directionChange cycles: 984, micros: 2

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0     0.66%
    1         0        0       0     0.00%
    2         0        0       0     0.00%
    3         0        0       0     0.00%

117 133 149 165 197 213 229 246 278 294 310 326 342 358 406 14 0 0 3373138432 1990328054 572784642 4290238063 
16 16 16 32 16 16 17 32 16 16 16 16 16 48 4294966904 4294967282 0 3373138432 2912156918 2877423884 3717453421 

# 
```

```
tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9      12       0    0.5%    0.0%         0
01 - (         SYSTEM)    999      14       1    1.8%    0.5%       124
02 - (       GYRO/PID)   8002      55      47   44.5%   38.1%     41653
03 - (            ACC)    999      26       9    3.0%    1.3%      1057
04 - (       ATTITUDE)     99      18       4    0.6%    0.5%        58
05 - (             RX)     32      34      20    0.6%    0.5%        81
06 - (         SERIAL)     99  314493       0 3113.9%    0.0%       665
07 - (       DISPATCH)    998      13       0    1.7%    0.0%        30
08 - (BATTERY_VOLTAGE)     49      14       1    0.5%    0.5%         6
09 - (BATTERY_CURRENT)     49      12       1    0.5%    0.5%         2
10 - ( BATTERY_ALERTS)      4      13       1    0.5%    0.5%         0
11 - (         BEEPER)     99      12       1    0.6%    0.5%         6
14 - (           BARO)    153      30       6    0.9%    0.5%       112
15 - (       ALTITUDE)     39      14       3    0.5%    0.5%        11
17 - (      TELEMETRY)    249      24       0    1.0%    0.0%        81
18 - (       LEDSTRIP)     99      34       3    0.8%    0.5%        40
20 - (            OSD)     59     234      13    1.8%    0.5%       104
22 - (            CMS)     59      11       1    0.5%    0.5%         3
23 - (        VTXCTRL)      4      25       0    0.5%    0.0%         2
24 - (        CAMCTRL)      4      12       0    0.5%    0.0%         0
26 - (    ADCINTERNAL)      1      12       1    0.5%    0.5%         0
27 - (       PINIOBOX)     19      12       0    0.5%    0.0%         1
RX Check Function                   2       0                         2
Total (excluding SERIAL)                        62.3%   45.9%

# 
```

**Setup / Versions**
 - Flight controller: H7EXTREME
 - ESC: TekkoF3 Metall 65A with 32.7.0 of blheli32
 - Esc are connected via extra ground and pwn wires not twisted
-F40 pro III 1600kv

**Additional context**
Here is a short video: https://youtu.be/7YiYsa2Esek

[image]

joelucid, 2019-10-24:
Transferred back to Betaflight since this issue has no connection to bidir dshot snippets.

mikeller, 2019-10-24:
@maGfpv:

> set dshot_bidir = ON

Use `set dshot_bidir = off` and it will work.

closed 2019-10-24 by mikeller