betaflight/betaflight #9101: OpenLog is NOT suitable for serial black box recording with 4.0 and beyond

jflyper, 2019-10-25:
As a result of never ending demands for different log items for better and more diverse flight characteristics analysis, output bandwidth requirement for serial blackbox output finally exceeded 230.4Kbps which is the fastest supported by OpenLog.

From 4.0 and on, serial blackbox users must use a logger that supports faster speed such as OpenLager instead, and configure baudrate at 1.5Mbps at least for stable recording.

- See https://github.com/betaflight/betaflight/issues/9043#issuecomment-544333986 for bandwidth requirement for varying logging rates.

- OpenLager information is available at https://github.com/d-ronin/openlager/wiki

tavdog, 2022-06-11:
Close time ?

haslinghuis, 2022-06-11:
Information contained here: https://github.com/betaflight/betaflight/wiki/Serial-BlackBox-Logging

closed 2022-06-11 by haslinghuis

KarateBrot, 2022-06-11:
Additional info from the future:
We can now **disable blackbox headers** to save logging overhead. This can make OpenLog work again because it doesn't need to handle as much data. We have presets to disable blackbox headers. Try it and see if slower loggers will work again with less data to log!

haslinghuis, 2022-06-11:
@KarateBrot added to the wiki