betaflight/betaflight #9103: ESC init tones out of sync with bidirectional DShot enabled

benedikt-bartscher, 2019-10-26:
I had this issue on multiple quads, all with Betaflight 4.1 + BLHeli32 32.7. ESC startup tones are not synced with bidirectional DShot enabled!

jflyper, 2019-10-26:
Questions.

(1) How far a part are tones separated when this happens?

(2) Do you observe non-zero RPM telemetry errors in Motors tab?

(3) Do you have any particular issue during flights involving motor functionality?

RipperDrone, 2019-10-26:
Have it, too. On OMNINXT7 with BLH32 32.7 RacerStar Metal 50A 4in1 ESC: Sometimes they boot in sync, most times 2 boot in sync, then there is a delay of ~1sec before the others boot. Flying ok, though...

joelucid, 2019-10-26:
@RipperDrone: I have seen these issues when the dshot signal line is pulled high during boot. If board initialization takes long enough the escs can go into boot loader mode and it will take some time for them to timeout and go back to signal detection mode. If you have a scope could you record a trace of the dshot signal line while this is happening so I can see if that's the cause?

jflyper, 2019-10-26:
@RipperDrone Can you answer the questions in https://github.com/betaflight/betaflight/issues/9103#issuecomment-546563252 ?

benedikt-bartscher, 2019-10-26:
The tones are always different far appart from each other. Sometimes one ESC even starts after all other finished. I have 0% DShot Errors on these builds with DShot 300 and even DShot 600. I don't have any issues during flight, it's only the strange init sound.

RipperDrone, 2019-10-26:
@joelucid sry no scope here...

RipperDrone, 2019-10-26:
@jflyper I have answered #1 and #3 already in my previous comment. #2: Error rates <0.2%

IvoFPV, 2019-10-26:
@joelucid I have seen this on many quads as well, doesn't affect performance or anything, just sounds weird. Only with dshot bidir enabled. Dshot300, Dshot600, lower gyro/pid loop frequency, 0% telemetry errors, same thing, doesn't affect it. 
It seems random, sometimes tones are close enough together, sometimes like 0.5sec off on one ESC.

@RipperDrone Suggestion: don't use #(number) because that references other PRs or issues 😄

jflyper, 2019-10-26:
@RipperDrone All motors showing small errors? And if not, are motors indicating RPM telemetry fixed, or they change by power cycling?

benedikt-bartscher, 2019-10-26:
The problem isn't related to DShot erros! I have some quads with and some without Dshot errors and this problem appears indepentently of these Errors. It has something to do with the boot process of Betaflight, the problem does not appear when i first power the FC with 5V via USB and then plug in the Lipo!

RipperDrone, 2019-10-26:
@jflyper: Have power cycled 10x: Motors are always indicating 0.00% errors now. rpm telemetry active on all motors any time. They did show 0.1x% errors in 1 case when I tested last time. In any case, all motors running flawlessly, just like @benedikt-bartscher confirms. Delayed boots of individual ESCs seem to happen more often when power cycling on 6S than 4S (same quad). It looks like being timing / watchdog issue, or voltage settling time to low/high levels... I have quite a big 1000uF 63V capacitor on this build. Maybe it's related, maybe not :-)?

benedikt-bartscher, 2019-10-26:
I tested on quads with no capacitors and on quads with one 470uf and one quads with 4x470uf. All quads are 4s, all quads have the same problem. All quads have single escs!

RipperDrone, 2019-10-26:
@benedikt-bartscher BLH32 only, like on mine? I have JESC BLH_S quads on DShot_bidir and never seen it on them :-)

benedikt-bartscher, 2019-10-27:
@RipperDrone they are all BLH32, i've only tested JESC once, can't remember if startup tones were synced...

benedikt-bartscher, 2019-10-27:
Problem only happens with bitbanged dshot (set dshot_bitbang=on) not with timer based dshot (set dshot_bitbang=off)

mikeller, 2019-10-27:
@benedikt-bartscher: Can you please supply the information that you were asked to supply when opening this issue?

benedikt-bartscher, 2019-10-27:
Describe the bug: already did this

To reproduce: plug in a lipo

Expected behavior: synced esc init tones

Flight controller configuration: default config, only turned on bidirectional dshot via configurator (bitbanged dshot)

resource show all: doesn't really make sense, i havend remapped anything and tried with lots of different F7 boards

Setup / Versions: 
FCs: Matek F722-SE, Luminer Lux F7, SPRACING H7 Extreme and others. 
ESCs: Wraith32 V2.2, Mamba506 and others.
All with BF 4.1 and BLH32.7.

benedikt-bartscher, 2019-10-27:
Here's one example diff:
'''
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Sep 22 2019 / 13:06:12 (a54c33b95) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# name: mamo

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature OSD

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 2 115200 57600 0 115200

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 225
set gyro_lowpass2_hz = 350
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 900
set acc_calibration = 122,17,152
set min_check = 1005
set max_check = 1995
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_delay = 10
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 335
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set deadband = 2
set pid_process_denom = 1
set osd_vbat_pos = 2497
set osd_rssi_pos = 2504
set osd_tim_1_pos = 385
set osd_tim_2_pos = 2433
set osd_gps_speed_pos = 2167
set osd_gps_sats_pos = 2471
set osd_home_dir_pos = 2081
set osd_home_dist_pos = 2135
set osd_flight_dist_pos = 87
set osd_altitude_pos = 2103
set osd_warnings_pos = 14799
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2250
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5806
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900
set name = mamo

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 250
set dterm_lowpass2_hz = 200
set p_pitch = 68
set d_pitch = 57
set f_pitch = 133
set p_roll = 64
set d_roll = 53
set f_roll = 126
set f_yaw = 102
set d_min_roll = 30
set d_min_pitch = 33

rateprofile 0

# rateprofile 0
set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 120
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

# end the command batch
batch end
'''

mikeller, 2019-10-27:
@benedikt-bartscher: Can you please also supply the output of `resource show all` as requested when opening the issue?

Also, there seems to be an awful lot of completely configuration settings completely unrelated to 'only turned on bidirectional dshot via configurator' in the `diff` you supplied - it looks like resetting to defaults failed. Please do a `defaults` in CLI, and then only enable bidirectional Dshot, and try again.

benedikt-bartscher, 2019-10-27:
@mikeller that was just a random log i found on my Laptop yesterday. Here's a clean one from my main quad:
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# master
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```

resource show all:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2019-10-27:
The data you are posting is highly incongruent and does not make any sense - I suspect you are stuffing up your configuration somewhere along the way.

Can you please do what I told you in https://github.com/betaflight/betaflight/issues/9103#issuecomment-546658854, and carefully only turn on bidirectional Dshot, and then test again.

It is impossible to help you if you are testing with a messed up configuration.

benedikt-bartscher, 2019-10-27:
What? I did exactly what you told. There is no messed up configuration in the diff above. I typed in defaults, turned on bidirectional dshot in Configurator (10.6.0) and posted a diff!

benedikt-bartscher, 2019-10-27:
Here's another diff from my 2nd quad, same problem:
```
# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Oct  6 2019 / 06:34:08 (c6452a55c) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# master
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600

profile 0

rateprofile 0

# end the command batch
batch end
```

resource show all:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: ADC_EXT
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: USB_DETECT
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2019-10-27:
You posted two diffs that had extra settings, with the one that you claimed being the one you tested with having a lot of extra settings and features enabled.

> I typed in defaults, turned on bidirectional dshot in Configurator (10.6.0) and posted a diff!

I didn't ask you to post a diff - I asked you to try and reproduce with a clean configuration because the previous one was obviously not what you thought it was.

benedikt-bartscher, 2019-10-27:
I already told you, first diff was just a random diff i found on my laptop, because i had no quad lying around yesterday. I tested 2x quads now with a clean configuration as you told me. And posted the diffs and resource show all.

mikeller, 2019-10-27:
@benedikt-bartscher: 

> I already told you, first diff was just a random diff i found on my laptop, 

If you want to waste the time of the people who's help you are requesting by posting random unrelated stuff instead of what you are asked to provide then here is the wrong place to do it.

mythinclient, 2019-10-27:
I have the same problem with my Matek F722SE. If i reset to defaults and enable bidir, i get the same diff as @benedikt-bartscher 
Some of my friends are having this issue too. As @benedikt-bartscher told, it only happens with bitbanged dshot, not with timer based.

diff:
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722SE   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id MTKS

# master
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set gyro_1_align_pitch = 1800
set gyro_1_align_yaw = 1800
set gyro_2_align_yaw = 900

profile 0

rateprofile 0

# end the command batch
batch end
```

resource show all:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: SDCARD_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mythinclient, 2019-10-27:
> You posted two diffs that had extra settings, with the one that you claimed being the one you tested with having a lot of extra settings and features enabled.

That's not true, if i type defaults for my Matek board i get the same diff

bennebartsch, 2019-10-27:
Hey, benedikt-bartscher here. I had to write from this account because @mikeller blocked me from this conversation. I tried to do everything to answer his questions quick and precise. I just wanted to help and make betaflight better and be a part of this great community. Me and others in this thread are having this problem, and as i already told multiple times: I have it with lot's of flight controllers and ESCs. If you don't belive me, this is not my problem! Yes i posted a diff with some unrelated config, but i had this problem with that configuration and at the moment i had no access to my quads.

The extra config in my 2nd diff seems to come from the new unified targets, i think you should know this as a developer. Don't always blame others, think about it just a second. I am here to help solving this problem, i don't wanna waste my/your time. But i think you are doing this right now.

Some more key facts about this problem (i already talked about them in this thread, just to sum up):
- Problem happens on BF 4.1 with BLH32.7
- Problem happens on multiple flight controllers and ESCs
- There are many people having this problem
- Problem does not apper if you first plug in your FC via USB and then plug in a Lipo, so it's somehow related to BF boot process
- Problem only appears with bitbanged bidir dhsot, not with timer based bidir dshot
- Problem appears with 0% Dshot errors
- Sometimes tones are synced, sometimes not. To reproduce you have to plug in a lipo multiple times. Delay is always a bit differnt.

RipperDrone, 2019-10-27:
Instead of continuing this conversation in 'nit-picky' mode, can we agree on the following?:

@IvoFPV @benedikt-bartscher @mythinclient and myself have experienced the same issue with the following findings so far:

- seems to be related to BLHeli32 and DShot_bidir in non-timer based mode only
- not depending on ESC or FC brand / type
- not related to capacitors / ESC voltage supply side
- even on same setup, it varies btw 
  - 'all boot ok', and 'some ESC's boot up delayed'
  - delays from some 1/10ths of a sec to 1sec
  - delayed boot ESC's can be 1 or 2 (or maybe even 3?)
- it has been confirmed in configurations where ONLY dshot_bidir (bitbanged) has been activated from defaults (in my case along with 4k/4k, DShot300 setting to make sure NOT to overload the CPU)

Most sensible hint IMHO came from @joelucid suspecting some ESC's first entering boot mode because the signal line from the FC is not readily bootet and has stable signal before the ESC is to be bootet.

Please add to this summary if there is anything else... :-)

mythinclient, 2019-10-27:
I'd like to add that this problem does not appear when you first plug in the FC via USB and then plug in a lipo, as @benedikt-bartscher mentioned.

Btw: You should stop blocking people which like to contribute to betaflight, that is not good for open source software.

mikeller, 2019-10-27:
@mythinclient, @RipperDrone: Have any of you actually reproduced this behaviour in a minimal configuration, flight controller reset to defaults, with only Dshot and bidirectional Dshot enabled?
Trying to take potshots at which feature might be introducing the delay that leads to the double initialisation is quite pointless.

@mythinclient: Wasting the time of people who are trying to help by posting fake information is not 'good for open source' - it just means the time isn't spent helping other users.

mythinclient, 2019-10-27:
Me (and @benedikt-bartscher) already posted a clean diff with only bidirectional DShot enabled from GUI. And with this configuration i have the problem. So YES!

n2rovert, 2019-10-27:
Me too . Just as everyone described above. I only have 1 machine with Matek F722SE

RipperDrone, 2019-10-27:
@mikeller yes, defaults on OMNINXT7 4.1, then only changed dshot_bidir / bitbanged settings mentioned in my summary post, nothing else. Easy to reproduce on (m)any BLH32 setup(s), it seems.

RipperDrone, 2019-10-27:
Pls see my additions to summary: The fact that it doesn't show this behavior when USB is plugged in first actually supports @joelucid's idea that it has to do with timings for signal lines from FC vs timings for ESCs to go into boot mode due to signal line level from FC being HIGH when the ESC is powered. It then takes time to settle back into std boot mode, so tones are delayed.

@joelucid Since it doesn't happen in 100% of power cycling, but sometimes takes 3-4 tries before a d'delayed boot-up' scenario is encountered: This could mean that the timing glitch is very minor, and a tiny little change in grant periods in the boot / ESC comm sequence could fix this issue. Can you comment on in which way booting / signal detect timings are any different in bitbanged mode vs timer based mode? This seems to make THE difference. Looks like we need to make the FC pull the ESC lines to LOW a lot faster on startup...?

mikeller, 2019-10-27:
@mythinclient: When you did the reset to defaults, did you have a battery plugged in? I am a bit puzzled by the fact that the default settings that you post contain `set use_unsynced_pwm = OFF`, which suggests that your flight controller is auto-detecting as being wired to brushed motors - this could be caused by the ESCs doing something unexpected on the signal line during the reset, and will probably have to be addressed.

mythinclient, 2019-10-27:
> @mythinclient: When you did the reset to defaults, did you have a battery plugged in? I am a bit puzzled by the fact that the default settings that you post contain `set use_unsynced_pwm = OFF`, [...]

I just tried again, no lipo plugged in. Always gettings this diff after resetting to defaults

mythinclient, 2019-10-27:
> @mythinclient: Wasting the time of people who are trying to help by posting fake information is not 'good for open source' - it just means the time isn't spent helping other users.

as i can reproduce the diff @benedikt-bartscher posted i don't think it was "fake information"

mikeller, 2019-10-27:
@mythinclient: How odd. My (naked) MATEKF722SE board does not do this with 4.1.0. And I don't think having powered down ESCs should make a difference. You don't have an unconnected MATEKF722SE available that you could use to verify if you get this or not - it could be that you have a different board.

> as i can reproduce the diff @benedikt-bartscher posted i don't think it was "fake information"

He admitted himself that he posted random data instead of something that would have been helpful to reproduce the problem:

https://github.com/betaflight/betaflight/issues/9103#issuecomment-546681644

mythinclient, 2019-10-27:
> @mythinclient: How odd. My (naked) MATEKF722SE board does not do this with 4.1.0. And I don't think having powered down ESCs should make a difference. You don't have an unconnected MATEKF722SE [...]
Sorry, i only have a MATEKF722SE in one of my builds. I think we should open a new issue about this, cause it isn't really related to this problem. I cannot reproduce this with other (naked) boards i have lying around. @n2rovert , can you reproduce this?

klutvott123, 2019-10-27:
I'm not sure if the root cause of this issue is bidirectional/bitbanged dshot as I have been experiencing this long before enabling any of them on two quads. Sometimes the ESCs startups are just out of sync and it flies fine. Sometimes they are out of sync and one motor won't spin. Seems to happen with blheli_32 ESCs. Can't remember having experienced it with blheli_s. 
ESCs are Aikon 35A single and 4 in 1 with firmware 32.6 and 32.7.

Just did a test with ``defaults`` on MATEKF722 with 4.1 and was able to reproduce. 
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.0 Oct 16 2019 / 11:58:45 (c37a7c91a) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722
manufacturer_id MTKS

# master
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end
```

joelucid, 2019-10-27:
Does anyone have a jesc quad to test this on? It doesn't happen on mine. If this is specific to blheli32 this might be an issue for @sskaug.

RipperDrone, 2019-10-27:
I have 2 jesc quads (AIO toothpicks, Mamba and BetaFPV whoop PCBs), burnt 16 batteries this week. It never happened once that ESCs booted up delayed / sequentially.

mythinclient, 2019-10-28:
Just flashed JESC on one of my BLHeliS ESCs. Startup tones are always synced

joelucid, 2019-10-29:
@sskaug, any idea why blheli32 might show this behavior?

sskaug, 2019-10-29:
First of all - I assume it is the last two tones that are delayed?
There are two things in BLHeli_32 that can affect startup time. If the input signal is high for a longer period of time, then it can enter the bootloader. Secondly BLHeli_32 does input signal detection via DMA, and there are retries in place in order to assure that the DMA stream is captured from the beginning of a frame. As soon as I have time, I can dig into what is the cause for the variable delays.

RipperDrone, 2019-10-30:
@sskaug - no, the whole tone sequence erratically sounds with delay for individual ESCs, all 5 tones. I made another discovery: If I switch magnetometer on on an FC that doesn't have mag hardware, the FC boot process will take almost twice as long as with mag correctly witched off. This prolonged FC boot seems to settle all ESC signals to 'correct' levels --> I could never replicate the delayed ESC booting with this configuration. In std / non-mag mode where FC booting is a lot faster, the delayed ESC init occurs sometimes... (OMNINXT7, 6S)

paulo-serrao, 2019-12-09:
I have the same issue. 

ESCs: 4x HAKRC BLHeli_32 35A (latest blheli32)
FC: Foxeer F405 AIO

Running rpm filters at 4k/4k dshot300

I can provide diff if needed.

sskaug, 2019-12-09:
There is a fix in the latest test code: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.1%20SBUS%20and%20S.PORT%20testcode

Asizon, 2019-12-09:
> There is a fix in the latest test code: https://github.com/bitdump/BLHeli/tree/master/BLHeli_32%20ARM/Rev32.7.1%20SBUS%20and%20S.PORT%20testcode

Thanks! Im going to try now on Furling32 4in1, why sbus sport named? Im using Fport lol

TheMightyThor1212, 2019-12-31:
Did the test Hexs fix this problem? i have an HGLRC 4in1 ESC behaving this way. please advise

Asizon, 2019-12-31:
> Did the test Hexs fix this problem? i have an HGLRC 4in1 ESC behaving this way. please advise

Try it and see @TheMightyThor1212  :)

Asizon, 2019-12-31:
I have tried this beta build with my Furling32_Multi and Furling32 4in1, out of sync init tones persist

closed 2020-02-06 by stale[bot]

BombowaKuna, 2020-07-20:
I got the same isue on my new 6S quad. BF 4.2, Blheli_32 Rew 32.7, 20x20mm Holybro Kakute F7 Mini V2 Flight Controller & Tekko32 F3 45A Mini BL_32 4in1 ESC

Lozagon1234, 2021-03-29:
I changed one ESC BLHeli_S and started to have this issue. I FIXED it by updating to the latest firmware. (16.71 right now, it says its a beta idkw but it fixed the issue for me).