betaflight/betaflight #9105: Max power output limit feature discussion.

junwoo091400, 2019-10-26:
**PROBLEM: For a drone to be qualified for fpv racing competitions, the maximum power must be below certain threshold. But currently in betaflight, a way of limiting power exists as a 'throttle limit', but that just overall sacrifices performance and control authority**

**It would be great if the board can limit the power output to a certain value.**
I think this could be achived in a pid or mixer or main loop, where a moving average filter or a simple low-pass filter could track the power output, and when it violates the user-defined limit, certain deduction should be applied to the pwm output, etc.

**Throttle limiting. But this feature just limits the control authority. Instead, it would be better if you can just get all the control, but in extreme power peaking cases, limit the output. Giving user much more flexibility.**

Docteh, 2019-10-27:
What sorts of limits are they specifying for the competitions? Is it in watts? horsepower? RPM?

Would the existing `motor_output_limit` feature be of any use? 

The qualification procedure sounds interesting. do they mount a drone on something and have it go full throttle?

junwoo091400, 2019-10-27:
About the qualification, the competition organizer places a power meter on each drone, to check on it after the race is finished. The limit for now is known to be around 1800W.

And as you've suggested, motor_output_limit could be used. But that limits motor output capability severely(I think this is especially the case when you want to do a super fast roll maneuver, but with that limit, the full potential is blocked. Although you are using only 2 motors, hence under the power limit).

mikeller, 2019-10-27:
@junwoo091400: Do you have a link to the rules that are defining the requirements and procedures for this limit to be established?
I suspect that any attempt to game the system by applying a form of dynamic limiting (that only cuts in _after_ the limit has been exceeded) will most likely result in any firmware implementing it being banned from any competition implementing this limit, as this will give pilots an unfair advantage over pilots implementing a hard limit.
In this respect using `motor_output_limit` and repetitive testing to establish what limit results in the desired maximum power output is probably the best way to make your craft compliant with the rules.

junwoo091400, 2019-10-27:
@mikeller The competition doesn't have any publicly available rules, and no, there is no requirements / procedures for establishing the limit. And as far as I know, the organizer of this competition is aware of such feature, and allows dynamic-power limiting, viewing it as a 'technological strength'.

So I think the rules/regulations part is clear(btw, I was surprised also, by the fact that such feature is allowed, but I guess it's ok for certain competitions).

mikeller, 2019-10-27:
@junwoo091400: Well, in this case I suggest you define two profiles, one that uses full power for the race, and a second one that uses `motor_output_limit` to limit output to 1800W. And at the end of the race, before testing, you simply change to the second profile. There you go, you 'dynamically' adjust the power to the conditions. :stuck_out_tongue_closed_eyes:

junwoo091400, 2019-10-27:
@mikeller Haha, clever😛. But I think you might've mis-understood(or, you're just joking haha), the logging data is collected During the race, which gets used to determine if you've passed the power limit, after the race. So a Legit dynamic power limiter is needed😁😂🤣

mikeller, 2019-10-27:
@junwoo091400: I don't think we are getting anywhere here. Unless you can supply the exact regulations of how much overshoot of the power limit during the race is allowable for how long, any discussion of dynamic power limiting is pointless, as it will lead to overshoot of the power limit during the race (if it is ever to cut in), getting you disqualified.

junwoo091400, 2019-10-27:
Alright, straight to the point, No overshoot is allowed. Basically 1800W and up is all disqualified.

mikeller, 2019-10-27:
@junwoo091400: Ok, that makes it hard to implement anything dynamic - but your idea of using a combined motor output limit instead of a per-motor one should still bring you closer to what you want. It might be hard to calibrate though, as the power consumption of the motors is most likely not linear with the ESC output signal.

junwoo091400, 2019-10-27:
@mikeller Exactly. Staying under certain power at all times is definetly challenging. 

Summing up the motor output and limiting them(scaling down) sounds great.

I thought about implementing a code that will kick in whenever user is at the Edge(ex. 1700 ~ 1800 [W]). That switches into a motor output limiting mode.

Currently, not sure how to tackle this problem. Communication with esc(blheli) is maybe another possible way.🤔

junwoo091400, 2019-10-28:
@mikeller @Docteh Just in case, here's my development branch on implementing this feature. This is my first time actually implementing something into the code(other than bug-fixes), so could be messy & wrong! https://github.com/junwoo091400/betaflight/tree/max_power_limit_dev

mikeller, 2019-10-29:
@junwoo091400: It will be easier to review / discuss if you open a pull request for it - if you aren't finished with it you can mark it as 'Draft'.

junwoo091400, 2019-10-29:
@mikeller Sure. Will do that!

junwoo091400, 2019-10-29:
Opened the pull request #9132 for this issue.

closed 2019-12-05 by stale[bot]