betaflight/betaflight #9106: RunCam Camera Control via UART

benedikt-bartscher, 2019-10-26:
I tested two Runcam Cameras with Serial Control (Runcam Swift 3 and Runcam Hybrid) and hooked them up to an UART. I can control the cameras with buttons assigned in the Modes Tab. But i cannot send any Stick Commands to get to the Camera OSD.

IvoFPV, 2019-10-26:
Have you had any luck with this guide?
https://github.com/betaflight/betaflight/wiki/RunCam-Device-Protocol

benedikt-bartscher, 2019-10-26:
I did everything along this guide and as already told, the Commands in Modes tab are working. But joystick mode / stick commands aren't

Ozyris, 2019-10-30:
I have the opposite issue)) I get OSD commands from UART, but modes switching does not affect UART output. 
I guess the reason is a wrong or untimely response to the info request.

FMarcelino, 2019-11-19:
I have not able to talk with caddx turtle v2 (latest firmware) via UART. Neither commands bottoms or stick osd commands. I plug an scope on the tx line and saw the data goind to the camera during some initialization phase, but no more. 

My first question, is Caddx Turtle V2 really compatible with RunCam protocol?

FMarcelino, 2019-11-19:
[image]

mikeller, 2019-11-21:
@FMarcelino: We do not have any communication with Caddx, nor are we supplied with engineering samples of their products, so if you want an answer to this you will have to contact them directly.

FMarcelino, 2019-11-22:
Sorry and Thank you guys,
I am not surprised, unfortunately 🙁
I read a couple articles about this "battle" (Betaflight vs Caddx) and my hope was it had been fixed already.
But It seems not yet. I understand the Betaflight point of view and I consider it´s fair!

Regards,

FMO

bri3d, 2019-12-02:
I think this might be the same as #8283 , which I left this comment on: https://github.com/betaflight/betaflight/issues/8283#issuecomment-560639705 . tl;dr: button commands work until I try to stick into the OSD, then the whole RunCam setup stops working. 

I will try to both follow the code and probe what's getting sent later tonight to see if I can make any headway.

closed 2020-01-09 by stale[bot]