betaflight/betaflight #9107: Dshot 1200 not available for Diatone Mamba F722s Stack (MAMBAF722)

Scott594, 2019-10-26:
The Diatone F722s Stack supports Dshot 1200 according to Diatone, but the pulldown menu of available firmware doesn't list it. I know it's a minor issue, but thought you may want to know about it.

To reproduce, flash the target MAMBAF722 (Diat) to the flight controller and go to  the Configuration page, the ESC/Motor Features section and the ESC Motor Protocol selection. Dshot 600 is listed and works well, but Dshot 1200 is not listed. 

All Diatone literature I have seen shows Both the FC and ESC rated for DShot 1200.

IvoFPV, 2019-10-26:
In Betaflight 4.1.0 there is no more dshot1200. It has been removed as it provides no benefit over dshot600 at 8k/8k gyro/pid loop frequency, and doesn't work well with dshot bidirectional and dshot bitbang.
Diatone will have to change their documentation, otherwise you can use prior versions where dshot1200 is still available like Betaflight 4.0.6.

Scott594, 2019-10-26:
Thanks IvoFPV, I could have sworn I have that stack in another quad running 1200, but couldn't remember which one. That explains it, I'm perfectly happy with 600, but why not run the faster protocol if it's available? I just thought it had slipped thru the cracks, I don't remember seeing anything in the release notes, but it's hard too keep up with the changes. Sorry to have bothered you over this.

closed 2019-10-26 by Scott594