betaflight/betaflight #9109: Wrong satellites quantity in smartport telemetry

Alex-g-by, 2019-10-26:
Seems wrong satellites quantity in smartport telemetry in BF4.1.
Tmp2 (Sats quantity) value looks like 3251 (numbers may vary).Photo below
In this case, sats quantity is 51! It's impossible !

In version BF3.5.7 everything was in order.

Hardware:
Eachine Tyro109 quad (Customized F4 Flight Controller) - MATEK F405
Frsky r-xsr receiver
BN-220 GPS connected UART5
Taranis X9D Plus

'''
# 

# diff all

# version
# Betaflight / MATEKF405 (MKF4) 4.1.0 Oct 16 2019 / 11:42:57 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
mcu_id 003f00395048501520373350
signature 

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# feature
feature -AIRMODE
feature MOTOR_STOP
feature GPS

# serial
serial 2 2048 115200 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 6 4 1400 2100 0 0
aux 4 46 3 1400 2100 0 0
aux 5 13 2 1300 2100 0 0
aux 6 13 0 1300 1700 0 0
aux 7 28 1 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 0

# master
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 14
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set motor_poles = 12
set failsafe_delay = 10
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1500
set vbat_max_cell_voltage = 420
set vbat_min_cell_voltage = 340
set vbat_warning_cell_voltage = 360
set ibata_scale = 1095
set ibata_offset = -400
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_initial_alt = 30
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1000
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_rssi_alarm = 45
set osd_cap_alarm = 1100
set osd_vbat_pos = 2433
set osd_rssi_pos = 2146
set osd_tim_1_pos = 2423
set osd_flymode_pos = 2457
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2369
set osd_gps_speed_pos = 2136
set osd_gps_lon_pos = 2096
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2113
set osd_home_dir_pos = 2254
set osd_home_dist_pos = 2168
set osd_altitude_pos = 2155
set osd_warnings_pos = 14729
set osd_avg_cell_voltage_pos = 2401
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 1
set vtx_freq = 5665
set vcd_video_system = NTSC
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# '''

[image]
[image]
[image]

Thanks for your help !
Alexander

etracer65, 2019-10-26:
There were some changes to add the HDOP value to the telemetry value but it looks like the logic has problems and is corrupting the telemetry value. Related to #8472

etracer65, 2019-10-26:
@Alex-g-by Can you try this please:

[betaflight_4.1.1_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/3775374/betaflight_4.1.1_MATEKF405.hex.zip)

The second digit will contain a "position quality" indicator from 0 (worst) to 9 (best). The last 2 digits of the telemetry value should now show the correct number of satellites.

mikeller, 2019-10-27:
@etracer65: One thing to keep in mind here is that there could be a compounding factor in the bug that causes certain values to not be transmitted over SmartPort / FPort for some RX firmware versions: https://github.com/betaflight/betaflight-tx-lua-scripts/issues/151

etracer65, 2019-10-27:
@mikeller I don’t think there are any other issues. There’s a clear bug with how the HDOP value was added to the telemetry value (see linked PR). I was able to replicate the problem and also confirmed the fix.

mikeller, 2019-10-27:
@etracer65: Well, there _is_ the issue with certain RX firmware versions as discussed in betaflight/betaflight-tx-lua-scripts#151. I've had it before that this has thrown a spanner into diagnosing telemetry / MSP over SmartPort related issues, so I am letting you know to make sure you are not spending time diagnosing phantom issues. :wink:

Alex-g-by, 2019-10-27:
@etracer65 
Tested your BF_4.1.1.
Seems the problem is solved !
Thank you very much !
[image]

etracer65, 2019-10-27:
@Alex-g-by Thanks for testing.

PR #9112 (2019-10-28): Fix data type overflow in HDOP calculation in Smartport telemetry
Fixes #9109 

Calculation originally used uint16_t which overflowed and corrupted the satellite count portion of the telemetry value. Should have been `uint16_t`.

Related to #8472
closed 2019-10-27 by Alex-g-by