betaflight/betaflight #9113: Firmware 4.1.0 makes receiver data unstable on DALRCF405

datalifter, 2019-10-27:
Skystars TurboS F405 FC running STM32F405 with firmware 3.5.2 shipped on it is stable.  However upgrading to 4.1.0 makes the Receiver data appear corrupted with the animated perspective model spinning in all different direction.  FC unable to arm - not sure I would anyway at that point.

Downgrading via Legacy selection in BF 10.6.0 configurator to 3.5.2 fixed the problem.  I have not attempted to step through other firmware updates.

datalifter, 2019-10-28:
[image: Text file]
BTFL_cli_DIFF_Rooster_2019102…
<https://drive.google.com/file/d/1cPEkEEBkn1wOkyei5WNu-K7tx9-2Or2l/view?usp=drivesdk>

mikeller, 2019-10-29:
@datalifter:

> resource SERIAL_TX 3 B06

Looks like you are using a non-standard UART TX pin for your RX on 3.5.2.
Did you make the same modification in 4.1.0?

datalifter, 2019-10-29:
No swap on 4.1.0.  The swap was of the LED_STRIP to UART addresses.  I was
trying to troubleshoot why the LED strip was not responding.  Eventually
gave up on 3.5.2. for that feature.  Never tried it on 4.1.0.  Firmware
3.5.2 flies.

Question: When you Erase the EEPROM before flashing new Firmware I'm
assuming wipes all custom mappings?

I'll work on getting the a DIFF of 4.1.0 out to you in the next day or so.

Thanks

mikeller, 2019-10-29:
@datalifter:

> Question: When you Erase the EEPROM before flashing new Firmware I'm
assuming wipes all custom mappings?

Yes it does.

datalifter, 2019-11-07:
Sorry for the delay.  Attached is the "diff" file after applying firmware 4.1.0.  

1. When connecting to BFC for the first time after flashing it asked to apply custom default config.  I said yes.
2. Set Conf. for Spektrum 2048
3. Bound receiver, worked.
4. Receiver tab channels all jump around with any stick movement.

file attached.
[BTFL_cli_diff_410_20191106_185124.txt](https://github.com/betaflight/betaflight/files/3817677/BTFL_cli_diff_410_20191106_185124.txt)

datalifter, 2019-11-07:
Last update.  I rolled back to 4.0.5 and still same erratic channel inputs in Receiver tab.  I then rolled back to the last 3.x.x before 4.0.0.  3.5.7 works as expected.  I'll be staying there in this FC.

Cheers!

mikeller, 2019-11-07:
@datalifter: Interesting. If the problem exists in 4.0.5 as well but not in 3.5.7 then I am a bit perplexed as to where this is coming from - there was no significant change in the code dealing with SPEKTRUM RX between 3.5.7 and 4.0.5.

datalifter, 2019-11-07:
Hmm. Let me try 4.0.1. I can do that tonight.

datalifter, 2019-11-08:
Attached is a 5 sec. video of what the Receiver tab looks like after applying 4.0.1 and the ``diff``.

I'll stick with 3.5.7.  Thanks for the help.

[BTFL_cli_401_diff_20191107_183657.txt](https://github.com/betaflight/betaflight/files/3822760/BTFL_cli_401_diff_20191107_183657.txt)

DALRCF405 4.0.1 after SPEKTRUM2048 bind.MOV file
https://drive.google.com/open?id=1eNWA409415J0hh2Z8J6iBZTtWHfHJ606

mikeller, 2019-11-09:
@datalifter: It looks like some sort of mis-interpretation of the protocol. I am not sure what could be causing this, and I do not have Spektrum gear to be able to investigate, sorry.

AndersHoglund, 2019-11-10:
I am running a Spektrum 4649T on a SpeedyBeeF4 without any such problem. BUT I am using the SRXL protocol on a UART Tx line.  I would suggest you try that too, and also get the benefit of telemetry. 
But still, could there be a SPEKTRUM1024 vs SPEKTRUM2048 enumeration problem in later BF versions? Channels jumping around like that is a bit typical of a Tx vs Rx protocol missmatch, i.e. DSM2 vs DSMx or visa verse.

datalifter, 2019-11-10:
Interesting.  I've seen on YT that you need an adapter to flash the 4649T
to support telemetry? I've only got a 8-Ch. Gen 1. DX8 tx, and pretty much
used up all my channels for my quad configuration modes.

Good info!

AndersHoglund, 2019-11-11:
All 4649T firmware versions support TM. You need to update for RSSI support only. The DX8G1 should support TM, from what I have read. 

A few further tests done. Tried SPEKTRUM2048 (non-SRXL) on the same SpeedyBeeF4 and 4649T. Worked just fine. Tried setting SPEKTRUM1024 and I got jumping channels just like you., but also only 7 channels. Not the full set of twelve. 

How did you bind Rx andTx ? Did you use some other method other than `set spektrum_sat_bind=9` ?
I did try a few times to re-bind using a bind value of 7 to see what might happen. But I did not manage to get the Rx into bind mode. Using a bind value of 9 works just fine. Not sure what that was, never tried such negative test before....  

I would suggest you start over from scratch, i.e. reconfigure for SRXL and TM. Then rebind using a bind value of 9.

datalifter, 2019-11-12:
Hey Anders - thanks for the update and details.  The FC has the receiver
setup on UART1 by default and that's what I'm using.  I did not try the
```set spektrum_sat_bind=9``` method - I will tonight <with fingers
crossed>.  The RX would just start flashing rapidly indicating it went into
bind mode after hooking up the battery.  I then ran the normal steps from
the TX to complete the bind.  I did have to set the
```spektrum_sat_bind_autoreset
= off``` to keep it from wanting to bind after unplugging the battery.

Cheers

On Mon, Nov 11, 2019 at 2:39 PM Anders Höglund <notifications@github.com>
wrote:

> All 4649T firmware versions support TM. You need to update for RSSI
> support only. The DX8G1 should support TM, from what I have read.
>
> A few further tests done. Tried SPEKTRUM2048 (non-SRXL) on the same
> SpeedyBeeF4 and 4649T. Worked just fine. Tried setting SPEKTRUM1024 and I
> got jumping channels just like you., but also only 7 channels. Not the full
> set of twelve.
>
> How did you bind Rx andTx ? Did you use some other method other than set
> spektrum_sat_bind=9 ?
> I did try a few times to re-bind using a bind value of 7 to see what might
> happen. But I did not manage to get the Rx into bind mode. Using a bind
> value of 9 works just fine. Not sure what that was, never tried such
> negative test before....
>
> I would suggest you start over from scratch, i.e. reconfigure for SRXL and
> TM. Then rebind using a bind value of 9.
>
> —
> You are receiving this because you were mentioned.
> Reply to this email directly, view it on GitHub
> <https://github.com/betaflight/betaflight/issues/9113?email_source=notifications&email_token=ANSLFFKHPIPMMTAQI3E4EOLQTHNKTA5CNFSM4JFPU7S2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOEDYMEFI#issuecomment-552649237>,
> or unsubscribe
> <https://github.com/notifications/unsubscribe-auth/ANSLFFJONNMUA7IURFLN573QTHNKTANCNFSM4JFPU7SQ>
> .
>

-- 
Stephen Payne
Corvallis, OR / Swim-Bike-Run

Mobile (541)602-2419 <5416022419>
Email payneo@gmail.com

AndersHoglund, 2019-11-12:
Did the Rx go into bind mode all by itself by just connecting battery, you doing nothing else ? Weird. 

> I did have to set the `spektrum_sat_bind_autoreset = off`  to keep it from wanting to bind after unplugging the battery.

That is a major misunderstanding. It is the other way around. With `spektrum_sat_bind_autoreset = off` BF will not go out of bind mode automatically. It will keep on sending the `spektrum_sat_bind` value to the Rx on every power up. With `spektrum_sat_bind_autoreset = ON` BF will only try sending the bind value once and the automatically reset that.

datalifter, 2019-11-12:
Maybe I'm not describing this correctly.  Out of the box, the RX goes into
bind mode every time a battery is plugged in.  With either 3.5.7 or 4.1.0
newly flashed firmware I had to set the ```set spektrum_sat_bind_autoreset
= off``` value to prevent the RX from going into bind mode each time.  I
also applied ```set spektrum_sat_bind = 1```. But after you mentioned
```...bind=9``` I did a search and found the following:

Other bind options values:
3 for dsm2 1024/22ms
5 for dsm2 2048/11ms
7 for dsmx 2048/22ms
9 for dsmx 2048/11ms

Now I'm curious and will go try the 7 and 9 options.

Thanks!

AndersHoglund, 2019-11-12:
This makes no sense. The Rx should not go into bind mode that way. And as said earlier,  `spektrum_sat_bind_autoreset` does not work like that. All it does is control whether `spektrum_sat_bind` should be automatically reset to zero or not, hence `_autoreset `in the name. It is `spektrum_sat_bind` that actually controls if and how binding should be done, when set to Zero (automatically or manually) no binding will be done. What you are seeing is probably a coincident with something else that is triggering binding to occur. There are many ways to make a Rx go into bind mode, holding data line low when powering up is one. There has been such issues with shared pins on F3 boards etc. 

Again, start over from scratch. Maybe even test if UART3 would make any difference. 
But before that, make sure you have read the WiKi :  https://github.com/betaflight/betaflight/wiki/Spektrum-Rx-setup

datalifter, 2019-11-13:
@AndersHoglund tonight I tried your suggestions, short of taking the quad apart and switching UARTs.  No joy on several attempts.

I read through the WiKi and tried all those options, even using the 22ms and 11ms options while also switching the TX to 11ms and re-binding.  Nothing.  The RX would always bind but BFC RX tab would never show that it was receiving a signal using the SFXL option.  The only time I was able to get BFC RX to show input is falling back to the SPEKTRUM2048 option in the BF CFG tab.

I've attached the last ```diff all``` attempt with 4.1.0.

I see two last options.
1. Try the Legacy 4.1.0, I watched YT today with J. Bardwell mentioning that these are different firmwares?
2. Try different UART.  The DALRCF405 Wiki suggests UART4. 

Falling back to 3.5.7

[BTFL_cli_diff_all_410_20191112_191416.txt](https://github.com/betaflight/betaflight/files/3839265/BTFL_cli_diff_all_410_20191112_191416.txt)

mikeller, 2019-11-14:
@datalifter: Taking this back one step - is the 3.5.2 version that is working for you the firmware that was provided by the supplier of the board, or does it work as well if you use the official 3.5.2 target? If only the supplier provided firmware works then it's a case of the supplier using a custom build with changes we don't know about, which means we cannot help you unless the supplier adds an official target for this board.

datalifter, 2019-11-14:
@mikeller I'm running BF 3.5.7 from the Legacy firmware selection and all works great and fly's great too.  I'm confident a custom supplier firmware is not an issue - thanks for bringing that up, did not think that was a consideration.  Good to know.

Is the Legacy firmware version 4.1.0 the same as the non-Legacy version 4.1.0? I heard something on YT that they are slightly different.

mikeller, 2019-11-14:
@datalifter: They are different insofar as the non-legacy version uses a different approach of how the target specific parts are configured. So it is possible that the legacy version works, and the non-legacy version doesn't. If so, please report back.

datalifter, 2019-11-15:
Last update. I loaded up the legacy 4.1.0 and had the same result. I tried several of the RX type options with no luck.  I went back to 3.5.7 and everything works.  

I'll let this issue rest.

Happy Flying!

closed 2019-11-15 by datalifter