betaflight/betaflight #9118: 10.6.0 - CLRACINGF7 (legacy) 4.1.0 Target Bricks CLRACINGF7 Dual V2 Boards

Miloz44, 2019-10-27:
The CLRACINGF7 Dual (legacy) target no longer exists in 10.6.0 and trying to flash the CLRACINGF7 (legacy) 4.1.0 target on the CLRACINGF7 Dual V2 Boards has bricked this flight controller. 

The CLRACINGF7 (CLRA) 4.1.0 target does work with CLRACINGF7 Dual V2 Boards though I could not see Dshot 1200 so tried flashing CLRACINGF7 (legacy) 4.1.0 version and now only a power LED illuminates, the board is not recognized by the computer at all and bootloader could not be accessed by the button or ImpulseRC Driver Fixer and could not be found on Zadig.

The board was flashed to CLRACINGF7 (legacy) target in version 4.1.0 with full chip erase on.

mikeller, 2019-10-27:
@Miloz44:

- Support for Dshot1200 was dropped in 4.1.
- I am not aware of a CLRACINGF7 Dual V2 board - the normal CLRACINGF7 target has support for dual gyros, so I don't know if you are meant to use this target, or if this board needs a new, not yet published target. Maybe @bnn1044 can help?

bnn1044, 2019-10-27:
@mikeller it used same normal CLRACINGF7 target i will test it.

bnn1044, 2019-10-27:
@Miloz44 @mikeller  I just flashed it. first boot lock up active light stay on. 
 
unplug and plug it back it. work as normal. 

# status
MCU F722 Clock=216MHz, Vref=3.27V, Core temp=46degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3571, Max available config: 16384
GYRO=ICM20602, ACC=ICM20602
System Uptime: 17 seconds, Current Time: 2019-10-27T14:51:16.797+00:00
CPU:4%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 1 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP

# version
# Betaflight / CLRACINGF7 (CLR7) 4.1.0 Oct 16 2019 / 11:28:58 (c37a7c91a) MSP API: 1.42

etracer65, 2019-10-27:
Having an intermittent freeze during the reboot immediately following flashing the firmware is unfortunately normal for F722-based boards. Pretty much all the F722-based boards do this and it's some bug we haven't been able to find yet. It only happens immediately after flashing though so not a major problem - just power cycle.

mikeller, 2019-10-27:
@Miloz44: Sorry, I completely neglected to ask: Did you in fact try to power cycle the board, and verify that it did not boot normally after the power cycle?

Miloz44, 2019-10-30:
Yer the reason I tried legacy was because I thought it was a bug but later found out that it was a feature and removed, I guess that's my fault for not researching. I do believe there was another target, but it was an earlier build in 4.0.6 and earlier, but I could be wrong cause I can't find it but I don't know how to look at old code. 
@bnn1044 did you test the 4.1.0 legacy build or the CLRA cause mine worked fine when I flashed it with the CLRA version. 
I flashed CLRA, opened the config tab then immediately flashed the legacy version I don't know if that might have contributed to it.
@etracer65 I've power cycled it multiple times and I don't get any signal lights only power, it possible that it has frozen in bootloader. why would that cause the computer any computer not to recognize that anything was plugged in? 
Others have looked at it and can't figure it out either.

mikeller, 2019-10-30:
@Miloz44: That does not sound good - looks like you've got a dead board. If the board was new I'd return it under warranty.

Miloz44, 2019-10-30:
I had it flying perfectly before updating it was one of my best quads.

mikeller, 2019-10-30:
@Miloz44: Sorry to hear that. But if it is not detected as 'DFU' when pushing the boot loader button, and another board is when you try it on the same port, then it is very likely that either the MCU or the 3.3V voltage regulator is dead.

Miloz44, 2019-10-30:
Could it be anything else I just didn't think that updating a board could kill it, though perhaps if it was damaged prior. I'll do some diagnostics anything I should be looking for?

mikeller, 2019-10-30:
@Miloz44: Yeah, it wasn't the upgrade, it must have been damaged prior. If you can get DFU to work again then you'll be able to reflash it.

etracer65, 2019-10-30:
> Could it be anything else I just didn't think that updating a board could kill it, though perhaps if it was damaged prior. I'll do some diagnostics anything I should be looking for?

You should try disconnecting everything from the UARTs (RX and TX pins) like receiver, GPS, smartaudio/tramp, etc. Or at least make sure the devices connected aren't powered. Traffic on the UARTs can interfere with the MCU entering DFU. After disconnecting everything try powering with the boot button pressed. If it still won't enter DFU then I'm afraid it's probably a failed board.

Miloz44, 2019-10-30:
I can confirm the 3.3V works and  I have disconnected everything from it.

closed 2019-12-06 by stale[bot]