betaflight/betaflight #9124: new "tree mode" for getting the quad back from beeing stuck in branches high in a tree

wildflyer, 2019-10-28:
when you are stuck in unreachable altitude in a tree, arming or turtle mode are of no use because at least one propeller is not spinning at all, so you have to get your quad back with a rope or climb the tree etc. with a new "tree mode" you could get most stuck quad fast and easy without climbing etc ...
"tree mode" would be activated with a switch like arming switch. when activated, the yaw and pitch an your remote will work as individual motor activators: yaw left spins only motor 1, yaw right spins motor 2, pitch up motor 3 and pitch down motor 4. the thottle amount of each motor ist the proportional stick movement. so if your quad is stuck the you can spin the motors individually to "move" your quad so it can fall down to safety. 

if it is possible to reverse spin direction of the motors, you could move throttle stick to max (in tree mode) and this would reverse the spin directions.

this new feature would save a lot of time when rescueing your quad from a tree. 

greetings
wildflyer

IvoFPV, 2019-10-28:
You could spin up just one motor in turtle mode by pushing the stick in the corner. Although it would be reversed, so your option to run just one forward or reversed would be beneficial in some situations.

fiveangle, 2019-10-28:
While a very specialized use case, we have certainly all been there.  On a remarkably tall eucalyptus I had my Bluebeam antenna caught in a crack of the thin bark and only if I had the ability to simultaneously spin front rotors "down" and rear "up" would I have been able to get the quad w/o the 3 hrs of lassoing that it eventually took (and broke the then-dead Blackout when it hit the dirt from 80' up  :) 

If someone decides to put work into this, perhaps some thought on how it might be implemented to allow discreet selection of each motor + direction (if reversible ESCs are configured) ?

richardbulger, 2019-10-29:
Turtle mode has saved me frome multiple trees, but failed to help on the tallest one I've got stuck in (2 hours of rope tossing did the job). Having finer control of which motors to individually spin up certainty would of helped.
Could the both sticks be used? Eg 
throttle up = motor 1 clockwise
throttle down = motor 1 anticlockwise
Yaw left = motor 2 clockwise
Yaw right = motor 2 anticlockwise

Pitch up = motor 3 clockwise
Pitch down = motor 3 anticlockwise
Roll left = motor 4 clockwise
Roll right = motor 4 anticlockwise

wildflyer, 2019-10-29:
> Turtle mode has saved me from multiple trees, but failed to help on the tallest one I've got stuck in (2 hours of rope tossing did the job). Having finer control of which motors to individually spin up certainty would have helped.
> Could both sticks be used? Eg
> throttle up = motor 1 clockwise
> throttle down = motor 1 anticlockwise
> Yaw left = motor 2 clockwise
> Yaw right = motor 2 anticlockwise
> 
> Pitch up = motor 3 clockwise
> Pitch down = motor 3 anticlockwise
> Roll left = motor 4 clockwise
> Roll right = motor 4 anticlockwise

problem here would be the throttle stick, if you activate tree mode and forget to exactly center the throttle stick you could burn motor or esc. I think it would be safer to toggle the directions with the throttle stick.
greetings
joachim

richardbulger, 2019-10-29:
@wildflyer fair point, hadn't considered that. I suppose you could prevent arming tree mode based on throttle position, but might be a bit cumbersome. (Can it be named crab or spider mode, following the animal theme?)

fujin, 2019-11-04:
You already have yaw and diagonal control in flip over after crash (anti-turtle) mode. I don't see "tree mode" becoming a thing, sorry.

wildflyer, 2019-11-05:
> You already have yaw and diagonal control in flip over after crash (anti-turtle) mode. I don't see "tree mode" becoming a thing, sorry.

is it possible to spin each motor individually in anti turtle mode ? so if 3 motors are stuck and only 1 motor is free?

yoodog, 2019-11-07:
I would love that. As for pre-condition we could have arm check as: Tree Mode switch + ARM switch (no pre arm) + throttle position centered. 

All sticks would need some dead band at center because:
- throttle Motor 4, yaw motor 3 = than you want only Motor 3, you move it right for NORMAL direction, you will inevitably give some input on Motor 4 (throttle band), I guess 30 points +/- from central . position would be ok as there is no point to give super slow spin to a motor when you trying to get from brach  (normally you would try to flip over )
- implement current limit per motor (so if you set 20amps and spin 1 motor it limits to 20, but if you spin 2 motors it limits to 40 = not ultimate solution but still - RPM reading can help to detect getting stuck and cut input right away) on BF side to prevent burn out when you hit the tree while trying to flip out it.

kalle992, 2019-12-05:
why make it complicated?

lets just offset the turtle mode input by 45 degree so its easier to controll individual motors, but still posible to us two motors at a time.

make throttle input be direction, down is reversed, up is normal.

yaw can be to activate all but inn diffrent direction motor 1-3 normal 2-4 reversed and throttle to make it flip on the other axies with yaw input

darkmatter2222, 2019-12-23:
> why make it complicated?
> 
> lets just offset the turtle mode input by 45 degree so its easier to controll individual motors, but still posible to us two motors at a time.
> 
> make throttle input be direction, down is reversed, up is normal.
> 
> yaw can be to activate all but inn diffrent direction motor 1-3 normal 2-4 reversed and throttle to make it flip on the other axies with yaw input
> 
> and maybe calling it, turtle mode v2.0

Still like the name "Tree mode"

Bugz000, 2020-12-16:
the simpler option would be have it relate to throttle stick
what i mean by this is i fly right hand throttle, so have that to pulse the motors

because you are wishing to spin one motor at a time, you only really need to identify which motor, which can be done with the other stick, holding it at the respective corner you wish to spin, and use the throttle to induce rotation

want to spin rear-right motor? one stick to bottom right, and throttle up, throttle mid can also be used to identify direction

i also think constant throttle isn't needed, as i've never seen anyone use turtle mode in anything more than quick pulses, so to combat the "finding dead-center" problem, i see no reason why tree mode (or turtle mode for that matter) couldn't use pulses of 100% throttle in varying DURATIONS based on throttle input, each a couple seconds apart to allow components to cool down, plus, having worked with trees for many years, brute force is often not possible to dislodge something that's stuck in it, but as we call "bouncing the branch" can often result in getting your item back, this can be used to free carabiners lodged in forks or weighted throw-bags, you never just rip it out with force, you flick and bounce the tree

so to sum up, pitch/roll stick to the corner you wish to spin, at throttle mid, if configured, or throttle 0 if not configured, and raise throttle to induce increasing frequency of 100% throttle pulses, max stick being full duty cycle

this way it should be easy to find a resonant frequency to bounce the twig/branch back and forth until either the drone becomes dislodged or a stick snaps, releasing the drone, and should protect against burnt ESC's 
(perhaps the throttle pulses can be varied somewhat alongside the throttle pulse duration so it's not quite 100%, but i'd also prefer a drone to come down with a burnt ESC than lose it forever)

UglyBabyGaming, 2022-04-20:
using throttle position could be an option for turtle or tree just like the 3d mode throttle down = reversed motors ; Turtle mode and full throttle could do Tree mode spinning motors in correct direction while using the turtle function

closed 2026-09-26 by blckmn