betaflight/betaflight #9126: Motor wont spin with Betaflight and CC3D

batata004, 2019-10-28:
Hi,

I succesfully flashed my CC3D board with v3.2.2 - everything went fine. Then I installed latest BETAFLIGHT on my windows, connected the board and it all worked.

Then first I made the BETAFLIGHT "see" the transmitter. I configured if correctly and I can see on RECEIVER tab all the 4 channels receiving the correct values. Then I went to MODES tab and configured the arming - everything just fine.

Now it came time to make the motors spin. First I plugged the battery to the motor and the motor started providing one beep at 2 secs interval (it indicates that the ESC is not "seeing" any signal yet). 

So I went to MOTORS tab, clicked the "I understand..." and started moving the master slider to the top but the motor kept beeping as if nothing was happening.

After many troubleshoot I discovred that I am using those very cheap ESCs that dont support ONESHOT or anything fancy, they only support PWM. So I had to go to CLI and execute "set fast_pwm_protocol = OFF" but this command does not exist anymore on new version. So I changed to "set enable_fast_pwm = OFF" and it accepeted. Then I hit SAVE.

But the problem still remains: the motor keeps beeping and it's not seeing the pulses from the FC. I  cant find a way to fix this. Even going to the CONFIGURATION tab I see no way to set the ESC mode, please check print below:

[image]

PS: the motor and esc works perfectly and everything is wired correctly as I am used to configure betaflight with crazybee.

IvoFPV, 2019-10-28:
>BetaFlight/CC3D_OPBL **2.1.4**
I succesfully flashed my CC3D board with v3.2.2 - everything went fine. 

https://github.com/betaflight/betaflight/releases/tag/v3.2.5
Get the latest version for your FC here and flash it again. It doesn't seem like you've flashed it at all...

batata004, 2019-10-28:
@IvoFPV THANK YOU MAN!! I took a look at the link you provided and in there there is the "betaflight_3.2.5_CC3D_OPBL.hex" file however there is not the BIN! I need the BIN file to flash, right? 

Would you mind telling me how to get the bin? Thanks a lot!

IvoFPV, 2019-10-28:
Download .hex and load it in the configurator by clicking Load Firmware [Local] and selecting that file then just flash it. Make sure you flash with Full chip erase enabled.

mikeller, 2019-10-28:
@IvoFPV: Pretty sure this is not how using the _OPBL version of the firmware works, as it will overwrite the OPBL boot loader.

@batata004: Unless you want to be able to switch back easily to OpenPilot, there is not much reason to not do what @IvoFPV tells you to. Just make sure that you flash `betaflight_3.2.5_CC3D.hex` (without _OPBL) in this case.

batata004, 2019-10-28:
You guys are amazing! Very supportive :) I flashed the `betaflight_3.2.5_CC3D_OPBL.hex` file and the problems remains, after opening BETAFLIGHT the ESC keeps beeping the motor as if no signal is being received on it.

I am pretty sure I wont come back to Librepilot! My only concern of flashing the non `_OPBL` is that if I cant fix this problem with the motor I will be toasted! You think this problem with the motor would be fixed if I use the `betaflight_3.2.5_CC3D.hex` file?

batata004, 2019-10-28:
I said a stupid thing! 5 min ago I said I installed the `.hex` BUT I realized that I installed the old 3.2.2. bin verion. So I tried installing the hex now but LIBREPILOT does not allow .hex, only  opfw and bin files! Please check print below dear friends:

[image]

mikeller, 2019-10-28:
No, it is not likely that `betaflight_3.2.5_CC3D.hex` will fix the problems you are having with the motors.

3.2.2 is probably just as good as 3.2.5 in order to diagnose / solve your problem. Also note that these versions are both very old, since support for your hardware was dropped a while ago, so support is limited.

Can you try setting `set motor_pwm_protocol = pwm` in CLI?

batata004, 2019-10-28:
@mikeller I just checked your link at https://github.com/betaflight/betaflight/releases/tag/v3.2.2 and it does not provide the bin file. Would you mind providing me the latest bin 3.2.x that works on CC3D? I actually lost the BIN file I was using earlier and cant find it anymore! After that I will try `set motor_pwm_protocol = pwm` and check if it solves!

BTW I googled the last 4 to 5 minutes to find the latest .bin file I flashed and I couldnt fine where it was! If you could point me in the right direction maybe it would solve my problem cause maybe I flashed a .bin version that was not from a good!

mikeller, 2019-10-28:
[betaflight_3.2.5_CC3D_OPBL.zip](https://github.com/betaflight/betaflight/files/3781035/betaflight_3.2.5_CC3D_OPBL.zip)

batata004, 2019-10-28:
@mikeller I succesfully flashed the bin file inside the zip file you provided on last comment and after that, when I connect my CC3D to windows, it says the message below:

[image]

Should I have any special driver installed? 

Important: right now I tried flashing a "random" old version of betaflight and I picked this https://github.com/betaflight/betaflight/releases/tag/v2.9.1 -> it has bin file. So I flashed the betaflight_2.9.1_CC3D_OPBL.bin file. Then I disconnected and connected CC3D back to windows (it now recognized the board again) and started betaflight. Then I went on CONFIGURATION tab and THERE WAS NOW AN OPTION TO change ONESHOT to PWM ON THE ESC! Then I tried spining the motor and it worked perfectly!!! \o/

So the question now is how to make a more updated version (like the 3.2.5.) to work. I have no idea why windows didnt recognize the version @mikeller provided.

mikeller, 2019-10-28:
@batata004: Look at the startup page in configurator - there's some hints on how to get your PC to connect to the flight controller.

batata004, 2019-10-28:
@mikeller do yo think I really should install the CP210x Drivers and STM USB VCP Drivers? Cause the 2.9.1 version was perfectly recognized by my pc without any additional driver. If there is no other option I will install it, but just wanna make sure I am not installing anything useless.

batata004, 2019-10-28:
@mikeller I went ahead and installed the 3 drivers.

Latest CP210x Drivers can be downloaded from here
Latest STM USB VCP Drivers can be downloaded from here
Latest Zadig for Windows USB driver installation can be downloaded from here

Restarted windows and it still does not recognize when I connect my CC3D board onto USB using the 3.2.5. version you provided me. Using the 2.9.1 version windows recognizes it perfectly! What do you think I should do?

mikeller, 2019-10-29:
@batata004: I'd stick with 2.9.1 if this is working for you - as I said, both of them are very old and only marginally supported, and if you want to make use of any recent improvements you'll have to upgrade your flight controller to a F4 or F7 based model.

batata004, 2019-10-29:
Thank you so much mikeller!!

batata004, 2019-10-29:
@mikeller  I AM SO SAD!!! After making the motor spin with the 2.9.1 version now I cant make the receiver work! It was working perfectly previously with the other bin version I had flashed in my CC3D which I dont remember the number!

@mikeller would you mind please sending me another .bin version? The version 3.2.5 that you provided I tried everything I could and windows still does not recognize my CC3D. Also, there is something very weird with the 3.2.5 version you provided me cause after flashing it and disconnecting/connecting everything back again to power, ONLY the green light of the cc3d will turn on and nothing else! When I used 2.9.1 version, when I powered my CC3D the blue light and green one would turn on.

Would you mind providing me another version of 3.x bin lower than 3.2.5 so I can try it? 

Thanks in advance!

mikeller, 2019-10-29:
@batata004: Are you sure that you unzipped the file I posted? GitHub does not allow posting of `.bin` files (ironically), so I zipped it.

batata004, 2019-10-29:
@mikeller Yes, I unzipped it, flashed it and didnt work :( I dont want to take your time, so if you have a 3.x file in hand with bin format I would be thankful for your atention, otherwise no problem you were already very supportive!

mikeller, 2019-10-29:
Built / zipped the same version again - I do not think this is a problem of the version, more of something during the build not working right. Try this:

[betaflight_3.2.5_CC3D_OPBL.zip](https://github.com/betaflight/betaflight/files/3785428/betaflight_3.2.5_CC3D_OPBL.zip)

batata004, 2019-10-31:
I will try that today night, now I am at work. Thank you so much!!

batata004, 2019-10-31:
For some reason, just comparing the 2 zips that you sent me, they have a small difference in the size (packed column below). Maybe that bit will fix my problem!

[image]

[image]

batata004, 2019-11-01:
Didnt work either! Both files you provided were different, despite being the build of the same version (I dont know why) but both didnt work. I checked the RELEASES of Betaflight and the latest version that has the BINARIES file is 3.1.7 and I tried flashing it and it worked like a charm! I am still trying to find the latest version I can make work on my CC3D!

If anyone has the latest bin possible I would be glad and I think it would be nice to lots of people in poor countries, like Brazil, where CC3D boards are the main FC used by people cause it's really cheap.

mikeller, 2019-11-01:
@batata004: The difference in size is most likely due to the timestamp of when a firmware was built being included in the firmware - that is normal.
As for 3.2.5 not working, I start to suspect that the CC3D OPBL version is not working for 3.2 in general. How about you take the plunge and try then non-OPBL version?

batata004, 2019-11-05:
THE NON OPBL WORKED!!! Thank you SO MUCH!!!

closed 2019-11-05 by batata004