betaflight/betaflight #9130: Talon Fusion F7

zowie05, 2019-10-29:
**Describe the bug**
Talon Fusion F7 CLRacingF7 unified and the VTX wouldn't power on with a switch set up to do so. I Re-flashed to the legacy and the VTX powered on using the switch. 

**To Reproduce**
Flash Talon Fusion F7 to CLRACINGF7 (CLRA) 

**Expected behavior**
The VTX would power on through the built in pit mode

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Sorry I already re-flashed back to the working Legacy target.

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller Talon Fusion F7 bought from Pyro Drone ;
 - Other components [R-XSR, TBS Unify Pro 32 

**Additional context**

mikeller, 2019-10-29:
You are using the wrong target - use the TALONF7 target.

zowie05, 2019-10-29:
Cool I'll use that one. On Heli-Nation it still says the target is CLRACINGF7

mikeller, 2019-10-30:
@bnn1044: Can you ask HeliNation to update their documentation for this board?

bnn1044, 2019-10-30:
@mikeller  I will get them update today along with PIT mode today.

thanks

closed 2019-12-08 by stale[bot]