betaflight/betaflight #9131: Thoughts about "cinematic cruise"

tigert, 2019-10-29:
I was wondering whether it would be possible to have a flight mode that puts a bit more weight on camera stabilization without a gimbal. Basically there are two things that you want to avoid when flying "cinematic" freestyle: sudden and abrupt tiny control inputs and altitude variations due to throttle inputs.

You can use expo, but it still is more or less direct mapping to your control inputs, it just puts less weight on the center area of the sticks. What I have in mind is a bit similar to the "smooth ink" tool in Photoshop and other design sotware, where you draw with a tablet and pen, and want to avoid your hand shaking showing up in the drawing: https://www.youtube.com/watch?v=nUlkxjUYY9I

The idea might work also for betaflight, and it definitely would not work for general control input, but I think it would be useful as a mode you can switch on and off. Basically if you had throttle and roll/pitch/yaw weighted so that your control inputs would "pull" the quad to a desired direction as if you had a rubber band attached to your stick end, it might result in a smooth, fluid flight path? 

This is just a thought I wanted to share and see whether this would actually work, and I am not sure what kind of practical problems it might have. But maybe this could lead to something interesting.

tigert, 2019-10-29:
Oh, that "pulled string mode" might be interesting as a concept - this, if you could do something similar with your sticks, would allow also abrupt moves and those changes in direction, if you would "keep the string tight" the quad would follow your input smoothly, but if you centered the stick and then pulled the other direction, it would do a sharper change. Hmm. I wonder if this kind of thing would work, and whether it would feel strange or if it would make sense to fly..

haslinghuis, 2019-10-29:
Quote ctzsnooze: 
For smooth cinematic footage, these are some useful things to do
```
- set ff_boost = 0
- use as little feed forward as you can
- set feed forward transition to 0.5
- configure input and derivative smoothing at 10-15hz PT1; or as low as you can go before the delay drives you mad
- turn the ADC off in FrSky Tx’s
- set a deadband in betaflight
- use lots of expo and low rates
- run D at not less than the P value
```

haslinghuis, 2019-10-29:
Betaflight does not have a Race / Freestyle drone switch either. iNav is using presets, Maybe an idea?

bizmar, 2019-11-02:
Filter based RC Smoothing does that exactly :) It's a bit complicated to understand and setup at the moment. Because the numbers are dependent on the speed of new RC commands. Try playing with the Auto smoothness setting in the Remote tab. Try 30, 40, 50. Be aware that latency will go up, a little. If everything goes well (the system correctly reads your average RX rate) even the highest setting of 50, you may want more smoothing for cinematic flying.. (test this where you have a lot of room)

closed 2019-12-09 by stale[bot]