betaflight/betaflight #9133: BARO  MODE

GGYHCLZ, 2019-10-30:
I want  to  use  BARO.  My  Board  is  Matek.
THE  Firmware 3.4.0  support  it.
But  i can't  use  this  mode   after  it  was  configured.

The  barometer  is   well.
It  can  be  detected.

GGYHCLZ, 2019-10-30:
Matek F405

GGYHCLZ, 2019-10-30:
[image]

GGYHCLZ, 2019-10-30:
[image]

mikeller, 2019-10-30:
Baro mode support was dropped as it did not work properly and was not maintained any more.

GGYHCLZ, 2019-10-30:
OK.
I know.

Docteh, 2019-11-06:
@GGYHCLZ  https://github.com/betaflight/betaflight/issues/5893 From the year ago comment it sounds like BARO was removed for 3.4. 

Showing as a mode to choose in 3.4.0 was a forgotten detail. Did the mode show in 3.4.1?

mikeller, 2019-11-06:
@Docteh: There was a bug as in the code for BARO mode was removed for 3.4, but the mode itself was left in, so it was in 3.4, but with no function.

singlag, 2019-11-11:
should add BARO mode back for cinewhoop type flying

closed 2019-12-18 by stale[bot]