betaflight/betaflight #9138: [Feature Request] add throttle to launch control OSD

JyeSmith, 2019-11-01:
When using launch control you can sometimes jump the start when positioning the throttle just before the trigger.

Can the throttle percentage and launch_trigger_throttle_percent be added to the launch control temporary OSD.  This will allow you to push forward on the throttle to setup a 'hair trigger' and minimise the chance of jumping the start.

Cheers

etracer65, 2019-11-01:
There is already a "Throttle Position" element that shows the corresponding throttle percentage. Just use that.

JyeSmith, 2019-11-01:
I'd like for it to disappear after launching and keep a minimal OSD when racing.

etracer65, 2019-11-01:
I'm not sure your concern for not using an extra element that already does what you want justifies changing the element to solve your problem and yet always make the redundant throttle % display for everybody else who may not want it. So using the Throttle Position element gives you the choice to display it or not.

etracer65, 2019-11-01:
And if you want the Throttle Position element to disappear then configure a second OSD profile without that element and switch to it after launching.

mikeller, 2019-11-01:
@JyeSmith, @etracer65: As a compromise, how about starting to blink the 'Launch Control' indicator when the throttle is within 10 percent (or so) of the launch throttle, or add some other form of 'binary' indicator?

JyeSmith, 2019-11-02:
That sounds like a great idea!  It also avoids the need to remeber the trigger percentage.

etracer65, 2019-11-02:
Sounds like a good idea. Should be easy to add.

etracer65, 2019-11-02:
@JyeSmith Try this: #9142

PR #9142 (2019-11-04): Blink the Launch Control warning element when throttle is within 10% of the trigger setting
Fixes #9138 

Provides a visual indication that throttle is close to the trigger limit.
closed 2019-11-03 by JyeSmith