betaflight/betaflight #9139: LUXMINIF7  has NOGYRO and 0%cpu problem

3Dor, 2019-11-01:
Hey,
I have two of the Lumenier LUX7 double gyros fc's that I update to 4.1.0 on "LUXMINIF7" and both of them has NOGYRO, cpu 0% and many more problems like esc is not detected and more.
when I received the fc I connected it by usb to Betaflight and it worked well and I had gyros working and cpu load and everything seemed to work, until I updated the firmware on Betaflight to 4.1.0 "LUXMINIF7". The 4.1.0 is the only version for this fc so I cannot try a different one and I'm sure that it's a huge bug on this firmware, so I assume my fc is ok but I don't have a good firmware for it to work. My main question is if you can send me please .hex file of the stock firmware that the LUX F7 comes with? because this is the only one works and I  couldn't get it on  the internet.
Thanks

etracer65, 2019-11-01:
LUXMINIF7 is not the correct target. That's for a new 20x20 single gyro board. You should contact the manufacturer to determine which target to use.

closed 2019-11-09 by etracer65

mikeller, 2019-11-09:
@x4FF3 is the target maintainer for Lumenier hardware, he should be able to provide the correct target definition.

Asizon, 2019-11-10:
Good read @3Dor
https://www.getfpv.com/media/wysiwyg/pro...Manual.pdf
[image]