betaflight/betaflight #9144: Detected bug on Gyro detection

Asizon, 2019-11-03:
**Describe the bug**
I have detected a gyro detection bug on Unified targets and normal target probably on STMF722 multigyro boards. Introduced by #7914 on commit #1557

**To Reproduce**
A lot of time (8s vs 1s) for initialize gyro when it is powered, maily on unified target 

**Expected behavior**
Fast initialization for gyro/fc

**Flight controller configuration**
AirbotF7 Dualgyro
Stm32F7X2
ICM20602

DIFF in a Buged version
# diff all

# version
# Betaflight / STM32F7X2 (S7X2) 4.0.0 Apr  3 2019 / 13:29:47 (d0deea94a) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name AIRBOTF7
manufacturer_id AIRB
mcu_id 004900203137511930343533
signature 

# name

# resources
resource BEEPER 1 B00
resource MOTOR 1 C08
resource MOTOR 2 B06
resource MOTOR 3 C09
resource MOTOR 4 B07
resource LED_STRIP 1 A15
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource I2C_SCL 1 B08
resource I2C_SDA 1 B09
resource LED 1 B12
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 C00
resource ADC_CURR 1 C01
resource FLASH_CS 1 A03
resource OSD_CS 1 C15
resource GYRO_CS 1 D02
resource GYRO_CS 2 C04

# timer
timer A15 0
timer A08 0
timer C08 0
timer B06 0
timer C09 0
timer B07 0

# dma
dma ADC 1 1
# ADC 1: DMA2 Stream 4 Channel 0
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin C08 0
# pin C08: DMA1 Stream 7 Channel 5
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin C09 0
# pin C09: DMA1 Stream 2 Channel 5
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2

# mixer

# servo

# servo mix

# feature
feature OSD

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set blackbox_device = SPIFLASH
set dshot_burst = ON
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 179
set beeper_inversion = ON
set beeper_od = OFF
set max7456_spi_bus = 2
set dashboard_i2c_bus = 1
set flash_spi_bus = 1
set gyro_1_bustype = SPI
set gyro_1_spibus = 3
set gyro_1_sensor_align = CW90
set gyro_2_spibus = 1

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

etracer65, 2019-11-03:
So just to be clear, you have a board that's defined to have 2 gyros but you've physically disconnected one? If the configuration still expects a gyro to be connected it's not really surprising that the detection takes time while it's trying to talk to a nonexistent device and having to wait for communication to timeout.

If you've removed gyro 2 for example, then do:
```
set gyro_2_bustype = NONE
save
```

Asizon, 2019-11-03:
@etracer65 I dont remove any gyro, this boards like OmnibusF4V6 comes only with one gyro, but with possibility to add more gyros, then they are Dualgyro boards

Asizon, 2019-11-03:
It only happens with unified target, let's wait for @mikeller , I think he will fear the solution

Asizon, 2019-11-03:
Note: This was done by trial and error flashing, the previous build works as expected.

etracer65, 2019-11-03:
Can you please try my suggestion?
```
set gyro_2_bustype = NONE
save
```
This will disable the second gyro. See if the initialization is quicker.

Then next try:
```
set gyro_2_bustype = GYROAUTO
save
```
And see if the initialization is still normal.

Asizon, 2019-11-03:
@etracer65   with ```set gyro_2_bustype = NONE``` the initialization is normal,  with ```GYROAUTO``` the initialization is now slowly again

jflyper, 2019-11-03:
Is the second gyro detected anyways? If so, I suspect `GYROAUTO` looks for I2C first, then SPI.

Asizon, 2019-11-04:
@jflyper When there is no gyro connected it is not detected, if I connect it an external gyro it normally recognized, when mikeller proposed the improvement in gyro detection on #7914  I tested it before being implemented but with 2 gyroscopes installed in an F405. Never get to test with 1 gyro installed and not on a unified target which is the one that most affects

mikeller, 2019-11-04:
@Asizon: As @etracer65 said, the fix is, when using a dual gyro board with only one gyro installed, set `gyro_2_bustype = NONE` - this will work on both Unified and non-unified targets.

Asizon, 2019-11-04:
@mikeller The problem is that these boards are sold with a single gyroscope, by default they should have a fix for that, I am thinking for solutions

Asizon, 2019-11-04:
@x4FF3 What do you think about this?

Asizon, 2019-11-05:
Thanks for the help @etracer65 @jflyper @mikeller , finnaly solved with this
https://github.com/betaflight/unified-targets/pull/102
Closing this issue...

closed 2019-11-05 by Asizon