betaflight/betaflight #9145: SucceX F4 FC after upgrade to 4.1 two of the motors does not function

marcli, 2019-11-03:
**Describe the bug**
I have a turbobee R77 with the SucceX F4 flight controller.
It works fine on 4.0.6 but after flashing with 4.1 I can not get motor 1 and 2 to spin in the motors tab.
If I use the blheli configurator, it only lists esc 3 and 4.
After a downgrade to 4.0.6 it works again.
I also tested with the latest development version, 4.2 from 2nd sept, and it was the same issue.

**To Reproduce**
upgrade to betaflight 4.1

**Expected behavior**
All 4 motors should work.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct 16 2019 / 11:57:34 (c37a7c91a) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF411PRO   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name IFF411PRO
manufacturer_id IFRC
mcu_id 003e004a3437510b32363736
signature 

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# master
set yaw_motors_reversed = ON
set gyro_1_align_yaw = 1800

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0
```

resource show all
```
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
turbobee R77
SucceX F4 flight controller
STM32F411
BF target: MATEKF411

etracer65, 2019-11-03:
Check with the manufacturer to ensure you've flashed the correct target.

marcli, 2019-11-04:
you where right, its was the wrong target, when I ignored the documentation and used IFF411PRO instead it works fine. Closing this off.

closed 2019-11-04 by marcli