betaflight/betaflight #9146: CRAZYBEEF4FS (HAMO) does not bind

ElieLiabeuf, 2019-11-03:
RX does not bind when flashing CRAZYBEEF4FS (HAMO) (unified target, not the legacy one)

**To Reproduce**
* Flash Betaflight 4.1.0 unified target for CRAZYBEEF4FS
* Apply default when connecting the first time with configurator 10.6.0
* Enter `bind_rx` in the cli
* Put your TX in bind mode

**Expected behavior**
Rx and Tx should bind. Note that it works perfectly with the legacy target.

**Configuration UNIFIED TARGED - NOT WORKING**

```
# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct 16 2019 / 11:57:34 (c37a7c91a) MSP API: 1.42
# manufacturer_id: HAMO   board_name: CRAZYBEEF4FS   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FS
manufacturer_id HAMO
mcu_id 0058001e3437511330353933
signature 

# feature
feature -RX_PARALLEL_PWM

# master
set gyro_1_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: RX_SPI_EXTI
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: TIMUP 4
DMA1 Stream 7: TIMUP 2
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Configuration LEGACY - WORKING**

```
# diff all

# version
# Betaflight / CRAZYBEEF4FS (C4FS) 4.1.0 Oct 16 2019 / 11:30:19 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FS
mcu_id 0058001e3437511330353933
signature 

# master
set flysky_spi_tx_id = 1001
set flysky_spi_rf_channels = 42,129,63,135,47,59,115,100,148,94,26,107,35,81,90,72

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: RX_SPI_EXTI
A15: RX_SPI_CS
B00: ADC_BATT
B01: ADC_CURR
B02: RX_SPI_BIND
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: MOTOR 2
B07: MOTOR 3
B08: MOTOR 4
B09: LED
B10: MOTOR 1
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: BEEPER
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 1
TIM3: FREE
TIM4:
    CH1: MOTOR 2
    CH2: MOTOR 3
    CH3: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

Asizon, 2019-11-03:
Can you supply ```dump hardware``` from legacy target in default mode?

ElieLiabeuf, 2019-11-03:
here it is

```
# dump hardware

# version
# Betaflight / CRAZYBEEF4FS (C4FS) 4.1.0 Oct 16 2019 / 11:30:19 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name CRAZYBEEF4FS
manufacturer_id 

# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 NONE
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
resource SERIAL_RX 10 NONE
resource SERIAL_RX 11 NONE
resource LED 1 C13
resource LED 2 NONE
resource LED 3 NONE
resource RX_BIND 1 NONE
resource RX_BIND_PLUG 1 NONE
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 NONE
resource CAMERA_CONTROL 1 NONE
resource ADC_BATT 1 B00
resource ADC_RSSI 1 NONE
resource ADC_CURR 1 B01
resource ADC_EXT 1 NONE
resource PINIO 1 NONE
resource PINIO 2 NONE
resource PINIO 3 NONE
resource PINIO 4 NONE
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 A14
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 B09
resource GYRO_EXTI 1 A01
resource GYRO_EXTI 2 NONE
resource GYRO_CS 1 A04
resource GYRO_CS 2 NONE
resource PULLUP 1 NONE
resource PULLUP 2 NONE
resource PULLUP 3 NONE
resource PULLUP 4 NONE
resource PULLDOWN 1 NONE
resource PULLDOWN 2 NONE
resource PULLDOWN 3 NONE
resource PULLDOWN 4 NONE

# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)

# dma
dma SPI_TX 1 NONE
dma SPI_TX 2 NONE
dma SPI_TX 3 NONE
dma SPI_RX 1 NONE
dma SPI_RX 2 NONE
dma SPI_RX 3 NONE
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma ADC 2 NONE
dma ADC 3 NONE
dma UART_TX 1 NONE
dma UART_TX 2 NONE
dma UART_TX 3 NONE
dma UART_TX 4 NONE
dma UART_TX 5 NONE
dma UART_TX 6 NONE
dma UART_TX 7 NONE
dma UART_TX 8 NONE
dma UART_RX 1 NONE
dma UART_RX 2 NONE
dma UART_RX 3 NONE
dma UART_RX 4 NONE
dma UART_RX 5 NONE
dma UART_RX 6 NONE
dma UART_RX 7 NONE
dma UART_RX 8 NONE
dma pin A03 NONE
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin A02 NONE
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0

# master
set gyro_to_use = FIRST
set rx_spi_bus = 3
set rx_spi_led_inversion = OFF
set adc_device = 1
set blackbox_device = SERIAL
set dshot_burst = AUTO
set dshot_bitbang = OFF
set dshot_bitbang_timer = AUTO
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 179
set beeper_inversion = ON
set beeper_od = OFF
set beeper_frequency = 0
set system_hse_mhz = 8
set max7456_clock = DEFAULT
set max7456_spi_bus = 2
set max7456_preinit_opu = OFF
set led_inversion = 0
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_i2cBus = 0
set gyro_1_i2c_address = 0
set gyro_1_sensor_align = CW90
set gyro_1_align_roll = 0
set gyro_1_align_pitch = 0
set gyro_1_align_yaw = 900
set gyro_2_bustype = SPI
set gyro_2_spibus = 0
set gyro_2_i2cBus = 0
set gyro_2_i2c_address = 0
set gyro_2_sensor_align = CW0
set gyro_2_align_roll = 0
set gyro_2_align_pitch = 0
set gyro_2_align_yaw = 0
set mco2_on_pc9 = OFF

# end the command batch
batch end
```

Asizon, 2019-11-03:
I thin ```flysky_2A_channel_count``` is missing

mikeller, 2019-11-04:
@Asizon: There is no parameter for `flysky_2A_channel_count`, this is only a define. If this is not defined, it will be set in `src/main/rx/a7105_flysky_defs.h`: 

https://github.com/betaflight/betaflight/blob/04c97c849e086eb4c6b290ea58c2faae7d4aa394/src/main/rx/a7105_flysky_defs.h#L39-L41

ElieLiabeuf, 2019-11-04:
I try to reapply the dump hardware to the unified target without loading the default, same result. Rx blink so it's in bind mode but no binding.

How can we progress on this ? Is there something missing on the unified target build ?

Asizon, 2019-11-05:
@ElieLiabeuf thank for your comprovation with config. The unified target config seems correct. Now you says that are put in bindig mode, but not binding data received?tested in legacy target for 2 of more times for test it?
@mikeller thanks for this point, then ```FLYSKY_2A_CHANNEL_COUNT``` is defined on 14 because the max value when it not defined.

User diff all bare with STM32F411 target
```
# diff all bare
# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct 16 2019 / 11:57:34 (c37a7c91a) MSP API: 1.42
# manufacturer_id: HAMO   board_name: CRAZYBEEF4FS   custom defaults: NO
board_name CRAZYBEEF4FS
manufacturer_id HAMO
# resources
resource BEEPER 1 C15
resource MOTOR 1 B10
resource MOTOR 2 B06
resource MOTOR 3 B07
resource MOTOR 4 B08
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A09
resource PWM 3 A10
resource LED_STRIP 1 A00
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource LED 1 C13
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ADC_BATT 1 B00
resource ADC_CURR 1 B01
resource OSD_CS 1 B12
resource RX_SPI_CS 1 A15
resource RX_SPI_EXTI 1 A14
resource RX_SPI_BIND 1 B02
resource RX_SPI_LED 1 B09
resource GYRO_EXTI 1 A01
resource GYRO_CS 1 A04
# timer
timer A03 AF3
# pin A03: TIM9 CH2 (AF3)
timer B10 AF1
# pin B10: TIM2 CH3 (AF1)
timer B06 AF2
# pin B06: TIM4 CH1 (AF2)
timer B07 AF2
# pin B07: TIM4 CH2 (AF2)
timer B08 AF2
# pin B08: TIM4 CH3 (AF2)
timer A00 AF2
# pin A00: TIM5 CH1 (AF2)
timer A02 AF3
# pin A02: TIM9 CH1 (AF3)
timer A09 AF1
# pin A09: TIM1 CH2 (AF1)
timer A10 AF1
# pin A10: TIM1 CH3 (AF1)
# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin B10 0
# pin B10: DMA1 Stream 1 Channel 3
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
dma pin B07 0
# pin B07: DMA1 Stream 3 Channel 2
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin A00 0
# pin A00: DMA1 Stream 2 Channel 6
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0
# feature
feature -RX_PARALLEL_PWM
feature RX_SPI
# master
set rx_spi_protocol = FLYSKY_2A
set rx_spi_bus = 3
set dshot_burst = AUTO
set dshot_bitbang = OFF
set current_meter = ADC
set battery_meter = ADC
set ibata_scale = 179
set beeper_inversion = ON
set beeper_od = OFF
set system_hse_mhz = 8
set max7456_spi_bus = 2
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90
set gyro_1_align_yaw = 900
profile 0
profile 1
profile 2
rateprofile 0
rateprofile 1
rateprofile 2
rateprofile 3
rateprofile 4
rateprofile 5
```

ElieLiabeuf, 2019-11-14:
Yeah I have done the test multiple time, with full erase and the default config. Legacy build is binding, but the unified target is not.

mikeller, 2019-11-15:
@ElieLiabeuf: I had another look into this, but I could not find any difference between the legacy target and the Unified Target that could cause this problem. Unfortunately I do not have a FlySky TX, so I cannot actually verify this beyond looking at the source code.

ElieLiabeuf, 2019-11-15:
@mikeller Yeah I totally understand that. I'm able to change the code and build betaflight myself, maybe you could point me out the right part of the code and I'll do some testing. Not sure how to print debug or something like that though

mikeller, 2019-11-15:
@ElieLiabeuf: If I had an idea what is going wrong I would - but as I said, I can't see why this isn't working.

mikeller, 2019-11-24:
@ElieLiabeuf: I found a potential issue. Can you please test this binary:

[betaflight_4.1.2_STM32F411.zip](https://github.com/betaflight/betaflight/files/3883203/betaflight_4.1.2_STM32F411.zip)
(Select 'CRAZYBEEF4FS (HAMO)' in the drop down, then click 'Load Firmware [Local]', select the unzipped file, click 'Flash Firmware'.)

ElieLiabeuf, 2019-11-24:
@mikeller It works ! It binds and the control are working in the receiver tab. Thanks !

Alexkemp1980, 2019-11-24:
I had the same issue, so have been using Legacy for now.
Just tried the above Zip package, binds fine now

dischinator, 2019-11-24:
Nice I just have been struggling with this problem myself and stumbled over this issue.
Just tried the zipped build, however sadly bindind did not work on my Crazybee F4 3.0 (also Flysky)
When I issue the bind_rx not even the bind led is blinking (on unified led is blinking after bind command, but bid does not happen)

@ElieLiabeuf @Alexkemp1980  on legacy, do you also have the problem RSSI is not send via AUX 10? For me it is sitting at 1500 regardless of actual RSSI.

ElieLiabeuf, 2019-11-24:
@dischinator My board is the version 2.1. Not sure that the problem you are facing is the same. Binding led was blinking after the bind_rx command on my side. In the receiver tab have you set "SPI RX" and "A7105_FLYSKY_2A" ? Have you applied the unified target default conf after flashing ? May be you should consider opening another issue with your particular configuration.

Concerning the RSSI, it was never send via AUX 10 neither on the legacy version nor the unified target. It's available on the OSD but rssi_channel is set to 0. It must be a special feature of SPI RX that send the value by another way (I'm guessing).

dischinator, 2019-11-24:
@ElieLiabeuf yes "SPI RX" and "A7105_FLYSKY_2A" however I did not apply unified target default conf, will try that.

And your right I'll keep my problem out of this issue, too confusion otherwise.

Interesting maybe I have to set RSSI channel to zero to avoid having the 1500 default being displayed. Will try that too, thanks.

dischinator, 2019-11-24:
@ElieLiabeuf thanks for the RSSI hint, after setting rssi_channel to zero RSSI is displayed correctly.

Alexkemp1980, 2019-11-24:
@dischinator mine is version 3.0 as well
I did apply the 'custom settings' that prompt after firmware update.
My RSSI is working OK.

closed 2019-11-25 by mikeller

dischinator, 2019-11-25:
@Alexkemp1980 thats strange after I flashed the zip I did not get a 'custom settings' prompt.

I am going to try unified again once the patch from @mikeller is integrated. Looks like that might be 4.1.2

Asizon, 2019-11-25:
@dischinator now you have available the fix in the last developer build 4.2.0, for zip instalation in local mode, you need to read above Mikellers steps

Asizon, 2019-11-25:
@dischinator https://github.com/betaflight/betaflight/issues/9146#issuecomment-557856454

PR #9218 (2019-11-25): Fixed handling of SPI parameters for Unified Targets.
Potentially fixes #9146.