betaflight/betaflight #9147: FLY FishTower Flytower F4 Buzzer beeps permanently after activate Bidirectional DSHOT

ApophysX, 2019-11-04:
Hi,

I have a FLY FishTower Flytower F4 Flight Controller Targe OMNIBUSF4SD with BF 4.1.0 as soon as I activate Bidirectional DSHOT the buzzer beeps permanently, even if I deactivated all options under Beeper Configuration. Tested with OMNIBUSF4SD AIRB and Legacy.

closed 2019-11-04 by ApophysX

sagiru, 2020-04-13:
Same issue here. I am using a OMNIBUSF4SD Board with Betaflight 4.1.1. I activated the Bidirectional dshod and did the described remapping of the led_strip as described here[1]. After that the buzzer beeps permanently.

[1] https://github.com/betaflight/betaflight/wiki/Board---OMNIBUSF4

dentex, 2022-07-14:
I have the same problem on a Xrotor F4 flight controller (version G2):
https://github.com/betaflight/betaflight/issues/2127
Did you find a solution?