betaflight/betaflight #9149: RPM filter Dshot 600 with 4K 4K burning motors?

TonyAcroFPV, 2019-11-07:
Steps to reproduce the behavior:
I crashed when I went to re-arm and one of the motors instantly went up in smoke, while the other 3 motors twitched. The ESC is fine and I replaced the motor but the only thing I changed was from Dshot 300 to Dshot 600. I am running an F4 and my CPU load is fine. It flew very well I had been testing the RPM filter for a while. I was not getting any error at all too. I think what may have happened is that because I was only running 4k 4k It could not update the ESC in sync or something like that. 

# 

# dump

# version
# Betaflight / STM32F405 (S405) 4.1.0 Oct 16 2019 / 11:57:16 (c37a7c91a) MSP API: 1.42
# manufacturer_id: CLRA   board_name: CLRACINGF4   custom defaults: YES

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# name: Tony-FPV

# resources
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 B08
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 B08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
resource SERIAL_RX 9 NONE
[... 997 lines cut ...]
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 70
set pitch_rc_rate = 72
set yaw_rc_rate = 70
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

#

etracer65, 2019-11-07:
> I **crashed** when I went to re-arm and one of the motors instantly went up in smoke

Crashing has a habit of damaging motors. There is no problem running DSHOT600 with 4K PID loop. But there's also no point in running faster than DSHOT300.

closed 2019-11-07 by etracer65

TonyAcroFPV, 2019-11-08:
Like I said before though, none of the other motors where spin they were all twitching. I know there is no point of running Dshot 600 although I was just testing it.