betaflight/betaflight #9151: 4.1 Crosshair consists of wrong icons

maartenvermeyen, 2019-11-08:
The crosshair consists of the wrong icons as seen on the photo. I have tried to upload the included font again through configurator but the problems persists. Previous version was on 4.0.6 and there was no issue. It also seems I cannot change the font trough the configurator so I'm not sure the upload is successful. Upload seems to be successful in the IDE.

[image]

**Flight controller configuration**
``` # diff

# version
# Betaflight / STM32F411 (S411) 4.1.0 Oct 16 2019 / 11:57:34 (c37a7c91a) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF411PRO   custom defaults: YES

# start the command batch
batch start

board_name IFF411PRO
manufacturer_id IFRC

# name: 120RS

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 6,11::CTOBVI:0
led 1 7,11::CTOBVI:0
led 2 8,11::CTOBVI:0
led 3 9,11::CTOBVI:0

# aux
aux 0 0 0 1575 2100 0 0
aux 1 1 3 1575 2100 0 0
aux 2 13 1 1300 1700 0 0
aux 3 35 1 1700 2100 0 0

# adjrange
adjrange 0 0 2 900 1075 12 2 0 0
adjrange 1 0 2 1400 1600 12 2 0 0

# rxrange
rxrange 0 988 1992
rxrange 1 988 2011
rxrange 2 1007 2005
rxrange 3 988 2011

# rxfail
rxfail 6 s 1000

# master
set gyro_lowpass2_hz = 375
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = 15,-2,-290
set baro_hardware = NONE
set rssi_channel = 9
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT600
set vbat_min_cell_voltage = 320
set vbat_warning_cell_voltage = 340
set current_meter = NONE
set ibata_scale = 210
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_rssi_alarm = 50
set osd_vbat_pos = 2448
set osd_rssi_pos = 2104
set osd_link_quality_pos = 411
set osd_tim_2_pos = 2433
set osd_flymode_pos = 2456
set osd_throttle_pos = 343
set osd_vtx_channel_pos = 311
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 44
set osd_pidrate_profile_pos = 2401
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2441
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set name = 120RS

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set p_pitch = 41
set i_pitch = 81
set d_pitch = 34
set f_pitch = 86
set p_roll = 38
set i_roll = 77
set d_roll = 32
set f_roll = 81
set p_yaw = 27
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 18
set d_min_pitch = 20

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 80
set tpa_breakpoint = 1300

# end the command batch
batch end
```

IvoFPV, 2019-11-08:
Plug in the LiPo while uploading the font. Some boards require that.

maartenvermeyen, 2019-11-08:
Ok, I feel stupid now :). That solved it, thx for the support!

closed 2019-11-08 by maartenvermeyen