betaflight/betaflight #9154: Auto 'soft' land

jeffhendrix, 2019-11-08:
Would like to see a auto soft land. Kind of a hybrid between auto land and gps rescue.

When RX is lost, use the accelerometers to level the quad and establish 1G on the z axis gyro.
Then lower the throttle until .9G (or a settable parameter) on the z axis until a spike is seen. (drone has landed or bumped into something). Then shut down the motors and disarm.

I've had a couple of times where RX single was lost and the drone slammed into the ground, and right after, the RX came back. Going to auto level might give enough time for RX to reestablish. I would also rather have a drone come down slowly instead of falling out of the sky.

skorokithakis, 2019-11-09:
This is a great idea. INAV does landings very well and is a good example of this. As it stands, my quad failsafed earlier because the RX lost contact, and would have slammed into me/my car normally. Luckily, it was close enough that it just disarmed and dropped.

Soft landing would be much better than turning the quad into a projectile on failsafes. A projectile is better than losing it completely, but not by much.

etracer65, 2019-11-09:
> Soft landing would be much better than turning the quad into a projectile on failsafes.

So let me get this straight. You're contending that it's safer to have a fully armed quad with motors and propellers spinning but with absolutely no control or ability to disarm potentially descend onto a person rather than having an inert disarmed craft fall to the ground? Sorry, that doesn't work.

jeffhendrix, 2019-11-09:
I would absolutely rather have a drone come down slowly making noise (with the possibility of regaining RX link), than silently fall out of the sky, which is far more dangerous.

skorokithakis, 2019-11-09:
Yep, what @jeffhendrix said. If you think landing is dangerous, you haven't seen inav land. I would absolutely have it descend slowly making noise rather than silently and randomly fall from the sky.

mikeller, 2019-11-09:
@skorokithakis:

>  If you think landing is dangerous, you haven't seen inav land. 

I think you are not understanding what iNav does, and what Betaflight does:
iNav is focusing on autonomous flight (which includes controlled autonomous landings in areas that are clear of people / objects that can be injured / damaged by rotating props).
Betaflight focuses on performance, and does not have any autonomous modes per se, only damage minimisation modes in the case of loss of control.

If you want autonomous landing then iNav is what you should be using.

skorokithakis, 2019-11-09:
You're posing a false dichotomy. What's so fundamental about Betaflight turning your quad into a projectile on failsafe? Why can safe landing not be implemented? I'm not satisfied with "because it looks too much like autonomous flying", that doesn't answer the question.

You're basically saying you want safety, but not *too much* safety. The other day I had a dodgy connection between my FC and RX and I had to choose whether I wanted to risk my quad flying into my face on RX loss or going home. With a soft landing, I wouldn't have to choose.

mikeller, 2019-11-09:
@skorokithakis: Can you please define what 'safe landing' is for you, i.e. what you are attempting to protect?

etracer65, 2019-11-09:
@skorokithakis 

> The other day I had a dodgy connection between my FC and RX and I had to choose whether I wanted to risk my quad flying into my face on RX loss or going home. With a soft landing, I wouldn't have to choose.

The **only** safe choice would have been to go home and fix the problem. Flying with a failing radio link is not wise regardless of the failsafe methods available. If for some ridiculous reason you were flying in such a way that you were concerned about the quad "flying into your face" it would seem that doing so with the props spinning would be the least desirable outcome.

skorokithakis, 2019-11-09:
@mikeller My face, mainly.

> If for some ridiculous reason you were flying in such a way that you were concerned about the quad "flying into your face"

mikeller, 2019-11-09:
@skorokithakis: 

> @mikeller My face, mainly.

So in your opinion it is better to have a craft fly into your face with rotating props uncontrollably while executing a 'soft landing' than to have one crash into you with stopped props?

I am pretty sure that everybody who has ever had to deal with the type of injury caused by spinning high powered props typically found on the type of craft used for Betaflight will disagree with you.

skorokithakis, 2019-11-09:
> So in your opinion it is better to have a craft fly into your face with rotating props uncontrollably while executing a 'soft landing' than to have one crash into you with stopped props?

This sentence makes me think you've never seen a soft landing. Ignoring the blatant exaggeration, I will quote my earlier comment:

> I would absolutely have it descend slowly making noise rather than silently and randomly fall from the sky.

mikeller, 2019-11-09:
@skorokithakis: 

> This sentence makes me think you've never seen a soft landing. Ignoring the blatant exaggeration, I will quote my earlier comment:

You really do not seem capable of understanding the difference between 'controlled soft landing' and 'uncontrollable craft with spinning props flying into people'. I have seen both, and I do not wish to see the latter ever again.

> I would absolutely have it descend slowly making noise rather than silently and randomly fall from the sky.

Ignoring the fact that you are comparing apples to oranges here, as it is really only a choice between 'randomly falling from the sky while props are powered' and 'randomly falling from the sky while props are off', if you are concerned about the 'silently' part then the correct action for you is to add a beeper.

etracer65, 2019-11-09:
@skorokithakis

So setting aside the dubious proposition that a "soft landing" is somehow safer, How do you propose we accomplish this? Betaflight has no position or altitude hold capability. Meaning the code doesn't exist and would have to be written. And we've already stated that these autonomous types of flight capabilities are not within the goals of the project. But assume you wanted to pursue this and implement these capabilities. Then you're left with the following issues:

So the first problem is that you're flying along at high speed and suddenly failsafe. The self-level part is no problem, but you'll still be traveling forward at a high rate of speed. There is no "braking" or ability to stop and hold position. And implementing position hold would require GPS which only a small percentage of typical Betaflight users would have. So basically the quad would try to "descend slowly" while still traveling forward straight at your face with the props spinning.

Next, how do we "descend slowly"? To do that we need a measure of vertical velocity. This can be derived from altitude sources but unfortunately only a small percentage of flight controllers even have a barometer. And for people that do have them they rarely have them properly installed with the required shielding from airflow to make them accurate enough. The alternative is GPS which can also provide altitude. But the problem again is availability and further that GPS altitude is very inaccurate (commonly varying by +-5m) and updates too slowly (sometimes a slow as 1Hz, but typically 5Hz). Since we're operating close to the ground and intending to move slowly, a high level of accuracy is required.

So finally as already stated, if this is a capability that's important to you then consider using a flight control firmware like iNav that has goals more closely aligned with your needs.

skorokithakis, 2019-11-09:
> And implementing position hold would require GPS which only a small percentage of typical Betaflight users would have.

I'm talking about soft landing at the end of the RTH procedure, which already needs a GPS to work, so that solves that problem. The barometer is a small issue, but the craft flying over you at `gps_rescue_initial_alt` centimeters and then descending slowly (or not so slowly) seems preferable to speeding towards you.

mikeller, 2019-11-09:
@skorokithakis: There is no RTH procedure in Betaflight. If you want RTH you need to use iNav.

skorokithakis, 2019-11-09:
There is GPS Rescue, which randomly crashes the craft near you instead of away from you. What I'm saying is it could soft-land instead, the way INAV does.

mikeller, 2019-11-09:
@skorokithakis: 

> There is GPS Rescue, which randomly crashes the craft near you instead of away from you. What I'm saying is it could soft-land instead, the way INAV does.

This seems to be based on you misunderstanding how GPS rescue works, and how it is intended to be used.
Please read https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode (and in particular (https://github.com/betaflight/betaflight/wiki/GPS-rescue-mode#what-this-is-not).
As stated before, if you want Return to Home functionality, which is different to what GPS rescue is, please use iNav.

Closing this.

closed 2019-11-09 by mikeller